coolsnowwolf/lede #2842: 默认编译报错

EstFoxs, 2020-01-23:
make[3]: Entering directory `/home/bbq/lede/package/lean/v2ray'
mkdir -p /home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/bbq/lede/tmp/.root-copy.flock -c 'cp -fpR /home/bbq/lede/build_dir/target-x86_64_musl/v2ray-4.21.3/.pkgdir/v2ray/. /home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.v2ray_installed
echo "v2ray" >> /home/bbq/lede/staging_dir/target-x86_64_musl/pkginfo/v2ray.default.install
make[3]: Leaving directory `/home/bbq/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#0.02#0.00#0.11
make[3]: Entering directory `/home/bbq/lede/package/lean/v2ray-core'
mkdir -p /home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/bbq/lede/tmp/.root-copy.flock -c 'cp -fpR /home/bbq/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.pkgdir/v2ray-core/. /home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/'
cp: cannot overwrite directory '/home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/./usr/bin/v2ray' with non-directory
make[3]: *** [/home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.v2ray-core_installed] Error 1
make[3]: Leaving directory `/home/bbq/lede/package/lean/v2ray-core'
time: package/lean/v2ray-core/compile#0.04#0.02#0.14
make[2]: *** [package/lean/v2ray-core/compile] Error 2
make[2]: Leaving directory `/home/bbq/lede'
make[1]: *** [/home/bbq/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/bbq/lede'
make: *** [world] Error 2

KFERMercer, 2020-01-23:
网络问题导致的 v2ray-core 编译错误. 暂时不要选v2ray-core, 还是用v2ray代替.

KFERMercer, 2020-01-23:
@Mattraks 太巧了吧

EstFoxs, 2020-01-23:
等修复呗

Mattraks, 2020-01-23:
需要将Network --->  <*> v2ray
*取消

KFERMercer, 2020-01-23:
> 需要将Network ---> <*> v2ray
> *取消

我用我的分支 (还没merge上游) 试过了也不行. v2ray那边服务器抽风了.

Mattraks, 2020-01-23:
你在Makefile加入以下使用代理编译
export GO111MODULE=on
export GOPROXY=https://goproxy.cn

hcym, 2020-01-23:
瞎折腾啊，这个go编译最烦人，十几个莫名其妙源地址踢来踢去，还是用大雕的二进制文件吧

KFERMercer, 2020-01-23:
> 你在Makefile加入以下使用代理编译
> export GO111MODULE=on
> export GOPROXY=https://goproxy.cn
> 
> 目前大雕合并的源码已经集成了go代理

现在的情况是actions里面也跑不出来.

D-Bood, 2020-01-23:
`cp: cannot overwrite directory /home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/./usr/bin/v2ray' with non-directory`
是这个问题导致的

原因是之前编译v2ray的时候在
`/home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/`
生成了`./usr/bin/v2ray`这个文件夹，而新增加的`v2ray-core`编译生成的`./usr/bin/v2ray`则是一个可执行文件，所以导致冲突

我这边是把
`/home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/./usr/bin/v2ray`
删除，然后手动修改.config，注释掉`CONFIG_PACKAGE_v2ray=y` (因为我之前已经编译过了)，确认`CONFIG_PACKAGE_v2ray-core=y`之后重新编译，搞定了

garypang13, 2020-01-23:
> `cp: cannot overwrite directory /home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/./usr/bin/v2ray' with non-directory`
> 是这个问题导致的
> 
> 原因是之前编译v2ray的时候在
> `/home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/`
> 生成了`./usr/bin/v2ray`这个文件夹，而新增加的`v2ray-core`编译生成的`./usr/bin/v2ray`则是一个可执行文件，所以导致冲突
> 
> 我这边是把
> `/home/bbq/lede/staging_dir/target-x86_64_musl/root-x86/./usr/bin/v2ray`
> 删除，然后手动修改.config，注释掉`CONFIG_PACKAGE_v2ray=y` (因为我之前已经编译过了)，确认`CONFIG_PACKAGE_v2ray-core=y`之后重新编译，搞定了
> 
> 可能我说的有些地方不是很对，大概解决过程就是这样的，目前还没发现有什么问题😂

注释掉  CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y 后在Actions中编译成功了.

D-Bood, 2020-01-23:
> 注释掉 CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y 后在Actions中编译成功了.
对，我把这个给忘了说了

lphgor, 2020-01-23:
最近的更新都很激进呀，大杂烩式的更新，还改动默认配置

KFERMercer, 2020-01-23:
已经提交修复了, 等待拉取 ↑

closed 2020-01-23 by coolsnowwolf