coolsnowwolf/lede #2844: 默认配置编译报错

demo0102, 2020-01-23:
Entering directory '/home/username/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.93'
make[5]: *** No rule to make target 'tools/objtool/objtool', needed by '/home/username/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/sch_cake.o'.  Stop.
Makefile:1520: recipe for target '_module_/home/username/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388' failed
make[4]: *** [_module_/home/username/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388] Error 2
make[4]: Leaving directory '/home/username/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.93'
Makefile:42: recipe for target '/home/username/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/.built' failed
make[3]: *** [/home/username/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/.built] Error 2
make[3]: Leaving directory '/home/username/lede/package/kernel/kmod-sched-cake'
time: package/kernel/kmod-sched-cake/compile#0.93#11.82#16.49
package/Makefile:107: recipe for target 'package/kernel/kmod-sched-cake/compile' failed

xuesumin, 2020-01-23:
cp: cannot overwrite directory '/home/jerry/lede/staging_dir/target-x86_64_musl/root-x86/./usr/bin/v2ray' with non-directory
make[3]: *** [/home/jerry/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.v2ray-core_installed] Error 1
make[3]: Leaving directory `/home/jerry/lede/package/lean/v2ray-core'
time: package/lean/v2ray-core/compile#0.08#0.04#0.28
make[2]: *** [package/lean/v2ray-core/compile] Error 2
make[2]: Leaving directory `/home/jerry/lede'
make[1]: *** [/home/jerry/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/jerry/lede'
make: *** [world] Error 2

demo0102, 2020-01-23:
> cp: cannot overwrite directory '/home/jerry/lede/staging_dir/target-x86_64_musl/root-x86/./usr/bin/v2ray' with non-directory
> make[3]: *** [...]

 你发错位置了

closed 2020-01-23 by coolsnowwolf