coolsnowwolf/lede #2851: V2死活过不去，求助

kl9800, 2020-01-24:
早上7点多，删除了全部项目然后新拉的代码，
1.  git clone https://github.com/coolsnowwolf/lede
2.  cd lede
2.  ./scripts/feeds update -a 
3.  ./scripts/feeds install -a
4.  make clean 
5.  make menuconfig
6.  make download V=s (dl下的包全部成功)
7.  make  -j1  V=s
######################### 报错信息 ######################

Cleaning module download cache (golang/go#27455)

mv -f /home/koll/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build/bin/main /home/koll/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build/bin/v2ray
upx --lzma --best /home/koll/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build/bin/v2ray
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2013
UPX 3.91        Markus Oberhumer, Laszlo Molnar & John Reiser   Sep 30th 2013

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
v2ray  4/8  [***************************************************.]   28.5%  | make[3]: *** [/home/koll/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built] Segmentation fault (core dumped)
make[3]: Leaving directory `/home/koll/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#52.19#8.78#29.16
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory `/home/koll/lede'
make[1]: *** [/home/koll/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/koll/lede'
make: *** [world] 错误 2

blizzardster, 2020-01-24:
一样的问题，我手动更新了dL文件文件正常，

kl9800, 2020-01-24:
手动更新“v2ray-core-4.22.1.tar.gz” ？？

kl9800, 2020-01-24:
啊看错了我以为你说已经好了呢

blizzardster, 2020-01-24:
> 手动更新“v2ray-core-4.22.1.tar.gz” ？？

据issues有人说，替换版本就好了，但是我编译成功过一次，因该不是版本的问题

blizzardster, 2020-01-24:
> 啊看错了我以为你说已经好了呢

用了这个依赖库也不行
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex node-uglify gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx-ucl libelf-dev autoconf automake libtool autopoint device-tree-compiler libuv-dev git-core

blizzardster, 2020-01-24:
L大说是UPX不支持高通，问题是我们编译X86

kl9800, 2020-01-24:
> L大说是UPX不支持高通，问题是我们编译X86

我也是X86

blizzardster, 2020-01-24:
> 用了这个依赖库也不行
> sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex node-uglify gcc-multilib [...]
我已经过了，现在卡其他地方了，晕，我在重新编译下试试

hgh188, 2020-01-24:
一样，搞了一天都没办法过

blizzardster, 2020-01-24:
> 一样，搞了一天都没办法过

我已经在重新编译了，用了git pull，有更新了，我在看看

EstFoxs, 2020-01-24:
v2ray  4/8  [***************************************************.]   28.5%  | make[3]: *** [/home/fox/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built] Segmentation fault (core dumped)
make[3]: Leaving directory `/home/fox/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#18.96#2.23#27.86
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory `/home/fox/lede'
make[1]: *** [/home/fox/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/fox/lede'
make: *** [world] Error 2
fox@fox:~/lede$ git pull
Already up-to-date.

KFERMercer, 2020-01-24:
可能是upx版本太低.

＃＃ 统计一下你们编译用的系统和内核版本, 请在下面回复, 以便定位问题.

kl9800, 2020-01-24:
乌班图14

KFERMercer, 2020-01-24:
推荐系统: ubuntu18 debian9 deepin15

kl9800, 2020-01-24:
好吧换掉系统去试试

kl9800, 2020-01-24:
sudo apt-get -y install 。。。 这个后面需要增加什么吗？

KFERMercer, 2020-01-24:
今天的CI自动编译顺利通过就已经证明了v2ray没有问题了

EstFoxs, 2020-01-24:
收到，换系统，刚好有18

hcym, 2020-01-24:
编译没问题，arm和x64都过了，还没刷

KFERMercer, 2020-01-24:
有没有用上述给出的建议系统编译依然失败的? 请回复

kl9800, 2020-01-24:
正在换系统，我会把结果发到这里，预计8小时以后。。。

hgh188, 2020-01-24:
取消upx后编译成功 但V2RAY不能够运行

AmadeusGhost, 2020-01-24:
编译机内存最好给大一点
[image]

AmadeusGhost, 2020-01-24:
> L大说是UPX不支持高通，问题是我们编译X86

不是这意思，原话是armv8上可能有问题，升级upx版本试试

kl9800, 2020-01-24:
已经换了Ubuntu18， 正在重新编译

tcsr200722, 2020-01-24:
deepin 15编译失败

hcym, 2020-01-24:
别折腾编译了，v2不能运行，ss正常，非要用v2的话退回去吧

d975xbx, 2020-01-24:
编译没问题，问题是SSR PLUS里面的v2ray不能启动。

coolsnowwolf, 2020-01-24:
[image]

kl9800, 2020-01-24:
换了Ubuntu18以后，原本V2那里卡住的问题已经通过了

closed 2020-01-24 by kl9800