coolsnowwolf/lede #2852: 关于今天V2ray出现的问题的解答

KFERMercer, 2020-01-24:
根据大家的反馈, 现在主要是两个问题: 编译失败和不能运行.

- 待编辑

KFERMercer, 2020-01-24:
今天回老家过年了, 可能编辑有点慢, 大家见谅.

tcsr200722, 2020-01-24:
ovo/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/lenovo/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.trojan_installed
echo "trojan" >> /home/lenovo/lede/staging_dir/target-x86_64_musl/pkginfo/trojan.default.install
make[3]: Leaving directory '/home/lenovo/lede/package/lean/trojan'
time: package/lean/trojan/compile#0.14#0.05#0.23
make[3]: Entering directory '/home/lenovo/lede/package/lean/v2ray'
rm -f /home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built
touch /home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built_check
( cd /home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build ; export GOPATH=/home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build GOCACHE=/home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/cache GOTMPDIR=/home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/tmp GOROOT_FINAL=/usr/lib/go-1.13 CC=x86_64-openwrt-linux-musl-gcc CXX=x86_64-openwrt-linux-musl-g++ GOOS=linux GOARCH=amd64 GO386=387 GOARM= GOMIPS= GOMIPS64= CGO_ENABLED=1 CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1:v2ray-core-4.22.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/lenovo/lede/staging_dir/target-x86_64_musl/usr/include -I/home/lenovo/lede/staging_dir/target-x86_64_musl/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1:v2ray-core-4.22.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" ; echo "Finding targets" ; targets=$(go list v2ray.com/core/main) ; for pattern in ; do targets=$(echo "$targets" | grep -v "$pattern") ; done ; echo ; if [ "" = 1 ]; then echo "Calling go generate" ; go generate -v  $targets ; echo ; fi ; if [ "" != 1 ]; then echo "Building targets" ; case amd64 in arm)             installsuffix="v" ;; mips|mipsle)     installsuffix="" ;; mips64|mips64le) installsuffix="" ;; esac ; ldflags="-linkmode external -extldflags '-L/home/lenovo/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/lenovo/lede/staging_dir/target-x86_64_musl/lib -L/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/home/lenovo/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib -Wl,-z,now -Wl,-z,relro'" ; pkg_gcflags="" ; pkg_ldflags="-s -w" ; for def in v2ray.com/core.version=4.22.1 v2ray.com/core.build=R5 v2ray.com/core.codename=OpenWrt; do pkg_ldflags="$pkg_ldflags -X $def" ; done ; go install ${installsuffix:+-installsuffix $installsuffix} -trimpath -ldflags "all=$ldflags" -v ${pkg_gcflags:+-gcflags "$pkg_gcflags"} ${pkg_ldflags:+-ldflags "$pkg_ldflags $ldflags"}  $targets ; retval=$? ; echo ; if [ "$retval" -eq 0 ] && [ -z "$(find /home/lenovo/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build/bin -maxdepth 0 -type d \! -empty 2>/dev/null)" ]; then echo "No binaries were generated, consider adding GO_PKG_SOURCE_ONLY:=1 to Makefile" ; echo ; fi ; echo "Cleaning module download cache (golang/go#27455)" ; go clean -modcache ; echo ; fi ; exit $retval ; )
Finding targets
go: downloading github.com/golang/protobuf v1.3.2
go: downloading golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
go: extracting github.com/golang/protobuf v1.3.2
go: extracting golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
go: downloading go.starlark.net v0.0.0-20190919145610-979af19b165c
go: downloading golang.org/x/net v0.0.0-20190311183353-d8887717615a
go: downloading google.golang.org/grpc v1.24.0
go: extracting go.starlark.net v0.0.0-20190919145610-979af19b165c
go: downloading golang.org/x/sys v0.0.0-20190215142949-d0b11bdaac8a
go: extracting golang.org/x/net v0.0.0-20190311183353-d8887717615a
go: extracting google.golang.org/grpc v1.24.0
go: downloading github.com/gorilla/websocket v1.4.1
go: extracting golang.org/x/sys v0.0.0-20190215142949-d0b11bdaac8a
go: extracting github.com/gorilla/websocket v1.4.1
go: downloading google.golang.org/genproto v0.0.0-20180831171423-11092d34479b
go: downloading golang.org/x/text v0.3.0
go: downloading github.com/refraction-networking/utls v0.0.0-20190909200633-43c36d3c1f57
go: extracting google.golang.org/genproto v0.0.0-20180831171423-11092d34479b
go: extracting github.com/refraction-networking/utls v0.0.0-20190909200633-43c36d3c1f57
go: extracting golang.org/x/text v0.3.0
go: finding github.com/golang/protobuf v1.3.2
go: finding google.golang.org/grpc v1.24.0
go: finding golang.org/x/net v0.0.0-20190311183353-d8887717615a
go: finding golang.org/x/sys v0.0.0-20190215142949-d0b11bdaac8a
go: finding go.starlark.net v0.0.0-20190919145610-979af19b165c
go: finding google.golang.org/genproto v0.0.0-20180831171423-11092d34479b
go: finding github.com/refraction-networking/utls v0.0.0-20190909200633-43c36d3c1f57
go: finding golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
go: finding golang.org/x/text v0.3.0
go: finding github.com/gorilla/websocket v1.4.1

Building targets

[... cut ...]
package/Makefile:107: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/lenovo/lede'
package/Makefile:103: recipe for target '/home/lenovo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lenovo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lenovo/lede'
/home/lenovo/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

blizzardster, 2020-01-24:
过了，现在卡这里
go: downloading golang.org/x/sys v0.0.0-20190215142949-d0b11bdaac8a
go: downloading github.com/gorilla/websocket v1.4.1
go: extracting github.com/gorilla/websocket v1.4.1
go: downloading google.golang.org/genproto v0.0.0-20180831171423-11092d34479b
go: downloading golang.org/x/text v0.3.0
go: extracting golang.org/x/sys v0.0.0-20190215142949-d0b11bdaac8a

coolsnowwolf, 2020-01-24:
问题找到了：

你们的 UPX 版本过低，请升级到 UPX 3.95 以上版本

https://github.com/upx/upx/releases

hyird, 2020-01-24:
整个docker编译环境得了

coolsnowwolf, 2020-01-24:
[image]

blizzardster, 2020-01-24:
> 根据大家的反馈, 现在主要是两个问题: 编译失败和不能运行.
> 
> * 待编辑
> 
> ## 最新消息: 梯子塌了, 推迟更新到初一

是的V2RAY，挂了

hcym, 2020-01-24:
幽默，铁拐李自己瘸腿了，哈哈😄，不过我的也是挂了个，还有三个不瘸

hyird, 2020-01-24:
我白嫖的2个 都没有挂。。

EstFoxs, 2020-01-24:
v2ray挂了，有剩一台小ji活着。

segadrive, 2020-01-24:
本来就是好好的呀

demo0102, 2020-01-25:
同样的问题，升级upx版本重新编译好了

d975xbx, 2020-01-25:
大神，3.94的怎么升级到3.96？

coolsnowwolf, 2020-01-25:
所以大家明白为什么提供我预编译的 V2Ray 版本更好了吧?

z826540272, 2020-01-25:
> 所以大家明白为什么提供我预编译的 V2Ray 版本更好了吧?

那为啥还改,好用体积小就行.

KFERMercer, 2020-01-25:
维护是个灾难

z826540272, 2020-01-25:
能不能把v2ray的体积再精简一些,去除用不到的功能,比如SS

outemancut, 2020-01-26:
> 同样的问题，升级upx版本重新编译好了

如何更新？

rampageX, 2020-01-26:
UPX bug 最少的还是 3.94

lrz2010, 2020-01-27:
我升级了UPX v2ray还是打不开啊 请问如何关闭UPX？

SouHong, 2020-01-28:
ubuntukylin-16.04-desktop-amd64系统中UPX 版本3.91编译的K3固件可以正常运行。
[image]

KFERMercer, 2020-01-28:
新版本已经默认把upx关掉了.

lrz2010, 2020-01-28:
> 新版本已经默认把upx关掉了.

现在我不管开不开UPX 都无法运行

pirately, 2020-01-28:
各位大佬，请问是否v2ray现在是tcp模式可以，但是mKcp不行？

unifreq, 2020-01-29:
个人认为低容量闪存的路由器也不需要upx, 因为squashfs本身就是带压缩的了，例如在ar7xxx架构的老路由上压缩率也是很高的。

tefiszx, 2020-01-31:
> 新版本已经默认把upx关掉了.

怎么把upx支持打开啊？编译的时候加参数还是改配置文件啊？

closed 2020-01-31 by coolsnowwolf