coolsnowwolf/lede #2857: 最新ssrplus 连接v2ray服务器失败

pirately, 2020-01-25:
ssrplus是最新的1.36,利用云编译。v2ray服务器端是4.22.1.服务器是搬瓦工。电脑端用V2RayN，手机端用V2RayNG连接搬瓦工的V2Ray没有问题，但是路由器用ssrplus连接搬瓦工，总是显示连接测试失败。请问，这是为何？
V2Ray是，mKcp，WeChatVedio模式，加密方式是Auto

KFERMercer, 2020-01-25:
确实有连接某些v2ray配置失败的问题, 核心是用v2ray的源码编译应该没问题吧, 难不成是模板的问题?

hcym, 2020-01-25:
今天刚刚编译的只连接了一个ipv6的v2，目测正常，普通mkcp

kl9800, 2020-01-25:
我也是昨天编译全部通过了，但是发现V2不能用，所以安装了旧版本

d975xbx, 2020-01-25:
编译没问题，ssrplus不能启动

hcym, 2020-01-25:
大雕把那个压缩文件关闭后编译可以用了，昨天没关闭的确实不能启动v2

kl9800, 2020-01-25:
我就是关闭upx之后，git pull https://github.com/coolsnowwolf/lede

kl9800, 2020-01-25:
然后编译完以后不能用

kl9800, 2020-01-25:
好像忘了升级UPX了。。。

kl9800, 2020-01-25:
乌班图系统换成了18
升级了UPX到3.96，
然后开启了UPX，
然后编译
所有问题得以解决，V2也可以正常使用了

pirately, 2020-01-25:
> 大雕把那个压缩文件关闭后编译可以用了，昨天没关闭的确实不能启动v2

请问，怎么关闭？是"CONFIG_TARGET_IMAGES_GZIP=n"？

pirately, 2020-01-25:
> 乌班图系统换成了18
> 升级了UPX到3.96，
> 然后开启了UPX，
> 然后编译
> 所有问题得以解决，V2也可以正常使用了

请问怎么操作？

pirately, 2020-01-25:
> 确实有连接某些v2ray配置失败的问题, 核心是用v2ray的源码编译应该没问题吧, 难不成是模板的问题?

K大，我换了Vultr的服务器，路由器自己还是连不上。

kl9800, 2020-01-25:
> > 乌班图系统换成了18
> > 升级了UPX到3.96，
> > 然后开启了UPX，
> > 然后编译
> > 所有问题得以解决，V2也可以正常使用了
> 
> 请问怎么操作？

哪步？

KFERMercer, 2020-01-25:
> 我换了Vultr的服务器，路由器自己还是连不上。

先看下其他人的反馈

kl9800, 2020-01-25:
> > 大雕把那个压缩文件关闭后编译可以用了，昨天没关闭的确实不能启动v2
> 
> 请问，怎么关闭？是"CONFIG_TARGET_IMAGES_GZIP=n"？

1. 你可以回到 37757cf2ce75fade6ee13f601af9f4b10716ad0a
2. 或者手动改回UPX那俩文件
package/lean/v2ray/Config.in
package/lean/v2ray/Makefile
3.升级UPX到3.95+（必须）
然后再编译就好了
系统我是ubuntu 18

pirately, 2020-01-25:
> > > 大雕把那个压缩文件关闭后编译可以用了，昨天没关闭的确实不能启动v2
> > 
> > 
> > 请问，怎么关闭？是"CONFIG_TARGET_IMAGES_GZIP=n"？
> 
> 1. 你可以回到 [37757cf](https://github.com/coolsnowwolf/lede/commit/37757cf2ce75fade6ee13f601af9f4b10716ad0a)
> 2. 或者手动改回UPX那俩文件
>    package/lean/v2ray/Config.in
>    package/lean/v2ray/Makefile
>    3.升级UPX到3.95+（必须）
>    然后再编译就好了
>    系统我是ubuntu 18

请问怎么升级UPX?

kl9800, 2020-01-25:
> 请问怎么升级UPX?

1.  下载 upx-3.96-amd64_linux.tar.xz (我是ubuntu 18)
2.  tar -Jxf upx*.tar.xz
3.  sudo cp upx*/upx /usr/bin

结束

lrz2010, 2020-01-27:
> 我就是关闭upx之后，git pull https://github.com/coolsnowwolf/lede

请问如何关闭UPX

kl9800, 2020-01-27:
> > 我就是关闭upx之后，git pull https://github.com/coolsnowwolf/lede
> 
> 请问如何关闭UPX

85ca0440541aef1423360ca64858c87eb1d8fdb1

lrz2010, 2020-01-27:
> > > 我就是关闭upx之后，git pull https://github.com/coolsnowwolf/lede
> > 
> > 
> > 请问如何关闭UPX
> 
> [85ca044](https://github.com/coolsnowwolf/lede/commit/85ca0440541aef1423360ca64858c87eb1d8fdb1)

谢谢  但是我关闭了之后仍然无法运行 而且没有无线选项 是小米路由器3g

kl9800, 2020-01-27:
> > > > 我就是关闭upx之后，git pull https://github.com/coolsnowwolf/lede
> > > 
> > > 
> > > 请问如何关闭UPX
> > 
> > 
> > [85ca044](https://github.com/coolsnowwolf/lede/commit/85ca0440541aef1423360ca64858c87eb1d8fdb1)
> 
> 谢谢 但是我关闭了之后仍然无法运行 而且没有无线选项 是小米路由器3g

我当时关闭UPX以后虽然编译成功，但是V2就不能用了，所以你最好想办法打开UPX

pirately, 2020-01-27:
> > > > > 我就是关闭upx之后，git pull https://github.com/coolsnowwolf/lede
> > > > 
> > > > 
> > > > 请问如何关闭UPX
> > > 
> > > 
> > > [85ca044](https://github.com/coolsnowwolf/lede/commit/85ca0440541aef1423360ca64858c87eb1d8fdb1)
> > 
> > 
> > 谢谢 但是我关闭了之后仍然无法运行 而且没有无线选项 是小米路由器3g
> 
> 我当时关闭UPX以后虽然编译成功，但是V2就不能用了，所以你最好想办法打开UPX

请问，怎么打开upx？我是以前比较长时间用的云编译。这两天建的虚拟机，下的最新的源码，upx升级到了最新，但是v2ray依然运行失败。

kl9800, 2020-01-28:
package/lean/v2ray/Config.in
package/lean/v2ray/Makefile

kl9800, 2020-01-28:
> 请问，怎么打开upx？我是以前比较长时间用的云编译。这两天建的虚拟机，下的最新的源码，upx升级到了最新，但是v2ray依然运行失败。

package/lean/v2ray/Config.in
package/lean/v2ray/Makefile

kl9800, 2020-01-28:
> package/lean/v2ray/Config.in
> package/lean/v2ray/Makefile

最新的代码 UPX默认是禁用的，所以你得修改这俩文件启用
package/lean/v2ray/Config.in   30行改成 y
package/lean/v2ray/Makefile   13行改回5 

85ca0440541aef1423360ca64858c87eb1d8fdb1  参照这个反着改回来即可

lrz2010, 2020-01-28:
> 译成功，但是V2就不能用了，所以你最好想办法打开UPX

我不管打开还是关闭都没法用

pirately, 2020-01-28:
> > > > > > > > 我就是关闭upx之后，git pull https://github.com/coolsnowwolf/lede
> > > > > > > 
> > > > > > > 
> > > > > > > 请问如何关闭UPX
> > > > > > 
> > > > > > 
> > > > > > [85ca044](https://github.com/coolsnowwolf/lede/commit/85ca0440541aef1423360ca64858c87eb1d8fdb1)
> > > > > 
> > > > > 
> > > > > 谢谢 但是我关闭了之后仍然无法运行 而且没有无线选项 是小米路由器3g
> > > > 
> > > > 
> > > > 我当时关闭UPX以后虽然编译成功，但是V2就不能用了，所以你最好想办法打开UPX
> > > 
> > > 
> > > 请问，怎么打开upx？我是以前比较长时间用的云编译。这两天建的虚拟机，下的最新的源码，upx升级到了最新，但是v2ray依然运行失败。
> > 
> > 
> > package/lean/v2ray/Config.in
> > package/lean/v2ray/Makefile
> 
> 最新的代码 UPX默认是禁用的，所以你得修改这俩文件启用
> package/lean/v2ray/Config.in 30行改成 y
> package/lean/v2ray/Makefile 13行改回5
> 
> [85ca044](https://github.com/coolsnowwolf/lede/commit/85ca0440541aef1423360ca64858c87eb1d8fdb1) 参照这个反着改回来即可

改回来了，打开了upx。但是V2Ray运行也失败了。都是新建的虚拟机。

kl9800, 2020-01-28:
> 改回来了，打开了upx。但是V2Ray运行也失败了。都是新建的虚拟机。

那就想不到为什么了，好奇怪，是不是节点问题？

pirately, 2020-01-28:
> 那就想不到为什么了，好奇怪，是不是节点问题？

我换了一个服务器，重新用官方的V2Ray配置方式配置，也是一样的不能启动

pirately, 2020-01-28:
> 那就想不到为什么了，好奇怪，是不是节点问题？

用TCP模式可以，用mKcp就不行

pirately, 2020-01-28:
> > 我换了Vultr的服务器，路由器自己还是连不上。
> 
> 先看下其他人的反馈

K大，v2ray如果是tcp模式可以，如果是mKcp+WechatVideo则不行。已经替换成了狼大的-7版

closed 2020-02-19 by coolsnowwolf