coolsnowwolf/lede #2862: lean大的openwrt的的外网访问op的端口是否可以关闭？或者是否可以修改端口号80为其他端口号

norikohxb, 2020-01-25:
lean大的openwrt的的外网访问op的端口是否可以关闭？或者是否可以修改端口号80为其他端口号
一直在把lean大的op当作旁路由用，主路由用ros，现在自己变异而来friendlywrt+lean大的package，也已经在使用了，因为是作为主路由，把nano pi R1s当作主路由。所以想要问一下，这个外网访问的端口能否关闭？或者能否修改端口号，万望知情的大神们告诉我一下。

pV-nRT, 2020-01-25:
从wan口默认情况下是无法访问openwrt的web界面的，至于修改web界面端口的方法，网上有很多，善用搜索引擎

norikohxb, 2020-01-26:
当我把op作为一个旁路由的时候看，把它的wan接口到我的局域网，通过DHCP获得wan口IP，这个时候用wan的IP是可以通过80访问设置的。是不是在你启用了PPPOE的时候，OP的防火墙就自动屏蔽了80web的外网管理

pV-nRT, 2020-01-26:
我从vbox上面创建1了一个openwrt的虚拟机，其wan口使用桥接网卡的形式接入了我的局域网，如图获得了一个ip地址192.168.1.158，此时我在电脑上打开这个网址，无法打开，so，固件没有问题。你应该排除下自己的网络接法以及路由器设置
[image]

norikohxb, 2020-01-26:
你可能误解我的问题了，当把R1S的WAN口接在内部网络上，我其实是可以通过wan口的ip访问的，所以我是在请教，当他是自己独立做主路由的时候是否是可以通过外网+80端口远程控制路由器的。
我问的是这个意思。

hyird, 2020-01-26:
> 我的

你独立的时候 外网+80端口 默认是无妨访问的 
web坚挺的端口号可以修改 可以自行修改uhttpd的配置文件 或者安装luci-app-webadmin这个包 可以修改端口的

hyird, 2020-01-26:
> 你可能误解我的问题了，当把R1S的WAN口接在内部网络上，我其实是可以通过wan口的ip访问的，所以我是在请教，当他是自己独立做主路由的时候是否是可以通过外网+80端口远程控制路由器的。
> 我问的是这个意思。

你独立的时候 外网+80端口 默认是无妨访问的 
web坚挺的端口号可以修改 可以自行修改uhttpd的配置文件 或者安装luci-app-webadmin这个包 可以修改端口的

norikohxb, 2020-01-26:
这个马上编译进去，luci-app-webadmin这个才是我这个问题的精华答案啊。要的就是这个。
还有厦门电信你封锁了80端口投诉工信部有用吗？

norikohxb, 2020-01-26:
> > 你可能误解我的问题了，当把R1S的WAN口接在内部网络上，我其实是可以通过wan口的ip访问的，所以我是在请教，当他是自己独立做主路由的时候是否是可以通过外网+80端口远程控制路由器的。
> > 我问的是这个意思。
> 
> 你独立的时候 外网+80端口 默认是无妨访问的
> web监听的端口号可以修改 可以自行修改uhttpd的配置文件 或者安装luci-app-webadmin这个包 可以修改端口的
感谢你的答复啊

hyird, 2020-01-26:
> 这个马上编译进去，luci-app-webadmin这个才是我这个问题的精华答案啊。要的就是这个。
> 还有厦门电信你封锁了80端口投诉工信部有用吗？

封80 那应该是硬性规定   国内架设web 是要备案的  小心被封宽带  人家扫你端口 v2ex已经有人被封了 签个保证书才行

norikohxb, 2020-01-26:
现在加入了你说的那个webadmin，就可以修改端口了
我如何关闭 那个22端口 ssh的端口，因为我发现我在家里可以远程用xshell去连接我的那个nano pi R1S H5.这个22
端口如何在编译的时候就去掉，我不要让他可以远程被xshell登陆。

hyird, 2020-01-26:
> 现在加入了你说的那个webadmin，就可以修改端口了
> 我如何关闭 那个22端口 ssh的端口，因为我发现我在家里可以远程用xshell去连接我的那个nano pi R1S H5.这个22
> 端口如何在编译的时候就去掉，我不要让他可以远程被xshell登陆。
大哥。。 你能去  “系统>管理权”   那里看看先么

hyird, 2020-01-26:
> 现在加入了你说的那个webadmin，就可以修改端口了
> 我如何关闭 那个22端口 ssh的端口，因为我发现我在家里可以远程用xshell去连接我的那个nano pi R1S H5.这个22
> 端口如何在编译的时候就去掉，我不要让他可以远程被xshell登陆。

你能先去  “系统>管理权”  看看 再说么

pV-nRT, 2020-01-26:
> 现在加入了你说的那个webadmin，就可以修改端口了
> 我如何关闭 那个22端口 ssh的端口，因为我发现我在家里可以远程用xshell去连接我的那个nano pi R1S H5.这个22
> 端口如何在编译的时候就去掉，我不要让他可以远程被xshell登陆。

默认情况下这个22端口和80端口一样，都是不能被外网访问的，我怀疑你是在上一级路由里面把你的pi设置为dmz，这会导致你这台pi完全暴露在外网中

hyird, 2020-01-26:
> > 现在加入了你说的那个webadmin，就可以修改端口了
> > 我如何关闭 那个22端口 ssh的端口，因为我发现我在家里可以远程用xshell去连接我的那个nano pi R1S H5.这个22
> > 端口如何在编译的时候就去掉，我不要让他可以远程被xshell登陆。
> 
> 默认情况下这个22端口和80端口一样，都是不能被外网访问的，我怀疑你是在上一级路由里面把你的pi设置为dmz，这会导致你这台pi完全暴露在外网中

对的 默认 只监听lan区域  你要自己设置成未指定 或者wan 才可以被访问到   
不过dmz 也支持暴漏 下级路由的wan ，lan不会 你这个假设不成立

pV-nRT, 2020-01-26:
> > > 现在加入了你说的那个webadmin，就可以修改端口了
> > > 我如何关闭 那个22端口 ssh的端口，因为我发现我在家里可以远程用xshell去连接我的那个nano pi R1S H5.这个22
> > > 端口如何在编译的时候就去掉，我不要让他可以远程被xshell登陆。
> > 
> > 
> > 默认情况下这个22端口和80端口一样，都是不能被外网访问的，我怀疑你是在上一级路由里面把你的pi设置为dmz，这会导致你这台pi完全暴露在外网中
> 
> 对的 默认 只监听lan区域 你要自己设置成未指定 或者wan 才可以被访问到
> 不过dmz 也支持暴漏 下级路由的wan ，lan不会 你这个假设不成立

重新看了他的描述，确实是我疏忽了，他是在局域网内打开pi就出现了可以直接访问web界面的情况

norikohxb, 2020-01-27:
看到了，在管理权里面可以设置22端口号也可以设置访问网卡权限，不错。现在我要的都行了，谢谢你。还有openwrt好像不需要设置组播对吧？天生自带的。我记得主路由的是ROS还必须去开通一个组播设置，机顶盒看直播的节目才可以看。这个openwrt好像是天生自带这个功能。不需要再去开通。

norikohxb, 2020-01-27:
> > > 现在加入了你说的那个webadmin，就可以修改端口了
> > > 我如何关闭 那个22端口 ssh的端口，因为我发现我在家里可以远程用xshell去连接我的那个nano pi R1S H5.这个22
> > > 端口如何在编译的时候就去掉，我不要让他可以远程被xshell登陆。
> > 
> > 
> > 默认情况下这个22端口和80端口一样，都是不能被外网访问的，我怀疑你是在上一级路由里面把你的pi设置为dmz，这会导致你这台pi完全暴露在外网中
> 
> 对的 默认 只监听lan区域 你要自己设置成未指定 或者wan 才可以被访问到
> 不过dmz 也支持暴漏 下级路由的wan ，lan不会 你这个假设不成立

非常感谢受益非浅

9l, 2020-01-27:
> 这个马上编译进去，luci-app-webadmin这个才是我这个问题的精华答案啊。要的就是这个。
> 还有厦门电信你封锁了80端口投诉工信部有用吗？

没有用，这就是工信部要求的

norikohxb, 2020-01-28:
谢谢大家的回答！

closed 2020-01-28 by norikohxb