coolsnowwolf/lede #2864: Shadowsocks New Version 加入 obfs=tls

jyy5206, 2020-01-25:
@coolsnowwolf 

首先感谢您的奉献，让我学会了自己编译。（本人非 IT 业者）

我的机场本来仅提供【纯 SS】 服务。
从 2019 年国庆起，机场逐渐把【纯 SS】改为了【SS + obfs=tls】。
从昨天起，基本上都改为了【SS + obfs=tls】。

能否帮忙把 obfs=tls 功能加入到 Shadowsocks New Version 里？

不情之请，请多包涵。

谢谢！

dolphinpaopao, 2020-01-25:
需要这种方式的话建议用V2，单独的SS+obfs不太推荐了。

jyy5206, 2020-01-25:
谢谢跟帖。

完全理解并赞同您的意见。

我的机场 = Justmysocks = 小众，他们没有 V2Ray 的服务 :(
所以我的需求=小众，况且此项目是 Mr. Lean 业余时间自愿免费在开发/维护，我自己都觉得很不好意思。。。

我尝试过在R9.12.31版本中，手动安装了 simple-obfs，可是在 Shadowsocks New Version 添加/编辑服务器配置中，并没有蹦出来 obfs 的选择框。:(

所以只能求助于您们专业人士了。

不管结果如何，都表示诚挚的感谢！

dolphinpaopao, 2020-01-26:
这个需要适配，尝试right论坛上编译的魔改固件看是否有相关功能。
另外可以参考这篇文章源自Google获取：http://blog.kompaz.win/2019/01/13/20190113%20OpenWRT%E5%8D%87%E7%BA%A7%E5%88%B018.06.1/

jyy5206, 2020-01-26:
谢谢！
我先学习下。
有问题再来请教。

qdfansanyblog, 2020-01-27:
同用Justmysocks，上个月发工单报故障，建议官方加入v2ray和trojan协议，至令没回应，反面3号服务器也改用了 obfs=tls这种方式

hcym, 2020-01-27:
哈哈，瓦工鸡场，用原版ss官方1907编译吧，还行，n1的防火墙也不用加什么规则

jyy5206, 2020-01-27:
我刚刚也给 Justmysocks 发了 Ticket，建议增加 V2Ray 。
多一个人反应，应该多一份效果吧。
看看有没有效果。。。

jyy5206, 2020-01-27:
JMS回复倒挺快。
态度 = No 。

> We do not limit on what devices you use our services. However unfortunately we are not able to provide specific instructions for such custom setups.
> 
> Note: I am a robot and my reply is based on the information provided in the ticket, as well as diagnostics data I have access to.
> If your request was not properly addressed, please reply in this ticket and one of our engineers will take a look.
> 
> Thank you.

用来用去，在路由器上，还是觉得 Mr. Lean 的版本最快，设置起来也快。
还望各位大佬帮忙加入 obfs 下。
谢谢！

hcym, 2020-01-27:
担心多余，混淆都没必要，瓦工机场速度和我的v3一样，我自己搭建的就是不如v2，上次封的那么多，机场没事，ip也就是同段的换了几次

hplee0120, 2020-01-27:
其实你说的并不算小众，现在挺多主流机场都已经从以前的xxr替换为ss+obfs和v2了。不过不清楚什么原因大雕一直没添加，ssr++魔改版插件倒是有支持obfs，或者你可以考虑一下。不过我感觉用来用去还是原版比较简洁，魔改版太过于花里胡哨。

jyy5206, 2020-01-28:
2个没想到：
1）Justmysocks 用户这里也有；
2）SS+obfs 组合竟然正在变成主流。

看大佬们是否有空增加 SS+obfs 的组合了。。。

不好意思白嫖。如何让我能捐献几杯咖啡？

lyin888, 2020-01-31:
看看恩山这位网友的帖子
https://www.right.com.cn/forum/thread-1632897-1-1.html

lyin888, 2020-01-31:
https://github.com/honwen/openwrt-v2ray-plugin

bigbighill, 2020-02-01:
1、https://github.com/bigbighill/lede
集成了 v2ray-plusin 及openplash 支持 
obfs参数填入位置见图片

[image]
2、也可使用 openclash，以clash 支持ss +plugin
3、已加入gitgub active workflow，可自行修改。

jyy5206, 2020-02-01:
> 看看恩山这位网友的帖子
> https://www.right.com.cn/forum/thread-1632897-1-1.html

@lyin888 您好！

感谢您的指导！
根据此贴，我照做了。

结果：头几分钟可以使用SS+obfs了，之后就不行了。症状为红色字体“ShadowsocksR Plus+ 未运行”始终出现。无论如何折腾，都无法出现绿色”ShadowsocksR Plus+ 运行中“。
哪怕重启路由也没用。

细节：
我是 R9.12.31 版本，Netgear 4300V1 路由。
obfs插件：http://openwrt-dist.sourceforge.net/packages/base/mips_24kc/simple-obfs_0.0.5-4_mips_24kc.ipk
opkg install 之后，看到 /usr/bin 里有 obfs-local 了，应该是安装成功了。

vi /usr/lib/lua/luci/model/cbi/shadowsocksr/client-config.lua ：
（中间2段是额外插入的代码）

> o = s:option(Value, "server_port", translate("Server Port"))
> o.datatype = "port"
> o.rmempty = false
> 
> o = s:option(Value, "plugin", "Plugin")
> o:depends("type", "ss")
> 
> o = s:option(Value, "plugin_opts", "Plugin Opts")
> o:depends("type", "ss")
> 
> -- o = s:option(Value, "timeout", translate("Connection Timeout"))
> -- o.datatype = "uinteger"
> -- o.default = 60
> -- o.rmempty = false

之后，在“编辑服务器配置”页面，成功出现了 Plugin 和 Plugin Opts 二个输入框。

然后 vi /etc/init.d/shadowsocksr ：
（local plugin=$..... 起，到 fi ，是额外插入的代码）

> if [ "$stype" == "ss" ] ;then
> 	cat <<-EOF >$config_file
> 		{
> 		    "server": "$hostip",
> 		    "server_port": $(uci_get_by_name $1 server_port),
> 		    "local_address": "0.0.0.0",
> 		    "local_port": $(uci_get_by_name $1 local_port),
> 		    "password": "$(uci_get_by_name $1 password)",
> 		    "timeout": $(uci_get_by_name $1 timeout 60),
> 		    "method": "$(uci_get_by_name $1 encrypt_method_ss)",
> 		    "reuse_port": true,
> 		    "fast_open": $fastopen
>     local plugin=$(uci_get_by_name $1 plugin)
>     if [ -n "$plugin" -a -x "/usr/bin/$plugin" ]; then
>         sed -i "s@$hostip\",@$hostip\",\n    \"plugin\": \"$plugin\",\n    \"plugin_opts\": \"$(uci_get_by_name $1 plugin_opts)\",@" $config_file
>     fi
> 		}
> EOF
>        elif [ "$stype" == "ssr" ] ;then
> 	cat <<-EOF >$config_file

麻烦您有空帮我看下，我哪里做得不对？

谢谢！

jyy5206, 2020-02-01:
@bigbighill 您好！

您的指导看到了，谢谢！
好多地方不懂。
为不打扰您的时间，我先自己学习下。
等7788有个数之后，再以最少的问题，花您最少的时间，来请教您。
祝好！

bigbighill, 2020-02-01:
> if [ "$stype" == "ss" ] ;then
> cat <<-EOF >$config_file
> {
> "server": "$hostip",
> "server_port": $(uci_get_by_name $1 server_port),
> "local_address": "0.0.0.0",
> "local_port": $(uci_get_by_name $1 [...]

if [ "$stype" == "ss" ] ;then
cat <<-EOF >$config_file
{
"server": "$hostip",
"server_port": $(uci_get_by_name $1 server_port),
"local_address": "0.0.0.0",
"local_port": $(uci_get_by_name $1 local_port),
"password": "$(uci_get_by_name $1 password)",
"timeout": $(uci_get_by_name $1 timeout 60),
"method": "$(uci_get_by_name $1 encrypt_method_ss)",
"reuse_port": true,
"fast_open": $fastopen
}
EOF
local plugin=$(uci_get_by_name $1 plugin)
if [ -n "$plugin" -a -x "/usr/bin/$plugin" ]; then
sed -i "s@$hostip",@$hostip",\n "plugin": "$plugin",\n "plugin_opts": "$(uci_get_by_name $1 plugin_opts)",@" $config_file
fi

bigbighill, 2020-02-01:
另外在  /etc/inti.d/shadowoskr  最后部分添加 
	killall -q -9 v2ray-plugin

参见  https://github.com/bigbighill/lede/blob/master/package/lean/luci-app-ssr-plus/root/etc/init.d/shadowsocksr

jyy5206, 2020-02-02:
尊敬的 @bigbighill ，

非常感谢您的指导！
按照您的教程，成功用上了 SS+obfs 组合！

> 另外在 /etc/inti.d/shadowoskr 最后部分添加
> killall -q -9 v2ray-plugin

这个我没执行。
因为：
1）系统已顺利运行；
2）我的插件是obfs，不是v2ray。

再次感谢！

hcym, 2020-02-02:
姨，谢谢分享，我也试试看看，瓦工就有一个节点最快，用的混淆，目前用的原版ss，没大雕的方便

jyy5206, 2020-02-02:
感谢 @bigbighill  @lyin888 和众多专家的指导，成功用上了 SS+obfs 组合！

我从本项目索取颇多，反哺回报为必须。

总结路线图如下，供其他有需要的朋友参考：

环境：
R9.12.31
Netgear 4300V1

步骤：
1）进入 Openwrt web页面：系统 => 软件包

2）【下载并安装软件包】的框里输入： 
http://openwrt-dist.sourceforge.net/packages/base/mips_24kc/simple-obfs_0.0.5-4_mips_24kc.ipk
点击确认。数秒钟后页面顶部会提示安装完毕。
（此为安装obfs插件）

3）更改文件  vi /usr/lib/lua/luci/model/cbi/shadowsocksr/client-config.lua
第145行起，插入额外代码（中间2段浅灰字体）
我把上下文也列出来了，可有效避免小白（如我）插错地方。

o = s:option(Value, "server_port", translate("Server Port"))
o.datatype = "port"
o.rmempty = false

> o = s:option(Value, "plugin", "Plugin")
> o:depends("type", "ss")
> 
> o = s:option(Value, "plugin_opts", "Plugin Opts")
> o:depends("type", "ss")

-- o = s:option(Value, "timeout", translate("Connection Timeout"))
-- o.datatype = "uinteger"
-- o.default = 60
-- o.rmempty = false

和我环境相同的，可直接使用我自己修改好的文件，来替换原文件。
下载： https://github.com/jyy5206/SS_obfs/blob/master/client-config.lua

4）更改文件  vi /etc/init.d/shadowsocksr
第105行起，插入额外代码（最下面5段浅灰字体）
我把上下文也列出来了，可有效避免小白（如我）插错地方。

if [ "$stype" == "ss" ] ;then                                                                                                                    
        cat <<-EOF >$config_file                                                                                                                 
                {                                                                                                                                
                    "server": "$hostip",                                                                                                         
                    "server_port": $(uci_get_by_name $1 server_port),                                                                            
                    "local_address": "0.0.0.0",                                                                                                  
                    "local_port": $(uci_get_by_name $1 local_port),                                                                              
                    "password": "$(uci_get_by_name $1 password)",                                                                                
                    "timeout": $(uci_get_by_name $1 timeout 60),                                                                                 
                    "method": "$(uci_get_by_name $1 encrypt_method_ss)",                                                                         
                    "reuse_port": true,                                                                                                          
                    "fast_open": $fastopen                                                                                                       
                }  
EOF

>     local plugin=$(uci_get_by_name $1 plugin)
>     if [ -n "$plugin" -a -x "/usr/bin/$plugin" ]; then
>         sed -i "s@$hostip\",@$hostip\",\n    \"plugin\": \"$plugin\",\n    \"plugin_opts\": \"$(uci_get_by_name $1 plugin_opts)\",@" $config_file
>     fi

和我环境相同的，可直接使用我自己修改好的文件，来替换原文件。
下载： https://github.com/jyy5206/SS_obfs/blob/master/shadowsocksr

5）Done

当然，如果 Mr. Lean 团队能直接在源码中加入 SS+obfs 功能，那会大大方便我们小白。

善莫大焉！

jyy5206, 2020-02-02:
效果一样，但是更简单的方法：

SSH上去后，逐行执行：

> wget http://openwrt-dist.sourceforge.net/packages/base/mips_24kc/simple-obfs_0.0.5-4_mips_24kc.ipk
> opkg install simple-obfs_0.0.5-4_mips_24kc.ipk
> wget https://github.com/jyy5206/SS_obfs/blob/master/client-config.lua
> cp -f client-config.lua /usr/lib/lua/luci/model/cbi/shadowsocksr
> wget https://github.com/jyy5206/SS_obfs/blob/master/shadowsocksr
> cp -f shadowsocksr /etc/init.d

hcym, 2020-02-03:
感谢楼主和层主，完全正常，插件是用源码编译的，就是ss官方那个，又多了一个出路

closed 2020-02-03 by jyy5206

bigbighill, 2020-02-04:
> 尊敬的 @bigbighill ，
> 
> 非常感谢您的指导！
> 按照您的教程，成功用上了 SS+obfs 组合！
> 
> > 另外在 /etc/inti.d/shadowoskr 最后部分添加
> > killall -q -9 v2ray-plugin
> 
> 这个我没执行。
> 因为：
> 1）系统已顺利运行；
> 2）我的插件是obfs，不是v2ray。
> 
> 再次感谢！
可以把v2ray-plugin换成你用的obfs插件
否则重启服务时老进程没有kill，另外又生成多obfs进程。

jyy5206, 2020-02-04:
谢谢 @bigbighill 
github 上的文件已更新：https://github.com/jyy5206/SS_obfs

jyy5206, 2020-02-05:
为什么我放在 github 上的3个文件，wget 下来，发现文件长度、内容都不对了呢？
https://github.com/jyy5206/SS_obfs/blob/master/simple-obfs_0.0.5-4_mips_24kc.ipk
https://github.com/jyy5206/SS_obfs/blob/master/client-config.lua
https://github.com/jyy5206/SS_obfs/blob/master/shadowsocksr

比如 client-config.lua，正确的内容开头应该是：
```
-- Copyright (C) 2017 yushi studio <ywb94@qq.com> github.com/ywb94
-- Licensed to the public under the GNU General Public License v3.

local m, s, o,kcp_enable
local shadowsocksr = "shadowsocksr"
local uci = luci.model.uci.cursor()
```

但是我 wget 下来后，内容开头变成了：
```
<!DOCTYPE html>
<html lang="en">
  <head>
    <meta charset="utf-8">
  <link rel="dns-prefetch" href="https://github.githubassets.com">
  <link rel="dns-prefetch" href="https://avatars0.githubusercontent.com">
```
而且文件长度也对不上了。

不懂，求教。

picmax, 2020-02-06:
多搜一会就搞定了嘛。自己compile下面两个。LEDE原有的SSR还是原始的2.5板，完全落后于时代。

https://github.com/shadowsocks/openwrt-shadowsocks
https://github.com/shadowsocks/luci-app-shadowsocks

hcym, 2020-02-06:
原版是正常使用，没有大雕的方便

wind007, 2020-02-07:
学到了除了魔改 又多了一条路

wind007, 2020-02-07:
wget 有问题的兄弟 用这两个地址吧
https://raw.githubusercontent.com/jyy5206/SS_obfs/master/client-config.lua
https://raw.githubusercontent.com/jyy5206/SS_obfs/master/shadowsocksr

mmeiwenlong, 2020-02-07:
> wget的问题有兄弟的这用地址两个吧
> [https://raw.githubusercontent.com/jyy5206/SS_obfs/master/client-config.lua ](https://raw.githubusercontent.com/jyy5206/SS_obfs/master/client-config.lua)
> https://raw.githubusercontent.com/jyy5206/SS_obfs/master/shadowsocksr

我x86下载哪个
[image]

wind007, 2020-02-07:
> > wget的问题有兄弟的这用地址两个吧
> > [https://raw.githubusercontent.com/jyy5206/SS_obfs/master/client-config.lua ](https://raw.githubusercontent.com/jyy5206/SS_obfs/master/client-config.lua)
> > https://raw.githubusercontent.com/jyy5206/SS_obfs/master/shadowsocksr
> 
> 我x86下载哪个
> [image]

不带 server 那个

mmeiwenlong, 2020-02-07:
> 另外在/etc/inti.d/shadowoskr最后部分添加
> killall -q -9 v2ray-plugin
> 
> 参见 https://github.com/bigbighill/lede/blob/master/package/lean/luci-app-ssr-plus/root/etc/init.d/shadowsocksr

我的是x86-64架构。按照教程也改了。大还是不行

> > > wget的问题有兄弟的这用地址两个吧
> > > [https://raw.githubusercontent.com/jyy5206/SS_obfs/master/client-config.lua ](https://raw.githubusercontent.com/jyy5206/SS_obfs/master/client-config.lua)
> > > https://raw.githubusercontent.com/jyy5206/SS_obfs/master/shadowsocksr
> > 
> > 
> > 我x86下载那里
> > [image]
> 
> 不带服务器那个

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
好像都正常就是不行

hcym, 2020-02-07:
你用楼主那两个文件覆盖一下编译，肯定行，什么x86，n1还有mtk都行

wind007, 2020-02-07:
> 我的是x86-64架构。按照教程也改了。大还是不行
>
>
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> 好像都正常就是不行

plugin 参数填了什么 ? 贴出来看看

hcym, 2020-02-07:
混淆插件用ss官方的obsf就可以，我用v2plug的反而不行

mmeiwenlong, 2020-02-07:
> plugin

plugin这个在哪

wind007, 2020-02-07:
参考一下吧
[image]

wind007, 2020-02-07:
参考一下吧
plugin 填 obfs-local 
plugin opts 填 你的混淆方式 和 混淆host 例子 obfs=tls;obfs-host=www.baidu.com

mmeiwenlong, 2020-02-07:
> 参考一下吧
> plugin 填 obfs-local
> plugin opts 填 你的混淆方式 和 混淆host 例子 obfs=tls;obfs-host=www.baidu.com

[image]
我在机场查询到的

wind007, 2020-02-07:
> > 参考一下吧
> > plugin 填 obfs-local
> > plugin opts 填 你的混淆方式 和 混淆host 例子 obfs=tls;obfs-host=www.baidu.com
> 
> [image]
> 我在机场查询到的

你按我的例子填 看看能不能用吧

mmeiwenlong, 2020-02-07:
> > > 参考一下吧
> > > 插件填obfs本地
> > > 插件OPTS填你的混淆方式和混淆主机例子obfs = TLS; obfs宿主= www.baidu.com
> > 
> > 
> > [image]
> > 我在机场查询到的
> 
> 你按我的例子填看看能不能用吧

选择节点也启动了。就是不能走出去而节点是好的。
[image]

wind007, 2020-02-07:
> > > > 参考一下吧
> > > > 插件填obfs本地
> > > > 插件OPTS填你的混淆方式和混淆主机例子obfs = TLS; obfs宿主= [www.baidu.com](http://www.baidu.com)
> > > 
> > > 
> > > [image]
> > > 我在机场查询到的
> > 
> > 
> > 你按我的例子填看看能不能用吧
> 
> 选择节点也启动了。就是不能走出去而节点是好的。
> [image]

你plugin 填 obfs-local     
plugin opts 填 obfs=tls;obfs-host=www.baidu.com
看看能不能用吧

mmeiwenlong, 2020-02-07:
> > > > > 参考一下吧
> > > > > 插件填充obfs本地
> > > > > 插件OPTS填充你的替换方式和替代主机实例obfs = TLS；obfs宿主= [www.baidu.com](http://www.baidu.com)
> > > > 
> > > > 
> > > > [image]
> > > > 我在机场查询到的
> > > 
> > > 
> > > 你按我的例子填看看能不能用吧
> > 
> > 
> > 选择宣布也启动了。就是不能走出去而异步是好的。
> > [image]
> 
> 你plugin填obfs-local
> 插件opt填obfs = tls; obfs-host = www.baidu.com
> 看看能不能用吧

这样会显示红色不能运行了。这里还有个问号
[image]

wind007, 2020-02-07:
> > > > > > 参考一下吧
> > > > > > 插件填充obfs本地
> > > > > > 插件OPTS填充你的替换方式和替代主机实例obfs = TLS；obfs宿主= [www.baidu.com](http://www.baidu.com)
> > > > > 
> > > > > 
> > > > > [image]
> > > > > 我在机场查询到的
> > > > 
> > > > 
> > > > 你按我的例子填看看能不能用吧
> > > 
> > > 
> > > 选择宣布也启动了。就是不能走出去而异步是好的。
> > > [image]
> > 
> > 
> > 你plugin填obfs-local
> > 插件opt填obfs = tls; obfs-host = [www.baidu.com](http://www.baidu.com)
> > 看看能不能用吧
> 
> 这样会显示红色不能运行了。这里还有个问号
> [image]

那就不懂了，帮不了你了

jyy5206, 2020-02-07:
@mmeiwenlong 
Plugin: obfs-local
Plugin Opts: obfs=tls;obfs-host=www.bing.com （注意：整行都没有空格）

wooleo, 2020-02-08:
谢谢，我也是just的机场，带obfs的服务器用的人少一些，能稍微稳定一点

hcym, 2020-02-09:
线路最重要，其他真的无所谓，我还想着怎么取消混淆呢

jyy5206, 2020-02-12:
今天发现 SS+objs 已经合并到源码了。
一次性编译成功。

就我使用的 Netgear 4300V1 路由，共享固件如下：
全新安装: https://github.com/jyy5206/OpenWRT-firmware-R20.1.24-for-Netgear-4300V1/blob/master/openwrt-ar71xx-nand-wndr4300-ubi-factory.img?raw=true
更新安装: https://github.com/jyy5206/OpenWRT-firmware-R20.1.24-for-Netgear-4300V1/blob/master/openwrt-ar71xx-nand-wndr4300-squashfs-sysupgrade.tar?raw=true

再次感谢 Mr. Lean 和众多专家一如既往的贡献！

zhEdward, 2020-02-29:
卧槽。同款4300v1 找到救命稻草了，拿来先用啦@jyy5206，我的机场主强制更换SSR加了**协议参数**和**混淆参数**选项，几年前恩山论坛找的（只支持SS）固件嗝屁。

这么一搞，是时候考虑学习 `Github Actions` 走自我编译持续迭代了，防止再次被机场主扼住命运的喉咙。

SoPudge, 2020-03-09:
> @mmeiwenlong
> Plugin: obfs-local
> Plugin Opts: obfs=tls;obfs-host=www.bing.com （注意：整行都没有空格）

给后来人Mark一下，这个方案管用，我是X86，Lean的版本，这个参数可以用JMS的obfs混淆

royiie, 2020-03-12:
@SoPudge 是在LEAN OPENWRT那个版本里带的ShadowSocksR Plus+里添加的么？

jyy5206, 2020-03-13:
R20.1.15添加的
之后到现在，此功能都内置了

jyy5206, 2020-03-24:
今天发现首页有捐赠码了
很高兴的捐赠了点
晚上编程序累了，喝点咖啡吧 :)

DanielChan27, 2020-04-16:
> 感谢 @bigbighill @lyin888 和众多专家的指导，成功用上了 SS+obfs 组合！
>
> 我从本项目索取颇多，反哺回报为必须。
>
> 总结路线图如下，供其他有需要的朋友参考：
>
> 环境：
> R9.12.31
> Netgear 4300V1
>
> 步骤：
> 1）进入 Openwrt web页面：系统 => 软件包
>
> 2）【下载并安装软件包】的框里输入： [...]

感谢老铁们，真的有用！