coolsnowwolf/lede #2866: 编译完成了没有生成img文件咋办？

JackEHolmes, 2020-01-26:
[*] squashfs
Target Profile (Raspberry Pi 4B)

make[3]: Leaving directory `/home/jack/lede/feeds/packages/lang/golang/golang'
make[2]: Leaving directory `/home/jack/lede'
make[1]: Leaving directory `/home/jack/lede'

这是编译完成了吗？
怎么没有看到可以刷入树莓派的img文件
/lede/bin/targets/brcm2708/bcm2711# ls
config.seed  packages

lyin888, 2020-01-26:
固件过大了吧

AmadeusGhost, 2020-01-27:
[image]
你试一下看看

JackEHolmes, 2020-01-27:
> 固件过大了吧

那要怎么设置呢？

JackEHolmes, 2020-01-27:
> [image]
> 你试一下看看

试过了，还是没有生成img文件

KnowledegAndAction, 2020-01-28:
错误信息没贴全吧

pbwind, 2020-02-10:
我也是。 
编译N1的固件的时候，不生成img。
但是生成tar.gz ，但是这个文件太小了。 只有10M不到。

hcym, 2020-02-10:
n1就是这样格式，没有v2的话只有6M大小。

closed 2020-02-19 by coolsnowwolf