coolsnowwolf/lede #2878: K2T 内存溢出？

NickLiu635, 2020-01-27:
昨天（1月27日）晚上更新编译的版本，编译前有make clean。K2T，AR7xxx，添加了ATH10k-ct-htt，并去掉了大部分与路由无关的组件（只保留了kms），2.4G设备5台左右，5G设备5台左右
4点左右发现，路由器网页无法打开，已经连接wifi的设备正常使用，但新设备无法通过wifi连接到路由器，旧的设备断开后也无法再次连接到路由器。

相关日志如下

> Sat Jan 25 06:36:05 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f IEEE 802.11: authenticated
Sat Jan 25 06:36:05 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f IEEE 802.11: associated (aid 4)
Sat Jan 25 06:36:05 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED d4:63:c6:0d:56:7f
Sat Jan 25 06:36:05 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f RADIUS: starting accounting session 1348EB8B93BCDBED
Sat Jan 25 06:36:05 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f WPA: pairwise key handshake completed (RSN)
Sat Jan 25 06:41:10 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED d4:63:c6:0d:56:7f
Sat Jan 25 06:41:10 2020 daemon.info hostapd: wlan1: STA d4:63:c6:0d:56:7f IEEE 802.11: disassociated due to inactivity
Sat Jan 25 06:41:11 2020 daemon.info hostapd: wlan1: STA d4:63:c6:0d:56:7f IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Jan 25 06:45:51 2020 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Sat Jan 25 06:45:51 2020 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Sat Jan 25 08:12:08 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f IEEE 802.11: authenticated
Sat Jan 25 08:12:09 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f IEEE 802.11: associated (aid 4)
Sat Jan 25 08:12:10 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f RADIUS: starting accounting session 1348EB8B93BCDBED
Sat Jan 25 08:12:10 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f WPA: pairwise key handshake completed (RSN)
Sat Jan 25 08:42:38 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f IEEE 802.11: authenticated
Sat Jan 25 08:42:39 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f IEEE 802.11: associated (aid 4)
Sat Jan 25 08:42:39 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f RADIUS: starting accounting session 1348EB8B93BCDBED
Sat Jan 25 08:42:39 2020 daemon.info hostapd: wlan0: STA d4:63:c6:0d:56:7f WPA: pairwise key handshake completed (RSN)
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.645672] hostapd invoked oom-killer: gfp_mask=0x14201ca(GFP_HIGHUSER_MOVABLE|__GFP_COLD), nodemask=(null),  order=0, oom_score_adj=0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.659511] CPU: 0 PID: 1623 Comm: hostapd Not tainted 4.14.162 #0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.666698] Stack : 80463800 803a17f4 00000000 00000000 00000000 00000000 00000000 00000000
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.676305]         00000000 00000000 00000000 00000000 00000000 00000001 8296db78 c789acd6
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.685874]         8296dc10 00000000 00000000 00000000 00000038 803772b8 6420342e 31342e31
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.695346]         206e6f64 00000000 000000ff 656d6173 80000000 829a10e8 803b0698 00000004
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.704862]         803f7410 8296dd40 00000001 00200000 00000001 80233e24 00000000 80460000
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.714293]         ...
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.716828] Call Trace:
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.716925] [<803772b8>] 0x803772b8
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.724142] [<80233e24>] 0x80233e24
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.728445] [<8006b6ac>] 0x8006b6ac
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.732056] [<8006b6b4>] 0x8006b6b4
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.736375] [<800e76c0>] 0x800e76c0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.740035] [<800e6a5c>] 0x800e6a5c
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.743658] [<800e6a6c>] 0x800e6a6c
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.748019] [<800f6f08>] 0x800f6f08
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.751722] [<800e75d4>] 0x800e75d4
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.755653] [<800eb18c>] 0x800eb18c
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.759534] [<800e479c>] 0x800e479c
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.763234] [<801069c0>] 0x801069c0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.767125] [<8010a138>] 0x8010a138
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.770811] [<80072448>] 0x80072448
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.775472] [<800a1970>] 0x800a1970
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.779093] [<800a22e4>] 0x800a22e4
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.782697] [<800b513c>] 0x800b513c
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.787360] [<80379890>] 0x80379890
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.790968] [<8028bde0>] 0x8028bde0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.795641] [<8007796c>] 0x8007796c
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.799249]
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.800785] Mem-Info:
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.803140] active_anon:454 inactive_anon:24 isolated_anon:0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.803140]  active_file:36 inactive_file:44 isolated_file:0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.803140]  unevictable:0 dirty:0 writeback:0 unstable:0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.803140]  slab_reclaimable:201 slab_unreclaimable:1722
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.803140]  mapped:2 shmem:112 pagetables:64 bounce:0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.803140]  free:1972 free_pcp:14 free_cma:0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.836663] Node 0 active_anon:1816kB inactive_anon:96kB active_file:144kB inactive_file:176kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:8kB dirty:0kB writeback:0kB shmem:448kB writeback_tmp:0kB unstable:0kB all_unreclaimable? yes
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.860438] Normal free:7824kB min:8192kB low:10240kB high:12288kB active_anon:1816kB inactive_anon:96kB active_file:144kB inactive_file:176kB unevictable:0kB writepending:0kB present:65536kB managed:60496kB mlocked:0kB kernel_stack:304kB pagetables:256kB bounce:0kB free_pcp:56kB local_pcp:56kB free_cma:0kB
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.889601] lowmem_reserve[]: 0 0
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.893039] Normal: 256*4kB (UM) 165*8kB (UM) 118*16kB (UM) 12*32kB (UM) 10*64kB (M) 6*128kB (UM) 3*256kB (UM) 2*512kB (M) 0*1024kB 0*2048kB 0*4096kB = 7816kB
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.908483] 192 total pagecache pages
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.912267] 16384 pages RAM
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.915735] 0 pages HighMem/MovableOnly
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22039.919702] 1260 pages reserved
Sat Jan 25 09:40:15 2020 kern.info kernel: [22039.922946] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
Sat Jan 25 09:40:15 2020 kern.info kernel: [22039.932085] [  513]     0   513      303       17       5       0        0             0 ubusd
[... 237 lines cut ...]
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22043.341702] 116 total pagecache pages
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22043.346341] 16384 pages RAM
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22043.349231] 0 pages HighMem/MovableOnly
Sat Jan 25 09:40:15 2020 kern.warn kernel: [22043.353182] 1260 pages reserved
Sat Jan 25 09:40:15 2020 kern.info kernel: [22043.357545] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
Sat Jan 25 09:40:15 2020 kern.info kernel: [22043.367372] [  513]     0   513      303       17       5       0        0             0 ubusd
Sat Jan 25 09:40:15 2020 kern.info kernel: [22043.377146] [  514]     0   514      231       10       4       0        0             0 askfirst
Sat Jan 25 09:40:15 2020 kern.info kernel: [22043.387137] [  891]     0   891      312       26       4       0        0             0 logd
Sat Jan 25 09:40:15 2020 kern.info kernel: [22043.396379] [  912]     0   912      388       25       6       0        0             0 rpcd
Sat Jan 25 09:40:15 2020 kern.info kernel: [22043.405694] [  977]     0   977      269       10       4       0        0             0 dropbear
[... cut ...]
Sat Jan 25 09:48:32 2020 authpriv.warn dropbear[1889]: Login attempt for nonexistent user from 192.168.1.6:56154
Sat Jan 25 09:48:36 2020 authpriv.warn dropbear[1889]: Login attempt for nonexistent user from 192.168.1.6:56154
Sat Jan 25 09:48:43 2020 authpriv.info dropbear[1889]: Exit before auth: Error reading: Connection reset by peer
Sat Jan 25 09:48:50 2020 authpriv.info dropbear[1890]: Child connection from 192.168.1.6:56183
Sat Jan 25 09:48:55 2020 authpriv.notice dropbear[1890]: Password auth succeeded for 'root' from 192.168.1.6:56183

NickLiu635, 2020-01-29:
添加了getrandom之后，已稳定运行一天=。=我找不到原因。。

coolsnowwolf, 2020-01-31:
是内存太小，不适合 ath10k-ct

sqliuchang, 2020-02-03:
> 是内存太小，不适合 ath10k-ct

想问下lean大，ath10k ct htt这几个驱动有什么区别，Google了一圈没找到答案…

PR #2919 (2020-02-03): ath79: sync and fix wireless
fix #2742 
fix #2878 
fix #2891
closed 2020-02-03 by coolsnowwolf