coolsnowwolf/lede #2889: 最近一版shadowsocksr启动bug及补丁

unifreq, 2020-01-29:
lean大：
最近一版(20200124)的/etc/init.d/shadowsr start时出现：
sh: 0: unknown operand
经检查代码，问题出现在if语句里，有很多地方如 if [ $xxx = 0 ]类似的代码存在bug，建议都改成
if [ "$xxx" == "0" ]
我自己做了个补丁望采纳
`--- shadowsocksr.orig	2020-01-29 20:58:48.533735334 +0800
+++ shadowsocksr	2020-01-29 21:46:01.407520660 +0800
@@ -73,16 +73,16 @@ gen_config_file() {
 				hostip=`cat /etc/ssr_ip`
 			fi
 		fi
-		[ $2 = "0" -a  $kcp_flag = "1" ] && hostip="127.0.0.1"
+		[ "$2" == "0" -a  $kcp_flag == "1" ] && hostip="127.0.0.1"
 
-		if [ $2 = "0" ] ;then
+		if [ "$2" == "0" ] ;then
 			config_file=$CONFIG_FILE
-		elif [ $2 = "1" ]; then
+		elif [ "$2" == "1" ]; then
 			config_file=$CONFIG_UDP_FILE
 		else
 			config_file=$CONFIG_SOCK5_FILE
 		fi
-		if [ $(uci_get_by_name $1 fast_open 0) = "1" ] ;then
+		if [ "$(uci_get_by_name $1 fast_open 0)" == "1" ] ;then
 			fastopen="true";
 		else
 			fastopen="false";
@@ -155,7 +155,7 @@ start_rules() {
 	kcp_server=$server
 
 	local kcp_enable=$(uci_get_by_name $GLOBAL_SERVER kcp_enable 0)
-	if [ $kcp_enable = "1" ] ;then
+	if [ "$kcp_enable" == "1" ] ;then
 		kcp_flag=1
 	fi
 
@@ -163,7 +163,7 @@ start_rules() {
 	local lan_ac_ips=$(uci_get_by_type access_control lan_ac_ips)
 	local lan_ac_mode="b"
 	local router_proxy=$(uci_get_by_type access_control router_proxy)
-	if [ "$GLOBAL_SERVER" = "$UDP_RELAY_SERVER" -a $kcp_flag = 0 ]; then
+	if [ "$GLOBAL_SERVER" == "$UDP_RELAY_SERVER" -a "$kcp_flag" == "0" ]; then
 		ARG_UDP="-u"
 	elif [ -n "$UDP_RELAY_SERVER" ]; then
 		ARG_UDP="-U"
@@ -179,18 +179,18 @@ start_rules() {
 
 #deal	gfw firewall rule
 	local gfwmode=""
-	if [ "$run_mode" = "gfw" ]; then
+	if [ "$run_mode" == "gfw" ]; then
 		gfwmode="-g"
-	elif [ "$run_mode" = "router" ]; then
+	elif [ "$run_mode" == "router" ]; then
 		gfwmode="-r"
-	elif [ "$run_mode" = "oversea" ]; then
+	elif [ "$run_mode" == "oversea" ]; then
 		gfwmode="-c"
-	elif [ "$run_mode" = "all" ]; then
+	elif [ "$run_mode" == "all" ]; then
 		gfwmode="-z"
 	fi
 
 	local dports=$(uci_get_by_type global dports 1)
-	if [ $dports = "1" ] ;then
+	if [ "$dports" == "1" ] ;then
 		proxyport=" "
 	else
 		proxyport="-m multiport --dports 22,53,587,465,995,993,143,80,443 "
@@ -279,11 +279,11 @@ start_redir() {
 
 	#deal kcp
 	local kcp_enable=$(uci_get_by_name $GLOBAL_SERVER kcp_enable 0)
-	if [ $kcp_enable = "1" ] ;then
+	if [ "$kcp_enable" == "1" ] ;then
 		[ ! -f "/usr/bin/kcptun-client" ]  &&  return 1
 
 		local kcp_str=`/usr/bin/kcptun-client -v |grep kcptun|wc -l`
-		[ "0" = $kcp_str ] && return 1
+		[ "0" == "$kcp_str" ] && return 1
 		local kcp_port=$(uci_get_by_name $GLOBAL_SERVER kcp_port)
 		local server_port=$(uci_get_by_name $GLOBAL_SERVER server_port)
 		local password=$(uci_get_by_name $GLOBAL_SERVER kcp_password)
@@ -370,13 +370,13 @@ start_redir() {
 
 	#deal with dns
 
-	if [ "$(uci_get_by_type global pdnsd_enable)" = "1" ] ;then
+	if [ "$(uci_get_by_type global pdnsd_enable)" == "1" ] ;then
 	local dnsstr="$(uci_get_by_type global tunnel_forward 8.8.4.4:53)"
 	local dnsserver=`echo "$dnsstr"|awk -F ':'  '{print $1}'`
 	local dnsport=`echo "$dnsstr"|awk -F ':'  '{print $2}'`
-	if [ "$run_mode" = "gfw" ]; then
+	if [ "$run_mode" == "gfw" ]; then
 		ipset add gfwlist $dnsserver 2>/dev/null
-	elif [ "$run_mode" = "oversea" ]; then
+	elif [ "$run_mode" == "oversea" ]; then
 		ipset add oversea $dnsserver 2>/dev/null
 	else
 		ipset add ss_spec_wan_ac $dnsserver nomatch 2>/dev/null
@@ -385,8 +385,8 @@ start_redir() {
 		pdnsd_enable_flag=1
 	fi
 
-	if [ "$(uci_get_by_type global enable_switch)" = "1" ] ;then
-		if [ "$(uci_get_by_name $GLOBAL_SERVER switch_enable)" = "1" ] ;then
+	if [ "$(uci_get_by_type global enable_switch)" == "1" ] ;then
+		if [ "$(uci_get_by_name $GLOBAL_SERVER switch_enable)" == "1" ] ;then
 			if [ -z "$switch_server" ] ;then
 				local switch_time=$(uci_get_by_type global switch_time)
 				local switch_timeout=$(uci_get_by_type global switch_timeout)
@@ -401,7 +401,7 @@ start_redir() {
 }
 
 gen_service_file() {
-	if [ $(uci_get_by_name $1 fast_open) = "1" ] ;then
+	if [ "$(uci_get_by_name $1 fast_open)" == "1" ] ;then
 		fastopen="true";
 	else
 		fastopen="false";
@@ -423,9 +423,9 @@ EOF
 }
 
 start_service() {
-[ $(uci_get_by_name $1 enable) = "0"  ]  && return 1
+[ "$(uci_get_by_name $1 enable)" == "0"  ]  && return 1
 let server_count=server_count+1
-if [ $server_count = 1 ] ;then
+if [ "$server_count" == "1" ] ;then
 	iptables -N SSR-SERVER-RULE && \
 	iptables -t filter -I INPUT  -j SSR-SERVER-RULE
 fi
@@ -439,7 +439,7 @@ return 0
 gen_serv_include() {
 FWI=$(uci get firewall.shadowsocksr.path 2>/dev/null)
 [ -n "$FWI" ] || return 0
-if [ ! -f $FWI ] ;then
+if [ ! -f "$FWI" ] ;then
 	echo '#!/bin/sh' >$FWI
 fi
 extract_rules() {
@@ -457,7 +457,7 @@ EOF
 }
 start_server() {
 	SERVER_ENABLE=$(uci_get_by_type server_global enable_server)
-	[ "$SERVER_ENABLE" = 0 ] && return 0
+	[ "$SERVER_ENABLE" == "0" ] && return 0
 	mkdir -p /var/run /var/etc
 
 	config_load $NAME
@@ -468,7 +468,7 @@ start_server() {
 
 start_local() {
 	local local_server=$(uci_get_by_type socks5_proxy server)
-	[ "$local_server" = "nil" ] && return 1
+	[ "$local_server" == "nil" ] && return 1
 	mkdir -p /var/run /var/etc
 	gen_config_file $local_server 2
 	/usr/bin/ssr-local -c $CONFIG_SOCK5_FILE -u  \
@@ -479,7 +479,7 @@ start_local() {
 }
 
 rules() {
-	[ "$GLOBAL_SERVER" = "nil" ] && return 1
+	[ "$GLOBAL_SERVER" == "nil" ] && return 1
 	mkdir -p /var/run /var/etc
 	UDP_RELAY_SERVER=$(uci_get_by_type global udp_relay_server)
 	[ "$UDP_RELAY_SERVER" = "same" ] && UDP_RELAY_SERVER=$GLOBAL_SERVER
@@ -511,7 +511,7 @@ EOF
 conf-dir=/tmp/dnsmasq.oversea
 EOF
 	fi
-	if [ $(uci_get_by_type global adblock) = 0 ] ;then
+	if [ "$(uci_get_by_type global adblock)" == "0" ] ;then
 		rm -f /tmp/dnsmasq.ssr/ad.conf
 	fi
 
@@ -523,7 +523,7 @@ EOF
 	start_server
 	start_local
 
-	if [ $(uci_get_by_type global monitor_enable) = 1 ] ;then
+	if [ "$(uci_get_by_type global monitor_enable)" == "1" ] ;then
 		let total_count=server_count+redir_tcp+redir_udp+tunnel_enable+kcp_enable_flag+local_enable+pdnsd_enable_flag+switch_enable
 		if [ $total_count -gt 0 ] ;then
 			#param:server(count) redir_tcp(0:no,1:yes)  redir_udp tunnel kcp local gfw
@@ -532,7 +532,7 @@ EOF
 	fi
 
 	ENABLE_SERVER=$(uci_get_by_type global global_server)
-	[ "$ENABLE_SERVER" = "nil" ] && return 1
+	[ "$ENABLE_SERVER" == "nil" ] && return 1
 }
 
 boot() {
@@ -550,7 +550,7 @@ stop() {
 	if [ -z "$switch_server" ] ;then
 		kill -9 $(busybox ps -w | grep ssr-switch | grep -v grep | awk '{print $1}') >/dev/null 2>&1
 	fi
-	if [ $(uci_get_by_type global monitor_enable) = 1 ] ;then
+	if [ "$(uci_get_by_type global monitor_enable)" == "1" ] ;then
 		kill -9 $(busybox ps -w | grep ssr-monitor | grep -v grep | awk '{print $1}') >/dev/null 2>&1
 	fi
 	killall -q -9 ssr-monitor
`

1715173329, 2020-01-29:
温馨提示：```[ "$a" == "$b" ]``` 这种语法仅为bash特有
在 (a)sh 环境下，```[ "$a" == "$b" ]```和 ```[ "$a" = "$b" ]``` 并无任何差别

1715173329, 2020-01-29:
导致出现 `unknown operand` 的原因在于引入的变量为空值（例如 `$a` 为空），且变量未使用引号引入
那么原表达式 ```[ $a = "$b" ]``` 变为 ```[ = "$b" ]```，由于左边是空的，自然导致报错，但这并不影响整个脚本的正常执行

KFERMercer, 2020-01-29:
↑确实

closed 2020-02-03 by coolsnowwolf