coolsnowwolf/lede #2890: kmod-sched-cake编译出错

kevenshijian, 2020-01-29:
删除lede整个目录，重新编译了2次，都是这个错误，各位大侠帮忙解决一下 谢谢

make[3]: Leaving directory '/home/keven/lede/feeds/packages/net/shadowsocks-libev'
time: package/feeds/packages/shadowsocks-libev/compile#0.09#1.26#1.27
make[3]: Entering directory '/home/keven/lede/package/kernel/kmod-sched-cake'
rm -f /home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/.built
touch /home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/.built_check
cat /dev/null > /home/keven/lede/staging_dir/target-x86_64_musl/pkginfo/sched-cake.symvers; for subdir in .; do cat /home/keven/lede/staging_dir/target-x86_64_musl/pkginfo/*.symvers 2>/dev/null > /home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/$subdir/Module.symvers; done
make -C /home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.93 HOSTCFLAGS="-O2 -I/home/keven/lede/staging_dir/host/include -I/home/keven/lede/staging_dir/hostpkg/include -I/home/keven/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Jan 24 10:22:41 2020" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/keven/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.19.93 CC="x86_64-openwrt-linux-musl-gcc" SUBDIRS="/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388" modules
make[4]: Entering directory '/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.93'
make[5]: *** No rule to make target 'tools/objtool/objtool', needed by '/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/sch_cake.o'.  Stop.
Makefile:1520: recipe for target '_module_/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388' failed
make[4]: *** [_module_/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388] Error 2
make[4]: Leaving directory '/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.93'
Makefile:42: recipe for target '/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/.built' failed
make[3]: *** [/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/.built] Error 2
make[3]: Leaving directory '/home/keven/lede/package/kernel/kmod-sched-cake'
time: package/kernel/kmod-sched-cake/compile#0.60#8.85#9.69
package/Makefile:107: recipe for target 'package/kernel/kmod-sched-cake/compile' failed
make[2]: *** [package/kernel/kmod-sched-cake/compile] Error 2
make[2]: Leaving directory '/home/keven/lede'
package/Makefile:103: recipe for target '/home/keven/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/keven/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/keven/lede'
/home/keven/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

kevenshijian, 2020-01-29:
补充一下，新拉的代码，ubuntu18.04，全程扶墙。

rgb5099888, 2020-01-30:
我也是差不多的问题，系统Ubuntu  14 LTS x64
全程富强

collect2: error: ld returned 1 exit status
make[5]: *** [cc1] Error 1
make[5]: Leaving directory `/home/xxx/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial/gcc'
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory `/home/xxx/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial'
make[3]: *** [/home/xxx/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-initial/.built] Error 2
make[3]: Leaving directory `/home/xxx/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#1214.06#81.41#1562.64
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory `/home/xxx/lede'
make[1]: *** [/home/xxx/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory `/home/xxx/lede'
make: *** [world] 错误 2

kevenshijian, 2020-01-30:
再补充一下 x64  全部默认 没有增减任何配置

hcym, 2020-01-30:
应该没问题，早上编译过，x86／64不是默认设置，插件只有翻墙和kms，其它usb网卡无线网卡什么的都去除。还有一个n1也是没有无线和usb光秃秃的，只有一个r7800是默认配置

hcym, 2020-01-30:
系统也是1804，本地编译。最近官方编译多，也没问题

kevenshijian, 2020-01-30:
奇怪，难道需要重新装了1804啊 心痛

dongfangy2011, 2020-01-31:
你看看是不是需要升级UPX？

hcym, 2020-01-31:
压缩默认关闭的，x86更不需要，我的系统也没安装。梯子出错的多，其次就是系统依赖环境。还有就是插件多的话互相有可能冲突，这个看日志吧，懂码的比较容易，我是看不懂

9l, 2020-01-31:
你单线程编译试试

coolsnowwolf, 2020-01-31:
No rule to make target 'tools/objtool/objtool', needed by '/home/keven/lede/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/sch_cake.o'. Stop.

少了 libelf-dev

closed 2020-02-03 by coolsnowwolf