coolsnowwolf/lede #2899: 为何最新版V2Ray只能使用TCP模式，不能使用mKcp+WechatVideo模式？

pirately, 2020-01-31:
请问，为何最新版V2Ray只能使用TCP模式，不能使用mKcp+WechatVideo模式？
V2Ray版本为4.22.1-6，固件为 R20.1.24，内核为4.19.98，x86
表现为，用TCP模式，连接测试显示OK，并且能出国留学。但是用mKcp模式，则连接测试显示fail，不过能ping延迟照样显示，但是不能出国留学。
请问这是为何？
电脑本身用V2RayN，手机用其他代理软件都是mKcp没有问题的。
软路由都是格盘全新装的最新编译的固件。

EstFoxs, 2020-01-31:
mkcp正常使用
[image]

pirately, 2020-01-31:
> mkcp正常使用，不过个人感觉1.24版mkcp的速度慢了不少，经过这几天的使用，慢了将近2/3

[image]
[image]
[image]
[image]

mKcp和TCP截图。我是换了运营商也一样。难不成和系统有关系？您在VPS上用的是什么系统？我用的是Debian

EstFoxs, 2020-01-31:
> > mkcp正常使用，不过个人感觉1.24版mkcp的速度慢了不少，经过这几天的使用，慢了将近2/3
> > [image]
> 
> 这是我的截图：
> [image]
> [image]
> [image]
> [image]
> 
> mKcp和TCP截图。我是换了运营商也一样。难不成和系统有关系？您在VPS上用的是什么系统？我用的是Debian

debian系统也有，centos也有
都正常

pirately, 2020-01-31:
> 
> debian系统也有，centos也有
> 都正常

那您用的是哪个vps服务商？

coolsnowwolf, 2020-01-31:
测试可用

TheAutomatic, 2020-02-07:
我和楼主一样，tcp完全没问题；手机v2rayNG客户端跑mKCP也没问题，K3路由器lede跑mKCP不行

hcym, 2020-02-07:
kcp可以，用的n1和x86，k3和r7800有点吃力，效果要看线路的，用瓦工还是tcp好，用vurlt和cc什么kcp好一些

TheAutomatic, 2020-02-07:
> kcp可以，用的n1和x86，k3和r7800有点吃力，效果要看线路的，用瓦工还是tcp好，用vurlt和cc什么kcp好一些

我用的是Azure香港服务器，想弄ws+tls来着，域名ssl证书死活不给签发

closed 2020-02-19 by coolsnowwolf

verOn1204, 2020-08-08:
> > debian系统也有，centos也有
> > 都正常
> 
> 那您用的是哪个vps服务商？

请问有解决么？我也碰到这个问题了