coolsnowwolf/lede #2901: 请大神添加对NanoPi R1S-H5的支持。

ppp47, 2020-01-31:
刚入手了一款NanoPi_R1S-H5的路由器在编译固件的时候没有找到对应的支持，请大神有空添加一下支持方便小白编译固件，谢谢！！
http://wiki.friendlyarm.com/wiki/index.php/NanoPi_R1S-H5/zh

kl9800, 2020-01-31:
同求，那玩意体积超级小，省了大地方，多谢多谢，R2S要出了！！！

hcym, 2020-01-31:
也在等2s，单臂应该armv8就够用

mcucore, 2020-01-31:
刚找到R2S资料，RK3328，双千兆（TX,RX实测941M）,代替N1或软路由不错哈！
http://wiki.friendlyarm.com/wiki/index.php/NanoPi_R2S/zh

lqhorochi, 2020-01-31:
还有一个NanoPi R2S,暂时还没上市
修正R1S-H5的问题

klever1988, 2020-02-01:
[0001-add-r1s-h5-support.zip](https://github.com/coolsnowwolf/lede/files/4142357/0001-add-r1s-h5-support.zip)

解压出patch文件，把patch放到lede目录，在make menuconfig之前用
git apply 0001-add-r1s-h5-support.patch
应用代码补丁，就可以在make menuconfig找到r1s设备并编译了。

已知问题，找不到无线模块，无线模块无法驱动
其中，内核dts文件来自https://github.com/jerrykuku/openwrt-nanopi-r1s-h5，感谢老竭力。

poctopus, 2020-02-03:
https://github.com/skytotwo/NanoPi-R1S-Build-By-Actions/suites/366205371/artifacts/762509
已经有人编译出来了。

作者的网站
https://totoro.site/index.php/archives/70/

ppp47, 2020-02-06:
> [0001-add-r1s-h5-support.zip](https://github.com/coolsnowwolf/lede/files/4142357/0001-add-r1s-h5-support.zip)
> 
> 解压出补丁文件，把补丁放到文件目录中，在make menuconfig之前使用
> git apply 0001-add-r1s-h5-support.patch
> 应用代码补丁，就可以在make menuconfig找到r1s设备并编译了。
> 
> 已知问题，找不到无线模块，无线模块无法驱动
> 其中，内核dts文件来自https://github.com/jerrykuku/openwrt-nanopi-r1s-h5，感谢老竭力。

ppp47, 2020-02-06:
这两个方法我都试过了没有用啊,可能是我的问题
第二个的连接失效了无法下载固件
[image]

anyUesr, 2020-02-12:
我按着这位作者方式编译了一下 r2s 的固件，编译失败，看起来还是需要一些其他的修改

klever1988, 2020-02-12:
> 我按着这位作者方式编译了一下 r2s 的固件，编译失败，看起来还是需要一些其他的修改

https://github.com/klever1988/NanoPi-R1S/blob/master/.github/workflows/r2s_lean.yml

不过目前还没有设备可以测试

xxyxd, 2020-02-13:
> > 我按着这位作者方式编译了一下 r2s 的固件，编译失败，看起来还是需要一些其他的修改
> 
> https://github.com/klever1988/NanoPi-R1S/blob/master/.github/workflows/r2s_lean.yml
> 
> 不过目前还没有设备可以测试

希望到货后可以测试一下 ,  R2S确实太吸引人了 , 感谢<)Like

closed 2020-02-19 by coolsnowwolf

zhhuabj, 2020-03-04:
see https://github.com/tiantianxiaoxiaoxyz/AutoBuild-Nanopi-R2S/blob/master/.github/workflows/NanoPi-R2s%20RK3328%20OpenWrt%20Build.yml