coolsnowwolf/lede #2910: 编译失败

Yofo-me, 2020-02-01:
```
Makefile:77: recipe for target '/home/code/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-4.25/.configured_1eac54b570d78e3d5519700ed2c720b1' failed
make[3]: *** [/home/code/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-4.25/.configured_1eac54b570d78e3d5519700ed2c720b1] Error 1
make[3]: Leaving directory '/home/code/lede/package/devel/strace'
time: package/devel/strace/compile#14.28#7.42#19.40
package/Makefile:107: recipe for target 'package/devel/strace/compile' failed
make[2]: *** [package/devel/strace/compile] Error 2
make[2]: Leaving directory '/home/code/lede'
package/Makefile:103: recipe for target '/home/code/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/code/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/code/lede'
/home/code/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

```

> 怎么解决？

librec, 2020-02-01:
( cd /home/librec/lede/build_dir/hostpkg/go-1.13.6/src ; GOROOT_FINAL=/home/librec/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/librec/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/librec/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap.
Building Go toolchain1 using /home/librec/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap.
Building Go bootstrap cmd/go (go_bootstrap) using Go toolchain1.
Building Go toolchain2 using go_bootstrap and Go toolchain1.
Building Go toolchain3 using go_bootstrap and Go toolchain2.
go build cmd/compile/internal/ssa: /home/librec/lede/build_dir/hostpkg/go-1.13.6/pkg/tool/linux_amd64/compile: signal: killed
go tool dist: FAILED: /home/librec/lede/build_dir/hostpkg/go-1.13.6/pkg/tool/linux_amd64/go_bootstrap install -gcflags=all= -ldflags=all= -a -i cmd/asm cmd/cgo cmd/compile cmd/link: exit status 1
make[3]: *** [/home/librec/lede/build_dir/hostpkg/go-1.13.6/.built] Error 2
make[3]: Leaving directory `/home/librec/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#307.31#49.32#996.58
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory `/home/librec/lede'
make[1]: *** [/home/librec/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/librec/lede'
make: *** [world] Error 2
我们好像差不多，搞了一天了，有大佬帮忙看一下吗

peterwjp2017, 2020-02-02:
> ( cd /home/librec/lede/build_dir/hostpkg/go-1.13.6/src ; GOROOT_FINAL=/home/librec/lede/staging_dir/hostpkg/lib/go-cross GOROOT_BOOTSTRAP=/home/librec/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap [...]

全局搭梯子了么？看着像是下载go组件失败

librec, 2020-02-02:
> 全局搭梯子了么？看着像是下载go组件失败

在vps上编译的，之前都没问题，更新源代码之后就这样了。[手动狗头]

coolsnowwolf, 2020-02-02:
内存太小了，不足以编译 go

librec, 2020-02-02:
> 内存太小了，不足以编译 go

谢谢大雕！

closed 2020-02-19 by coolsnowwolf

baokeding, 2020-06-08:
我的是wsl 下的ubuntu。 32g的内存。 怎么会小呢