coolsnowwolf/lede #2911: 可以用nginx代替uhttpd吗

helllkz, 2020-02-01:
官方的好像支持的，L大的能实现吗，我想把PHP的环境部署到OP上面来，对uhttpd不熟，可以编译的时候把nginx弄上然后去掉uhttpd吗

4593645, 2020-02-02:
愿主祝福您，自己弄可以么？这种问题跟我提的问题一样，密码错误。试试password怕受伤

coolsnowwolf, 2020-02-02:
ngnix体积太大了，放不下 小 rom 种

helllkz, 2020-02-02:
> ngnix体积太大了，放不下 小 rom 种

X86的应该是没问题的，所以我在想能不能实现，毕竟现在包里面都有nginx的，但是uhttpd都用了80端口了，而且luci应该是很依赖的，我看好像还需要fcgi什么的

4593645, 2020-02-02:
去配置端口，具体哪里我忘记了。老了！

wgedu, 2020-02-03:
安装nginx-luci，或者编译时候选中，就可以了

closed 2020-02-19 by coolsnowwolf

yubanmeiqin9048, 2021-01-03:
> 安装nginx-luci，或者编译时候选中，就可以了

怎么在编译的时候去掉UHTTPD

coLATin, 2021-01-08:
同问，x86软路由不在乎磁盘空间，只想nginx替换掉uhttpd，方便做转发

yubanmeiqin9048, 2021-01-08:
> 同问，x86软路由不在乎磁盘空间，只想nginx替换掉uhttpd，方便做转发

我现在的解决方法是：uhttpd改监听端口，docker装nginx 然后反代uhttpd顺带配置其他服务，这样方便些

coLATin, 2021-01-13:
> > 同问，x86软路由不在乎磁盘空间，只想nginx替换掉uhttpd，方便做转发
> 
> 我现在的解决方法是：uhttpd改监听端口，docker装nginx 然后反代uhttpd顺带配置其他服务，这样方便些

也是，passwall，ssr还有helloworld没有一个在nginx下正常工作的，无奈