coolsnowwolf/lede #2913: 下载地址错误

kl9800, 2020-02-01:
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/antfs-2018-08-14-002407a1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/antfs-2018-08-14-002407a1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:05 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.

hcym, 2020-02-01:
浏览器下载下来放进去dl目录，源码下载非常考验梯子质量，正好这两天赶上gfw鸡飞狗跳

kl9800, 2020-02-01:
这是404，不是连不上，梯子没问题，是404，访问上层目录是可以的

9l, 2020-02-02:
这目录里没有这个文件

kl9800, 2020-02-02:
是的

leopardciaw, 2020-02-04:
全部都下载不了。

thelittlefox, 2020-02-05:
有兄弟有这个文件的备份吗？

leopardciaw, 2020-02-06:
> 有兄弟有这个文件的备份吗？

把原来下载的dl目录另存保留，删除整个目录，重来git来过一次，可能就好了。我的这样弄好了。

closed 2020-02-19 by coolsnowwolf