coolsnowwolf/lede #2916: 局域网无法用域名进入nas，外网可以。

authon, 2020-02-01:
转发都设置好了，PVE的可以，只有nas不行，听说是源码支持docker导致，请问大神这种情况怎么解决

lariboo, 2020-02-02:
与docker无关，不是它的锅。

hxlls, 2020-02-02:
编译时候不勾选docker 就解决了，是docker 的依赖引起的，具体是啥引起，因为个人对docker 没需求，所以没去查找

authon, 2020-02-02:
> 编译时候不勾选docker 就解决了，是docker 的依赖引起的，具体是啥引起，因为个人对docker 没需求，所以没去查找

谢谢，我试试

helllkz, 2020-02-03:
我还是不行啊，我没装docker，内网用外网域名访问也不能触发端口转发的。
不过也很奇怪，也不是每条端口转发都不能，和楼主一样PVE的8006端口可以触发，R7000的80端口可以触发，但是群晖的5000端口就不能触发，很奇怪，另外个虚拟机的nginx的80端口也无法触发

authon, 2020-02-03:
> 我还是不行啊，我没装docker，内网用外网域名访问也不能触发端口转发的。
> 不过也很奇怪，也不是每条端口转发都不能，和楼主一样PVE的8006端口可以触发，R7000的80端口可以触发，但是群晖的5000端口就不能触发，很奇怪，另外个虚拟机的nginx的80端口也无法触发

编译的时候不要选择安装docker，我把docker和dockerCE都取消了

hxlls, 2020-02-03:
不是取消docker ,是rm 掉配置文件，从新不选择docker 就完事了

authon, 2020-02-03:
> 不是取消docker ,是rm 掉配置文件，从新不选择docker 就完事了
是的，我说的取消就是注释掉

trackycn, 2020-02-05:
> 我还是不行啊，我没装docker，内网用外网域名访问也不能触发端口转发的。
> 不过也很奇怪，也不是每条端口转发都不能，和楼主一样PVE的8006端口可以触发，R7000的80端口可以触发，但是群晖的5000端口就不能触发，很奇怪，另外个虚拟机的nginx的80端口也无法触发

我测试也发现这个问题, 一开始编译的有docker的版本, 转发用不了, 后来编译个没有 docker 的版本, 发现还是不能用, 很奇怪, 抓包看了一下, 发现奇怪的问题:

如果外网端口和内网端口一致可以转发, 如果外网和内网端口不一致, 只能通过一次请求, 之后的请求就不行。

最后关闭 Flow Offloading 后, 问题解决了, 现在不确定是不是其他插件引起的, 还在测试中。

slitilanber, 2020-02-05:
NAT loopback相关问题，但我也不知怎么搞。
说来奇怪，我是lxc中运行的，运行了一段时间完全正常，后面不知咋的又不行了。

helllkz, 2020-02-06:
> 我测试也发现这个问题, 一开始编译的有docker的版本, 转发用不了, 后来编译个没有 docker 的版本, 发现还是不能用, 很奇怪, 抓包看了一下, 发现奇怪的问题:
>
> 如果外网端口和内网端口一致可以转发, 如果外网和内网端口不一致, 只能通过一次请求, 之后的请求就不行。
>
> 最后关闭 Flow Offloading 后, 问题解决了, 现在不确定是不是其他插件引起的, 还在测试中。

还真是这个问题，我关闭Flow Offloading后也没问题了，也是内外端口不一致，其实就算一致的端口之前也不行的

closed 2020-02-19 by coolsnowwolf