coolsnowwolf/lede #2917: 19.07.0编译SSR-plus报错

paladin4fan, 2020-02-02:
编译输出如下 好像是kmod-ipt-ipopt的问题？
Packaged contents of /home/agent/sdk-19.07.0/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/packages/ipkg-arm_cortex-a9_vfpv3/kmod-ipt-offload into /home/agent/sdk-19.07.0/bin/targets/mvebu/cortexa9/packages/kmod-ipt-offload_4.14.162-1_arm_cortex-a9_vfpv3.ipk
ERROR: module '/home/agent/sdk-19.07.0/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/linux-4.14.162/net/netfilter/xt_dscp.ko' is missing.
modules/netfilter.mk:281: recipe for target '/home/agent/sdk-19.07.0/bin/targets/mvebu/cortexa9/packages/kmod-ipt-ipopt_4.14.162-1_arm_cortex-a9_vfpv3.ipk' failed
make[2]: *** [/home/agent/sdk-19.07.0/bin/targets/mvebu/cortexa9/packages/kmod-ipt-ipopt_4.14.162-1_arm_cortex-a9_vfpv3.ipk] Error 1
make[2]: Leaving directory '/home/agent/sdk-19.07.0/package/linux'
time: package/linux/compile#26.53#420.56#499.64
package/Makefile:111: recipe for target 'package/linux/compile' failed
make[1]: *** [package/linux/compile] Error 2
make[1]: Leaving directory '/home/agent/sdk-19.07.0'
/home/agent/sdk-19.07.0/include/toplevel.mk:218: recipe for target 'package/lean/luci-app-ssr-plus/compile' failed
make: *** [package/lean/luci-app-ssr-plus/compile] Error 2

4860575, 2020-02-02:
你是用sdk编译的？sdk里没有依赖包

paladin4fan, 2020-02-02:
> 你是用sdk编译的？sdk里没有依赖包

用SDK+这个项目里的packages 刚刚解决了 是SDK的问题 我用19.07.1的SDK编译成功了

closed 2020-02-02 by paladin4fan