coolsnowwolf/lede #2918: 有个BUG，刷新软件源后提示和图标不消失

lloyd18, 2020-02-02:
[image]

hyird, 2020-02-02:
这。。。。。。。。。。。。。。。。

hxlls, 2020-02-02:
最新源码编译测试没问题，请更新，还有，图标这玩意，好像这套源码没有啊，你确认没走错门？
附上截图：
[image]

lloyd18, 2020-02-02:
> 最新源码编译测试没问题，请更新，还有，图标这玩意，好像这套源码没有啊，你确认没走错门？
> 附上截图：
> [image]
> [image]

可能是我说得不够清楚，可以刷新列表，但是刷新后这个提示是不是该消失了，但它一直在啊，请看标注的地方

hxlls, 2020-02-02:
这个重来不会消失，为什么你认为他需要消失？

lloyd18, 2020-02-02:
> 这个重来不会消失，为什么你认为他需要消失？

[image]

hxlls, 2020-02-02:
测试了，软件源问题，之前没认真看，自己修改下源配置就行了

lloyd18, 2020-02-02:
> 测试了，软件源问题，之前没认真看，自己修改下源配置就行了

请问怎么改，不懂

hxlls, 2020-02-02:
配置，把option check_signature 注释掉或者删除，这个是软件签名

hxlls, 2020-02-02:
[image]
你也可以把这个玩意去掉重新编译，大概位置是
[image]

lloyd18, 2020-02-02:
> [image]
> 你也可以把这个玩意去掉重新编译，大概位置是
> [image]

非常感谢，方法有用

closed 2020-02-03 by coolsnowwolf