coolsnowwolf/lede #2922: socks5 proxy not working in SSR Plus+/sock5代理无效

KnowledegAndAction, 2020-02-02:
除了sock5代理，其他功能一切正常。可以科学上网，仅仅sock5不起作用。
第一次编译的固件在我手动开旗ss-local进程后socks5是可以用的，但是因为缺少了部分功能，我备份设置后重新编译了新的固件，然后刷入，然后恢复旧固件的配置。新加入的功能应该增加了ss相关的一些组件（但没有减少），没有动过ssr相关的。

现在查看进程可以看到
/usr/bin/ssr-local -c /var/etc/shadowsocksr_s.json -u -l 1080 -b 0.0.0.0 -f /var/run/ssr-local.pid
且检查了json文件，没有问题。
但是我的ss server node是来自搬瓦工的justmysocks，ss协议的，不是ssr的，不知道是否和这个有关。我也尝试了手动启动ss-local，但是每次都会自动拉起ssr-local进程。或者只要浏览器通过1080端口去访问sock5代理，ssr-local都会被自动拉起，即使我手动kill后也是。

我希望有socks5代理的目的是因为通过浏览器决定是否走代理比较灵活，而且插件按钮点击比较方便。

需要提供哪些日志？
---------------------------------------------------------------------------------
Everything works fine except this. I can get over GFW under "GFW list mode" in SSR Plus+, just sock5 proxy not working.

 Actually it  once worked, with the first firmware(A) i had compiled 10 days ago. But I then compiled again and got the second firmware(B), because I wanted add some functions. I flashed this firmware B and  restored the backup file from the firmware(A), and sock5 proxy failed.

I can see the process ssr-local running with "ps":
/usr/bin/ssr-local -c /var/etc/shadowsocksr_s.json -u -l 1080 -b 0.0.0.0 -f /var/run/ssr-local.pid
SS server nodes are fine, config in json file is fine, port is right.
BTW, SS server nodes are from Justmysocks, which are SS server, not SSR, maybe that's why?

But I've also tried to run ss-local after killing ssr-local process, which doesn't work either. And I also notice that as soon as ss-local launching, or connecting lede's socks5 proxy port 1080 from my browser,  ssr-local process will be pulled up.
-----------------------------------------------------------------------------

coolsnowwolf, 2020-02-03:
只对 ssr 节点有效

KnowledegAndAction, 2020-02-03:
> 只对 ssr 节点有效

那我应该如何禁用ssr-local的自动拉起进程呢？现在我手动启动ss-local后ssr-local仍然会被自动拉起，即使我设置了不同的代理本地端口，ss-local的socks5仍然无效。
另外我把ss-libev也编译了，在ss-libev里设置好也不行。

KnowledegAndAction, 2020-02-03:
想起来了，在ssr plus+里disable sock5代理后再手动启动ss-local就可以了。

KnowledegAndAction, 2020-02-03:
遇到新问题，再手动启动ss-local Mon Feb  3 18:55:15 2020 daemon.info /usr/bin/ss-local[17333]: using tcp fast open
Mon Feb  3 18:55:15 2020 daemon.info /usr/bin/ss-local[17333]: initializing ciphers... aes-256-gcm
Mon Feb  3 18:55:15 2020 daemon.info /usr/bin/ss-local[17333]: listening at 0.0.0.0:1081
Mon Feb  3 18:55:15 2020 daemon.info /usr/bin/ss-local[17333]: tcp port reuse enabled
Mon Feb  3 18:55:15 2020 daemon.info /usr/bin/ss-local[17333]: udprelay enabled
Mon Feb  3 18:55:15 2020 daemon.info /usr/bin/ss-local[17333]: udp port reuse enabled
Mon Feb  3 18:55:15 2020 daemon.info /usr/bin/ss-local[17333]: running from root user
Mon Feb  3 19:07:54 2020 daemon.err /usr/bin/ss-local[17333]: remote_recv_cb_recv: Connection reset by peer
Mon Feb  3 19:14:44 2020 daemon.err /usr/bin/ss-local[17333]: remote_recv_cb_recv: Connection reset by peer
Mon Feb  3 19:20:07 2020 daemon.err /usr/bin/ss-redir[13691]: server recv: Connection reset by peer
Mon Feb  3 19:20:07 2020 daemon.err /usr/bin/ss-redir[13691]: server recv: Connection reset by peer
Mon Feb  3 19:26:17 2020 daemon.err /usr/bin/ss-redir[13695]: server recv: Connection reset by peer
Mon Feb  3 19:32:37 2020 daemon.err /usr/bin/ss-redir[13691]: server recv: Connection reset by peer
Mon Feb  3 19:32:37 2020 daemon.err /usr/bin/ss-redir[13695]: server recv: Connection reset by peer
Mon Feb  3 19:32:37 2020 daemon.err /usr/bin/ss-redir[13691]: server recv: Connection reset by peer
Mon Feb  3 19:34:00 2020 daemon.err /usr/bin/ss-local[17333]: getpeername: Socket not connected
Mon Feb  3 19:34:00 2020 daemon.err /usr/bin/ss-local[17333]: getpeername: Socket not connected
Mon Feb  3 19:34:09 2020 daemon.err /usr/bin/ss-local[17333]: getpeername: Socket not connected
Mon Feb  3 19:34:18 2020 daemon.err /usr/bin/ss-local[17333]: getpeername: Socket not connected
Mon Feb  3 19:34:39 2020 daemon.err /usr/bin/ss-local[17333]: getpeername: Socket not connected
Mon Feb  3 19:35:09 2020 daemon.err /usr/bin/ss-local[17333]: getpeername: Socket not connected
Mon Feb  3 19:35:22 2020 daemon.err /usr/bin/ss-local[17333]: getpeername: Socket not connected
Mon Feb  3 19:35:27 2020 daemon.err /usr/bin/ss-local[17333]: getpeername: Socket not connected

coolsnowwolf, 2020-02-03:
没必要纠结这个，有独立的 v2ray server ，里面就有 socks5 代理

KnowledegAndAction, 2020-02-04:
> 没必要纠结这个，有独立的 v2ray server ，里面就有 socks5 代理

请教在哪里设置？

KnowledegAndAction, 2020-02-05:
v2ray server里socks代理似乎强制用户名密码，chrome插件switchyomega不支持。

awter, 2020-02-06:
> v2ray server里socks代理似乎强制用户名密码，chrome插件switchyomega不支持。

#issuecomment-582475642

KnowledegAndAction, 2020-02-07:
> > v2ray server里socks代理似乎强制用户名密码，chrome插件switchyomega不支持。
> 
> [#2792 (comment)](https://github.com/coolsnowwolf/lede/issues/2792#issuecomment-582475642)

谢谢，虽然关于为何ss-local会自动失效的问题没有解决，但你这个方法有效，已经解决了我的需求。

closed 2020-02-07 by KnowledegAndAction