coolsnowwolf/lede #2924: fast.com 测试上传时会导致整个断网一会

blankhang, 2020-02-02:
软路由更新到2.1号 OpenWrt R20.1.24 / LuCI Master (git-20.032.31649-0d31472) 编译版本
发现有间歇性断网的问题
仅启用插件 SSR 
在19年10月17号编译的旧版本 无此问题
希望大神有空能看看这个问题
[image]

### 内核日志
[    0.000000] Linux version 4.19.98 (blank@blank) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r1900-ae48c010)) #0 SMP Fri Jan 31 13:48:59 2020
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=5d42a6a1-02 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] [Firmware Warn]: BIOS disabled fast string operation, re-enable failed.
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ec00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fedffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003fee0000-0x000000003fefefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003feff000-0x000000003fefffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003ff00000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 07/29/2019
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 1599.953 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000001] vmware: using sched offset of 4269933617 ns
[    0.000004] tsc: Detected 1599.953 MHz processor
[    0.008292] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.008299] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.008310] last_pfn = 0x40000 max_arch_pfn = 0x400000000
[    0.008365] MTRR default type: uncachable
[    0.008366] MTRR fixed ranges enabled:
[    0.008369]   00000-9FFFF write-back
[    0.008371]   A0000-BFFFF uncachable
[    0.008372]   C0000-CBFFF write-protect
[    0.008374]   CC000-EFFFF uncachable
[    0.008375]   F0000-FFFFF write-protect
[    0.008376] MTRR variable ranges enabled:
[    0.008379]   0 base 00000000000 mask 7FF00000000 write-back
[    0.008381]   1 base 000C0000000 mask 7FFC0000000 uncachable
[    0.008382]   2 disabled
[    0.008383]   3 disabled
[    0.008384]   4 disabled
[    0.008385]   5 disabled
[    0.008386]   6 disabled
[    0.008387]   7 disabled
[    0.008459] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.027342] found SMP MP-table at [mem 0x000f6a70-0x000f6a7f]
[    0.027480] BRK [0x02202000, 0x02202fff] PGTABLE
[    0.027487] BRK [0x02203000, 0x02203fff] PGTABLE
[    0.027489] BRK [0x02204000, 0x02204fff] PGTABLE
[    0.027944] BRK [0x02205000, 0x02205fff] PGTABLE
[    0.028158] ACPI: Early table checksum verification disabled
[    0.028284] ACPI: RSDP 0x00000000000F6A00 000024 (v02 PTLTD )
[    0.028299] ACPI: XSDT 0x000000003FEE965B 00005C (v01 INTEL  440BX    06040000 VMW  01324272)
[    0.028318] ACPI: FACP 0x000000003FEFEE73 0000F4 (v04 INTEL  440BX    06040000 PTL  000F4240)
[    0.028336] ACPI: DSDT 0x000000003FEEAA08 01446B (v01 PTLTD  Custom   06040000 MSFT 03000001)
[    0.028351] ACPI: FACS 0x000000003FEFFFC0 000040
[    0.028360] ACPI: FACS 0x000000003FEFFFC0 000040
[    0.028368] ACPI: BOOT 0x000000003FEEA9E0 000028 (v01 PTLTD  $SBFTBL$ 06040000  LTP 00000001)
[    0.028378] ACPI: APIC 0x000000003FEEA29E 000742 (v01 PTLTD  ? APIC   06040000  LTP 00000000)
[    0.028387] ACPI: MCFG 0x000000003FEEA262 00003C (v01 PTLTD  $PCITBL$ 06040000  LTP 00000001)
[    0.028397] ACPI: SRAT 0x000000003FEE9757 0008A8 (v02 VMWARE MEMPLUG  06040000 VMW  00000001)
[    0.028406] ACPI: HPET 0x000000003FEE971F 000038 (v01 VMWARE VMW HPET 06040000 VMW  00000001)
[    0.028415] ACPI: WAET 0x000000003FEE96F7 000028 (v01 VMWARE VMW WAET 06040000 VMW  00000001)
[... 1854 lines cut ...]
Sun Feb  2 02:05:14 2020 daemon.info pdnsd[24514]: pdnsd-1.2.9b-par starting.
Sun Feb  2 02:05:14 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Feb  2 02:05:14 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Feb  2 04:00:00 2020 cron.info crond[24299]: USER root pid 22174 cmd /etc/coremark.sh
Sun Feb  2 14:01:31 2020 daemon.notice netifd: wan (17032): udhcpc: sending renew to 192.168.1.1
Sun Feb  2 14:01:31 2020 daemon.notice netifd: wan (17032): udhcpc: lease of 192.168.1.8 obtained, lease time 86400
Sun Feb  2 16:47:02 2020 daemon.notice netifd: Interface 'lan' is now down
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.524660] br-lan: port 1(eth0) entered disabled state
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.530996] device eth0 left promiscuous mode
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.534482] br-lan: port 1(eth0) entered disabled state
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.711777] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Sun Feb  2 16:47:02 2020 daemon.notice netifd: Interface 'lan' is disabled
Sun Feb  2 16:47:02 2020 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Feb  2 16:47:02 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.819448] 8021q: adding VLAN 0 to HW filter on device eth0
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.822358] br-lan: port 1(eth0) entered blocking state
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.824371] br-lan: port 1(eth0) entered disabled state
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.828857] device eth0 entered promiscuous mode
Sun Feb  2 16:47:02 2020 daemon.notice netifd: Interface 'lan' is enabled
Sun Feb  2 16:47:02 2020 daemon.notice netifd: Interface 'lan' is setting up now
Sun Feb  2 16:47:02 2020 daemon.notice netifd: Interface 'lan' is now up
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.831665] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.834129] br-lan: port 1(eth0) entered blocking state
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.836237] br-lan: port 1(eth0) entered forwarding state
Sun Feb  2 16:47:02 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)

blankhang, 2020-02-02:
### 内核日志
[    0.000000] Linux version 4.19.98 (blank@blank) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r1900-ae48c010)) #0 SMP Fri Jan 31 13:48:59 2020
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=5d42a6a1-02 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] [Firmware Warn]: BIOS disabled fast string operation, re-enable failed.
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ec00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fedffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003fee0000-0x000000003fefefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003feff000-0x000000003fefffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003ff00000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 07/29/2019
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 1599.953 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000001] vmware: using sched offset of 4269933617 ns
[    0.000004] tsc: Detected 1599.953 MHz processor
[    0.008292] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.008299] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.008310] last_pfn = 0x40000 max_arch_pfn = 0x400000000
[    0.008365] MTRR default type: uncachable
[    0.008366] MTRR fixed ranges enabled:
[    0.008369]   00000-9FFFF write-back
[    0.008371]   A0000-BFFFF uncachable
[    0.008372]   C0000-CBFFF write-protect
[    0.008374]   CC000-EFFFF uncachable
[    0.008375]   F0000-FFFFF write-protect
[    0.008376] MTRR variable ranges enabled:
[    0.008379]   0 base 00000000000 mask 7FF00000000 write-back
[    0.008381]   1 base 000C0000000 mask 7FFC0000000 uncachable
[    0.008382]   2 disabled
[    0.008383]   3 disabled
[    0.008384]   4 disabled
[    0.008385]   5 disabled
[    0.008386]   6 disabled
[    0.008387]   7 disabled
[    0.008459] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.027342] found SMP MP-table at [mem 0x000f6a70-0x000f6a7f]
[    0.027480] BRK [0x02202000, 0x02202fff] PGTABLE
[    0.027487] BRK [0x02203000, 0x02203fff] PGTABLE
[    0.027489] BRK [0x02204000, 0x02204fff] PGTABLE
[    0.027944] BRK [0x02205000, 0x02205fff] PGTABLE
[    0.028158] ACPI: Early table checksum verification disabled
[    0.028284] ACPI: RSDP 0x00000000000F6A00 000024 (v02 PTLTD )
[    0.028299] ACPI: XSDT 0x000000003FEE965B 00005C (v01 INTEL  440BX    06040000 VMW  01324272)
[    0.028318] ACPI: FACP 0x000000003FEFEE73 0000F4 (v04 INTEL  440BX    06040000 PTL  000F4240)
[    0.028336] ACPI: DSDT 0x000000003FEEAA08 01446B (v01 PTLTD  Custom   06040000 MSFT 03000001)
[    0.028351] ACPI: FACS 0x000000003FEFFFC0 000040
[    0.028360] ACPI: FACS 0x000000003FEFFFC0 000040
[    0.028368] ACPI: BOOT 0x000000003FEEA9E0 000028 (v01 PTLTD  $SBFTBL$ 06040000  LTP 00000001)
[    0.028378] ACPI: APIC 0x000000003FEEA29E 000742 (v01 PTLTD  ? APIC   06040000  LTP 00000000)
[    0.028387] ACPI: MCFG 0x000000003FEEA262 00003C (v01 PTLTD  $PCITBL$ 06040000  LTP 00000001)
[    0.028397] ACPI: SRAT 0x000000003FEE9757 0008A8 (v02 VMWARE MEMPLUG  06040000 VMW  00000001)
[    0.028406] ACPI: HPET 0x000000003FEE971F 000038 (v01 VMWARE VMW HPET 06040000 VMW  00000001)
[    0.028415] ACPI: WAET 0x000000003FEE96F7 000028 (v01 VMWARE VMW WAET 06040000 VMW  00000001)
[    0.028439] ACPI: Local APIC address 0xfee00000
[    0.028473] system APIC only can use physical flat
[    0.028474] Setting APIC routing to physical flat.
[    0.050589] Zone ranges:
[    0.050600]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.050605]   DMA32    [mem 0x0000000001000000-0x000000003fffffff]
[    0.050608]   Normal   empty
[... 1151 lines cut ...]
[   69.610448] br-lan: port 1(eth0) entered blocking state
[   69.612099] br-lan: port 1(eth0) entered forwarding state
[   69.613962] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   69.624722] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   69.626701] 8021q: adding VLAN 0 to HW filter on device eth1
[   69.637209] e1000: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   69.640586] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  187.625219] br-wan: port 1(eth1) entered blocking state
[  187.626954] br-wan: port 1(eth1) entered disabled state
[  187.630825] device eth1 entered promiscuous mode
[  187.636371] br-wan: port 1(eth1) entered blocking state
[  187.638097] br-wan: port 1(eth1) entered forwarding state
[  187.640068] IPv6: ADDRCONF(NETDEV_UP): br-wan: link is not ready
[  188.669802] IPv6: ADDRCONF(NETDEV_CHANGE): br-wan: link becomes ready
[53316.524660] br-lan: port 1(eth0) entered disabled state
[... cut ...]
[53316.711777] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[53316.819448] 8021q: adding VLAN 0 to HW filter on device eth0
[53316.822358] br-lan: port 1(eth0) entered blocking state
[53316.824371] br-lan: port 1(eth0) entered disabled state
[53316.828857] device eth0 entered promiscuous mode
[53316.831665] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[53316.834129] br-lan: port 1(eth0) entered blocking state
[53316.836237] br-lan: port 1(eth0) entered forwarding state

blankhang, 2020-02-02:
### 系统日志
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203398] pci_bus 0000:03: resource 1 [mem 0xfd400000-0xfd4fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203401] pci_bus 0000:03: resource 2 [mem 0xe7a00000-0xe7afffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203404] pci_bus 0000:04: resource 0 [io  0x8000-0x8fff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203406] pci_bus 0000:04: resource 1 [mem 0xfd000000-0xfd0fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203409] pci_bus 0000:04: resource 2 [mem 0xe7600000-0xe76fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203412] pci_bus 0000:05: resource 0 [io  0xc000-0xcfff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203414] pci_bus 0000:05: resource 1 [mem 0xfcc00000-0xfccfffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203416] pci_bus 0000:05: resource 2 [mem 0xe7200000-0xe72fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203419] pci_bus 0000:06: resource 1 [mem 0xfc800000-0xfc8fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203422] pci_bus 0000:06: resource 2 [mem 0xe6e00000-0xe6efffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203424] pci_bus 0000:07: resource 1 [mem 0xfc400000-0xfc4fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203427] pci_bus 0000:07: resource 2 [mem 0xe6a00000-0xe6afffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203430] pci_bus 0000:08: resource 1 [mem 0xfc000000-0xfc0fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203432] pci_bus 0000:08: resource 2 [mem 0xe6600000-0xe66fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203435] pci_bus 0000:09: resource 1 [mem 0xfbc00000-0xfbcfffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203437] pci_bus 0000:09: resource 2 [mem 0xe6200000-0xe62fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203440] pci_bus 0000:0a: resource 1 [mem 0xfb800000-0xfb8fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203443] pci_bus 0000:0a: resource 2 [mem 0xe5e00000-0xe5efffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203445] pci_bus 0000:0b: resource 0 [io  0x5000-0x5fff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203448] pci_bus 0000:0b: resource 1 [mem 0xfd300000-0xfd3fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203450] pci_bus 0000:0b: resource 2 [mem 0xe7900000-0xe79fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203453] pci_bus 0000:0c: resource 0 [io  0x9000-0x9fff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203455] pci_bus 0000:0c: resource 1 [mem 0xfcf00000-0xfcffffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203458] pci_bus 0000:0c: resource 2 [mem 0xe7500000-0xe75fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203460] pci_bus 0000:0d: resource 0 [io  0xd000-0xdfff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203463] pci_bus 0000:0d: resource 1 [mem 0xfcb00000-0xfcbfffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203465] pci_bus 0000:0d: resource 2 [mem 0xe7100000-0xe71fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203468] pci_bus 0000:0e: resource 1 [mem 0xfc700000-0xfc7fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203470] pci_bus 0000:0e: resource 2 [mem 0xe6d00000-0xe6dfffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203473] pci_bus 0000:0f: resource 1 [mem 0xfc300000-0xfc3fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203475] pci_bus 0000:0f: resource 2 [mem 0xe6900000-0xe69fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203479] pci_bus 0000:10: resource 1 [mem 0xfbf00000-0xfbffffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203481] pci_bus 0000:10: resource 2 [mem 0xe6500000-0xe65fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203484] pci_bus 0000:11: resource 1 [mem 0xfbb00000-0xfbbfffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203486] pci_bus 0000:11: resource 2 [mem 0xe6100000-0xe61fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203489] pci_bus 0000:12: resource 1 [mem 0xfb700000-0xfb7fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203491] pci_bus 0000:12: resource 2 [mem 0xe5d00000-0xe5dfffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203494] pci_bus 0000:13: resource 0 [io  0x6000-0x6fff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203496] pci_bus 0000:13: resource 1 [mem 0xfd200000-0xfd2fffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203499] pci_bus 0000:13: resource 2 [mem 0xe7800000-0xe78fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203502] pci_bus 0000:14: resource 0 [io  0xa000-0xafff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203504] pci_bus 0000:14: resource 1 [mem 0xfce00000-0xfcefffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203506] pci_bus 0000:14: resource 2 [mem 0xe7400000-0xe74fffff 64bit pref]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203509] pci_bus 0000:15: resource 0 [io  0xe000-0xefff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203511] pci_bus 0000:15: resource 1 [mem 0xfca00000-0xfcafffff]
Sun Feb  2 01:58:53 2020 kern.debug kernel: [    2.203514] pci_bus 0000:15: resource 2 [mem 0xe7000000-0xe70fffff 64bit pref]
[... cut ...]
Sun Feb  2 16:47:02 2020 daemon.notice netifd: Interface 'lan' is now up
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.831665] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.834129] br-lan: port 1(eth0) entered blocking state
Sun Feb  2 16:47:02 2020 kern.info kernel: [53316.836237] br-lan: port 1(eth0) entered forwarding state
Sun Feb  2 16:47:02 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)

kongzheng333, 2020-02-02:
同 我把我感觉有问题的系统日志也放上来

kongzheng333, 2020-02-02:
### 系统日志
Sun Feb  2 19:52:18 2020 daemon.info procd: - init complete -
Sun Feb  2 19:52:19 2020 daemon.notice netifd: Network device 'eth0' link is up
Sun Feb  2 19:52:19 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Feb  2 19:52:19 2020 kern.info kernel: [   12.193261] r8169 0000:02:00.0 eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Feb  2 19:52:19 2020 kern.info kernel: [   12.193767] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sun Feb  2 19:52:20 2020 daemon.notice netifd: Network device 'eth1' link is up
Sun Feb  2 19:52:20 2020 daemon.notice netifd: Interface 'wan' has link connectivity
Sun Feb  2 19:52:20 2020 kern.info kernel: [   13.553986] r8169 0000:03:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
Sun Feb  2 19:52:20 2020 kern.info kernel: [   13.554488] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
Sun Feb  2 19:52:27 2020 daemon.info pdnsd[4710]: pdnsd-1.2.9b-par starting.
Sun Feb  2 19:52:27 2020 daemon.warn pdnsd[4710]: EOF encountered while reading entry count from disk cache file.
Sun Feb  2 19:52:27 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Feb  2 19:52:27 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Feb  2 19:54:21 2020 daemon.err uhttpd[2358]: cat: write error: Broken pipe
Sun Feb  2 20:07:36 2020 daemon.notice miniupnpd[2243]: remove port mapping 12846 UDP because it has expired
Sun Feb  2 20:07:37 2020 daemon.notice miniupnpd[2243]: remove port mapping 42795 UDP because it has expired
Sun Feb  2 20:07:37 2020 daemon.notice miniupnpd[2243]: remove port mapping 39952 UDP because it has expired
Sun Feb  2 20:07:37 2020 daemon.notice miniupnpd[2243]: remove port mapping 37303 UDP because it has expired
Sun Feb  2 20:07:37 2020 daemon.notice miniupnpd[2243]: remove port mapping 12848 UDP because it has expired
Sun Feb  2 20:07:37 2020 daemon.notice miniupnpd[2243]: remove port mapping 12847 UDP because it has expired
Sun Feb  2 22:17:23 2020 daemon.err uhttpd[2358]: sed: write error

kongzheng333, 2020-02-02:
目前还没出现 不过昨天断网时的系统日志和楼主有相似部分

Sun Feb 2 02:05:10 2020 daemon.warn pdnsd[23723]: Caught signal 15. Exiting.
Sun Feb 2 02:05:10 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Feb 2 02:05:10 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Feb 2 02:05:13 2020 cron.info crond[24299]: crond (busybox 1.30.0) started, log level 5
Sun Feb 2 02:05:14 2020 daemon.info pdnsd[24514]: pdnsd-1.2.9b-par starting.

luckyyyyy, 2020-02-02:
我也有断网问题 但是看日志 不是一个问题

coolsnowwolf, 2020-02-02:
eth1 的网口连接有问题

blankhang, 2020-02-02:
> eth1 的网口连接有问题

硬件连接接口吗... 
尝试过重新插拨 插紧后 仍然有此问题
仍然会间歇性断网 ping不通路由

dzhuqi, 2020-02-03:
我的试了几天也有这样情况。看拨号是连接上的，就是网页打不开，显示没有网络，重新连接下网口又可以正常上网，过段时间仍然会间歇性断网（软路由是j1900），目前刷回81的内核使用

blankhang, 2020-02-03:
> 我的试了几天也有这样情况。看拨号是连接上的，就是网页打不开，显示没有网络，重新连接下网口又可以正常上网，过段时间仍然会间歇性断网（软路由是j1900），目前刷回81的内核使用

能分享下81的镜像吗 我也测试看看

luckyyyyy, 2020-02-03:
> 我的试了几天也有这样情况。看拨号是连接上的，就是网页打不开，显示没有网络，重新连接下网口又可以正常上网，过段时间仍然会间歇性断网（软路由是j1900），目前刷回81的内核使用

贴日志看一下啦

ch1993128, 2020-02-05:
极路由B70，网件R8000，极路由1s都有百分之5丢包，硬件连接没有问题的。切以前版本也没有问题，日志看不出啥问题来。。。

dzhuqi, 2020-02-07:
(Powered by LuCI Master (git-20.032.31649-0d31472) / OpenWrt R20.1.24)今天的发现的断网情况，大家看一下我的系统日志（路由器设置每天重启），当时发现拨号是连接上的，就是没有网络，w口重新连接一下才正常`Fri Feb  7 05:00:36 2020 kern.info kernel: [    6.780397] usbcore: registered new interface driver asix
Fri Feb  7 05:00:36 2020 kern.info kernel: [    6.781468] usbcore: registered new interface driver ax88179_178a
Fri Feb  7 05:00:36 2020 kern.info kernel: [    6.797307] PPP generic driver version 2.4.2
Fri Feb  7 05:00:36 2020 kern.info kernel: [    6.798141] PPP MPPE Compression module registered
Fri Feb  7 05:00:36 2020 kern.info kernel: [    6.799004] NET: Registered protocol family 24
Fri Feb  7 05:00:36 2020 kern.info kernel: [    6.805075] usbcore: registered new interface driver snd-usb-audio
Fri Feb  7 05:00:36 2020 kern.info kernel: [    6.820113] usbcore: registered new interface driver rt2800usb
Fri Feb  7 05:00:36 2020 kern.info kernel: [    6.822663] usbcore: registered new interface driver ath9k_htc
Fri Feb  7 05:00:36 2020 user.info kernel: [    6.824543] kmodloader: done loading kernel modules from /etc/modules.d/*
Fri Feb  7 05:00:36 2020 kern.info kernel: [    7.063879] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts:
Fri Feb  7 05:00:36 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Feb  7 05:00:36 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: started, version 2.80 cachesize 2048
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: DNS service limited to local subnets
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify dumpfile
Fri Feb  7 05:00:36 2020 daemon.warn dnsmasq[3032]: warning: ignoring resolv-file flag because no-resolv is set
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using local addresses only for domain test
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using local addresses only for domain onion
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using local addresses only for domain localhost
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using local addresses only for domain local
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using local addresses only for domain invalid
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using local addresses only for domain bind
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using local addresses only for domain lan
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using nameserver 119.29.29.29#53
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: using nameserver 114.114.114.114#53
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: read /etc/hosts - 22 addresses
Fri Feb  7 05:00:36 2020 daemon.info dnsmasq[3032]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Fri Feb  7 05:00:37 2020 kern.info kernel: [    8.454757] xt_FULLCONENAT: RFC3489 Full Cone NAT module
Fri Feb  7 05:00:37 2020 kern.info kernel: [    8.454757] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
Fri Feb  7 05:00:37 2020 user.notice : Added device handler type: 8021ad
Fri Feb  7 05:00:37 2020 user.notice : Added device handler type: 8021q
Fri Feb  7 05:00:37 2020 user.notice : Added device handler type: macvlan
Fri Feb  7 05:00:37 2020 user.notice : Added device handler type: veth
Fri Feb  7 05:00:37 2020 user.notice : Added device handler type: bridge
Fri Feb  7 05:00:37 2020 user.notice : Added device handler type: Network device
Fri Feb  7 05:00:37 2020 user.notice : Added device handler type: tunnel
Fri Feb  7 05:00:37 2020 daemon.err block: /dev/loop0 is already mounted on /overlay
Fri Feb  7 05:00:37 2020 daemon.err block: /dev/sda1 is already mounted on /boot
Fri Feb  7 05:00:37 2020 daemon.err block: /dev/sda2 is already mounted on /rom
Fri Feb  7 05:00:37 2020 daemon.err block: /dev/sdb1 is already mounted on /mnt/sdb1
Fri Feb  7 05:00:37 2020 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Fri Feb  7 05:00:37 2020 cron.info crond[3166]: crond (busybox 1.30.0) started, log level 5
Fri Feb  7 05:00:37 2020 daemon.notice netifd: Interface 'lan' is enabled
Fri Feb  7 05:00:37 2020 daemon.notice netifd: Interface 'lan' is setting up now
Fri Feb  7 05:00:37 2020 daemon.notice netifd: Interface 'lan' is now up
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.086729] 8021q: adding VLAN 0 to HW filter on device eth0
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.087286] br-lan: port 1(eth0) entered blocking state
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.087664] br-lan: port 1(eth0) entered disabled state
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.088123] device eth0 entered promiscuous mode
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.088897] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.116881] 8021q: adding VLAN 0 to HW filter on device eth1
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.117457] br-lan: port 2(eth1) entered blocking state
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.117838] br-lan: port 2(eth1) entered disabled state
Fri Feb  7 05:00:37 2020 kern.info kernel: [    9.118303] device eth1 entered promiscuous mode
Fri Feb  7 05:00:37 2020 daemon.notice netifd: Interface 'loopback' is enabled
[... cut ...]
Fri Feb  7 09:50:22 2020 daemon.info dnsmasq-dhcp[25523]: DHCPREQUEST(br-lan) 192.168.1.139 1c:6f:65:86:37:e4
Fri Feb  7 09:50:22 2020 daemon.info dnsmasq-dhcp[25523]: DHCPACK(br-lan) 192.168.1.139 1c:6f:65:86:37:e4 z-GA-MA770T-UD3P
Fri Feb  7 09:50:23 2020 daemon.warn odhcpd[3136]: DHCPV6 CONFIRM IA_NA from 00044b58fe9ee41de69df59d509358ce368c on br-lan: not on-link
Fri Feb  7 09:50:23 2020 daemon.warn odhcpd[3136]: DHCPV6 SOLICIT IA_NA from 00044b58fe9ee41de69df59d509358ce368c on br-lan: ok 240e:62:b18d:5300::3bb/128
`

alphavip, 2020-02-07:
我用以前的config编译也是间歇断网，我路由器是dhcp连接光猫上网的，连接正常，在路由器上ping都正常，但就是连接路由器的设备间歇性断网，重新删除config重来，目前没发现断网，就只有一个libcurl对openssl依赖问题

closed 2020-02-19 by coolsnowwolf

jack338c, 2023-05-26:
一夜的情况，只要有 possible DNS-rebind attack detected: qqshow2-ufs.qq.com出现，就会有这个问题