coolsnowwolf/lede #2926: 编译到一半（编译开始大约1~2小时后）出现交互选项，应该怎么做？

SciUnivMoring, 2020-02-02:
[image]

blankhang, 2020-02-02:
我按了回车....

josephzjw, 2020-02-02:
我也出现这个状况，现在50G磁盘满了，之前已经`make clean`过了..

4593645, 2020-02-02:
什么机器环境？

kl9800, 2020-02-02:
> 什么机器环境？

乌班图18

qdfansanyblog, 2020-02-02:
一样出现这个选项，选的n回车

hue715, 2020-02-02:
出现交互前出现以下警报：（用make menuconfig保存的.config)
make[5]: Entering directory '/home/tom/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.98'
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
net/sched/Kconfig:44: warning: menuconfig statement without prompt
.config:3202:warning: override: M586MMX changes choice state
.config:4192:warning: symbol value 'm' invalid for NF_NAT_REDIRECT
.config:5933:warning: override: SPARSEMEM_MANUAL changes choice state
*
* Restart config...
*
*
* PHY Device support and infrastructure
*
PHY Device support and infrastructure (PHYLIB) [M/y/?] m

SciUnivMoring, 2020-02-02:
> 什么机器环境？

Ubuntu 18.04 LTS

josephzjw, 2020-02-03:
我后来发现好像是我的硬盘满了，我 `make dirclean`， `make clean` 以后重新编译好像就行了

closed 2020-02-03 by coolsnowwolf