coolsnowwolf/lede #2931: ssr ++自动订阅节点的兼容问题

tinnlux, 2020-02-03:
求在自动订阅节点以后在每个节点信息里自动加上	option insecure '1'这行。实测很多节点不开这个无法使用，或者默认增加	option insecure '0'也行，我手动替换也比一个节点一个节点手动增加这行快很多。谢谢！！

luckyyyyy, 2020-02-03:
这个自己改

vi /usr/share/shadowsocksr/subscribe.lua 
随便找个地方加一行

[image]

tinnlux, 2020-02-03:
@luckyyyyy 谢谢

tinnlux, 2020-02-03:
@luckyyyyy 源码文件是哪个？我想直接改我本地源码

tinnlux, 2020-02-03:
@luckyyyyy 您好！我看了看v2ray更新订阅节点源码部分，缺少判断是否需要insecure的语句啊。是否可以考虑有空的时候加上这个判断？

luckyyyyy, 2020-02-03:
@tinnlux 你先手动加一下 脚本里面直接加就好了 我对v2ray了解不多 不知道这个参数干嘛的

tinnlux, 2020-02-03:
@luckyyyyy 我手动加了，很完美，如果合并不到主线的话我就自己在源码添加好了。

closed 2020-02-03 by tinnlux

luckyyyyy, 2020-02-03:
> @luckyyyyy 我手动加了，很完美，如果合并不到主线的话我就自己在源码添加好了。

{"host":"","path":"","tls":"","add":"XXX","port":10090,"aid":2,"net":"tcp","type":"none","v":"2","ps":"[V2ray]日本1-GMO-×0.9-Netflix","id":"XXXXXXXX"}

因为从返回的解析中 这个参数似乎是无从判断的？怎么判断？
还有就是这个参数干嘛用的，我发现我的不加也不行。

tinnlux, 2020-02-04:
@luckyyyyy v2ray文档上有解释，我记得是允许什么，具体的记不清了。windows客户端这一项默认是true，这一项在订阅里是不添加判断的。我用了几个客户端都是默认true。希望还是能在下一次合并的时候把这个源码加上。

tinnlux, 2020-02-04:
@luckyyyyy 
我查到的定义是：
"allowInsecure": false
            //检测证书有效性
且这个定义只在客户端，服务端无此项。这个参数应该是跟底层传输有关，默认是true

luckyyyyy, 2020-02-04:
这个我查文档看到了已经。

发自我的iPhone

tinnlux, 2020-02-13:
@luckyyyyy passwall里关于这个的解释是允许不安全连接。当勾选时，V2Ray 不会检查远端主机所提供的 TLS 证书的有效性。我建议在主线里默认添加option insecure '1'，否则很多机场的v2节点都不能使用的。