coolsnowwolf/lede #2939: R7800最新源码编译后无法自动识别和挂载usb移动硬盘

pk49800, 2020-02-04:
主机型号：Netgear Nighthawk X4S R7800 
固件版本：OpenWrt R20.1.24 / LuCI Master (git-20.032.31649-0d31472)
内核版本：4.14.167
使用云编译的固件从R19最后一个版本升级至R20后，无法自动识别挂载USB移动硬盘。笔记本的USB3端口可正常访问移动硬盘，试过路由器上的两个USB3接口均无法识别。

bihcou, 2020-02-06:
同样问题，使用官源1907的话还会导致编译出错
去掉extra package里的automount后编译通过

coolsnowwolf, 2020-02-06:
看看是不是 antfs 对 ntfs 的格式支持问题

pk49800, 2020-02-07:
> 看看是不是 antfs 对 ntfs 的格式支持问题

经过测试，移动硬盘是NTFS格式的，无法自动识别挂载。插入FAT格式的USB3.1 U盘可以自动挂载，samba访问也没问题。

EnnawYang, 2020-02-07:
树莓派4b使用antfs挂载U盘失败，能识别但是不能挂载
```
Fri Feb  7 15:05:21 2020 daemon.err block: /dev/loop0 is already mounted on /rom/overlay
Fri Feb  7 15:05:21 2020 daemon.err block: /dev/mmcblk0p1 is already mounted on /boot
Fri Feb  7 15:05:21 2020 daemon.err block: /dev/mmcblk0p2 is already mounted on /rom
Fri Feb  7 15:05:21 2020 daemon.err block: No "mount.ntfs" utility available
Fri Feb  7 15:05:21 2020 daemon.err block: mounting /dev/sda1 (ntfs) as /mnt/sda1 failed (2) - No such file or directory
......
Fri Feb  7 15:05:23 2020 daemon.err block: No "mount.ntfs" utility available
Fri Feb  7 15:05:23 2020 daemon.err block: mounting /dev/sda1 (ntfs) as /mnt/sda1 failed (25) - Not a tty
```
内核部分日志
```
[    2.134075] ANTFS Module: Version 
[    2.139350] exFAT: Version 1.2.9
[    2.143442] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    2.150298] ehci-platform: EHCI generic platform driver
[    2.155925] ehci-pci: EHCI PCI platform driver
[    2.161213] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    2.167700] ohci-platform: OHCI generic platform driver
[    2.173539] uhci_hcd: USB Universal Host Controller Interface driver
[    2.180160] ohci-pci: OHCI PCI platform driver
[    2.184732] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    2.198241] init: - preinit -
[    2.446342] bcmgenet: Skipping UMAC reset
[    2.451563] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
[    2.459934] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    2.712430] scsi 0:0:0:0: Direct-Access     SanDisk  Ultra Fit        1.00 PQ: 0 ANSI: 6
[    2.721359] sd 0:0:0:0: [sda] 242614272 512-byte logical blocks: (124 GB/116 GiB)
[    2.730780] sd 0:0:0:0: [sda] Write Protect is off
[    2.735591] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
[    2.736249] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[    2.742209] random: procd: uninitialized urandom read (4 bytes read)
[    2.757126]  sda: sda1
[    2.761318] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

bihcou, 2020-02-07:
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'kmod-fs-antfs', which does not exist

发现了这个，难怪使用extra package里的automount不能通过

AmadeusGhost, 2020-02-19:
This issue can be closed，fixed from db014d1026c02106e47832bfdc85db6ff26d1d08

closed 2020-02-19 by coolsnowwolf