coolsnowwolf/lede #2943: 请Lean's支持一下极路由X c526a

lloyd18, 2020-02-04:
ipq4019+mt7615方案，目前唯一能用的第三方固件是PandoraBox 19年2月版本
可是PandoraBox已不再更新。
我在在另一位大腿的源码看到有c526a的dts，自己尝试可以编译成功但是刷进去无限重启
https://github.com/jimstar06/openwrt-LGA1150

极路由X(C526A)刷入PandoraBox简单教程 
https://www.right.com.cn/forum/thread-438488-1-1.html
[image]

AmadeusGhost, 2020-02-05:
https://github.com/AmadeusGhost/lede/commit/22386d6c9d53beeb4942637d7998dddfa08958bc
你可以试下
bdinfo能不能dd一下发上来，毕竟我没这机子，邮箱：938640796@qq.com
```
cat /proc/mtd
dd if=/dev/mtd8 of=/tmp/bdinfo.bin
```

lloyd18, 2020-02-05:
> [AmadeusGhost@22386d6](https://github.com/AmadeusGhost/lede/commit/22386d6c9d53beeb4942637d7998dddfa08958bc) 你可以试下
> bdinfo能不能dd一下发上来，毕竟我没这机子，邮箱：[938640796@qq.com](mailto:938640796@qq.com)
> 
> ```
> cat /proc/mtd
> dd if=/dev/mtd8 of=/tmp/bdinfo.bin
> ```

已经发过去了

AmadeusGhost, 2020-02-05:
> > [AmadeusGhost@22386d6](https://github.com/AmadeusGhost/lede/commit/22386d6c9d53beeb4942637d7998dddfa08958bc) 你可以试下
> > bdinfo能不能dd一下发上来，毕竟我没这机子，邮箱：[938640796@qq.com](mailto:938640796@qq.com)
> > ```
> > cat /proc/mtd
> > dd if=/dev/mtd8 of=/tmp/bdinfo.bin
> > ```
> 
> 已经发过去了

由图可得/dev/mtd11才是bdinfo，所以应该
```
dd if=/dev/mtd11 of=/tmp/bdinfo.bin
```
有的小尴尬

lloyd18, 2020-02-05:
> > > [AmadeusGhost@22386d6](https://github.com/AmadeusGhost/lede/commit/22386d6c9d53beeb4942637d7998dddfa08958bc) 你可以试下
> > > bdinfo能不能dd一下发上来，毕竟我没这机子，邮箱：[938640796@qq.com](mailto:938640796@qq.com)
> > > ```
> > > cat /proc/mtd
> > > dd if=/dev/mtd8 of=/tmp/bdinfo.bin
> > > ```
> > 
> > 
> > 已经发过去了
> 
> 由图可得/dev/mtd11才是bdinfo，所以应该
> 
> ```
> dd if=/dev/mtd11 of=/tmp/bdinfo.bin
> ```
> 
> 有点小尴尬

小问题 已重发

AmadeusGhost, 2020-02-05:
http://tmp.link/f/5e3ada71b4313
http://tmp.link/f/5e3adaa36c453
编译出来的固件你试试看？

lloyd18, 2020-02-05:
> http://tmp.link/f/5e3ada71b4313
> http://tmp.link/f/5e3adaa36c453
> 编译出来的固件你试试看？

不行，后台升级提示不支持的固件，pb-boot下提示固件刷写失败，

AmadeusGhost, 2020-02-05:
> > http://tmp.link/f/5e3ada71b4313
> > http://tmp.link/f/5e3adaa36c453
> > 编译出来的固件你试试看？
> 
> 不行，后台升级提示不支持的固件，pb-boot下提示固件刷写失败，我在用的固件大小是15M的，不知道和固件大小有没有关系

pb-boot下是用的sysupgrade.bin嘛？

lloyd18, 2020-02-05:
> > > http://tmp.link/f/5e3ada71b4313
> > > http://tmp.link/f/5e3adaa36c453
> > > 编译出来的固件你试试看？
> > 
> > 
> > 不行，后台升级提示不支持的固件，pb-boot下提示固件刷写失败，我在用的固件大小是15M的，不知道和固件大小有没有关系
> 
> pb-boot下是用的sysupgrade.bin嘛？

两个都试了，factory.ubi未识别，sysupgrade.bin刷到40% 提示固件刷写失败

lloyd18, 2020-02-06:
> 由图可得/dev/mtd11才是bdinfo，所以应该
>
> ```
> dd if=/dev/mtd11 of=/tmp/bdinfo.bin
> ```
>
> 有点小尴尬

大佬 会有后续的操作吗？

AmadeusGhost, 2020-02-06:
先刷一遍潘多拉，然后将.ubi固件上传到/tmp
```
mtd erase rootfs
mtd write /tmp/xxx.ubi rootfs
```

lloyd18, 2020-02-06:
> 先刷一遍潘多拉，然后将.ubi固件上传到/tmp
> 
> ```
> mtd erase rootfs
> mtd write /tmp/xxx.ubi rootfs
> ```

我现在用的就是潘多拉 要重新刷一遍吗？
mtd erase rootfs 提示
Could not open mtd device: rootfs
Could not open mtd device: rootfs

AmadeusGhost, 2020-02-06:
> > 先刷一遍潘多拉，然后将.ubi固件上传到/tmp
> > ```
> > mtd erase rootfs
> > mtd write /tmp/xxx.ubi rootfs
> > ```
> 
> 我现在用的就是潘多拉 要重新刷一遍吗？
> mtd erase rootfs 提示
> Could not open mtd device: rootfs
> Could not open mtd device: rootfs

不用，前面我说错了抱歉
```
mtd erase hiwifi_rootfs
mtd write /tmp/xxx.ubi hiwifi_rootfs
```

lloyd18, 2020-02-06:
> > > 先刷一遍潘多拉，然后将.ubi固件上传到/tmp
> > > ```
> > > mtd erase rootfs
> > > mtd write /tmp/xxx.ubi rootfs
> > > ```
> > 
> > 
> > 我现在用的就是潘多拉 要重新刷一遍吗？
> > mtd erase rootfs 提示
> > Could not open mtd device: rootfs
> > Could not open mtd device: rootfs
> 
> 不用，前面我说错了抱歉
> 
> ```
> mtd erase hiwifi_rootfs
> mtd write /tmp/xxx.ubi hiwifi_rootfs
> ```

操作了一遍，reboot后没什么改变

AmadeusGhost, 2020-02-06:
重启后不是op？
你截下主页的图我看看

lloyd18, 2020-02-06:
[image]
没变，刚才运行命令也没报错

lloyd18, 2020-02-06:
> 重启后不是op？
> 你截下主页的图我看看

我备份ubi分区看到容量也很大 和rootfs一样都是44MB

AmadeusGhost, 2020-02-06:
我问了1150大佬，之前是我搞错了(╥﹏╥)
```
mtd erase ubi
mtd erase hiwifi_rootfs
mtd write /tmp/xxx.ubi ubi
```

lloyd18, 2020-02-06:
> 我问了1150大佬，之前是我搞错了(╥﹏╥)
> 
> ```
> mtd erase ubi
> mtd erase hiwifi_rootfs
> mtd write /tmp/xxx.ubi ubi
> ```

哭了，还是不行，刷完重启自动进入pb-boot，然后在pb-boot可以刷入你的固件 但是无限重启

AmadeusGhost, 2020-02-06:
有ttl日志吗？

lloyd18, 2020-02-06:
> 有ttl日志吗？

没有 日志要怎么搞得到

AmadeusGhost, 2020-02-06:
百度路由器接ttl

lloyd18, 2020-02-06:
> 百度路由器接ttl

好像要专用工具，接线，动硬件，这就有点麻烦了

AmadeusGhost, 2020-02-07:
刷一遍潘多拉，然后
```
mtd erase ubi
mtd write /tmp/xxx.ubi ubi
```
再试试吧

ch1993128, 2020-02-07:
都尝试过的，我和楼上也是一样的问题，进不去op，在用官版。

ch1993128, 2020-02-07:
> 刷一遍潘多拉，然后
> 
> ```
> mtd erase ubi
> mtd write /tmp/xxx.ubi ubi
> ```
> 
> 再试试吧

意思说无限重启是uboot在写入固件时写入分区错误的问题？编译固件本身是可以用的？等去学校了尝试一下。。。。如果不动官方eeprom和其他分区的情况下可以用op那可太省心了

AmadeusGhost, 2020-02-07:
> > 刷一遍潘多拉，然后
> > ```
> > mtd erase ubi
> > mtd write /tmp/xxx.ubi ubi
> > ```
> > 
> > 
> > 再试试吧
> 
> 意思说无限重启是uboot在写入固件时写入分区错误的问题？编译固件本身是可以用的？等去学校了尝试一下。。。。如果不动官方eeprom和其他分区的情况下可以用op那可太省心了

spi上的分区都不会动到的，只动nand上的

lloyd18, 2020-02-07:
> 刷一遍潘多拉，然后
> 
> ```
> mtd erase ubi
> mtd write /tmp/xxx.ubi ubi
> ```
> 
> 再试试吧

又试了两遍，还是一样，应该是bootloader这个分区有问题

ch1993128, 2020-02-07:
> > > 刷一遍潘多拉，然后
> > > ```
> > > mtd erase ubi
> > > mtd write /tmp/xxx.ubi ubi
> > > ```
> > > 
> > > 
> > > 再试试吧
> > 
> > 
> > 意思说无限重启是uboot在写入固件时写入分区错误的问题？编译固件本身是可以用的？等去学校了尝试一下。。。。如果不动官方eeprom和其他分区的情况下可以用op那可太省心了
> 
> spi上的分区都不会动到的，只动nand上的

pb-boot好像也是这么写入的哦的哦，无限重启。。。。。LGA-1150那个源码编译出来的，我不知道最近更新没

ch1993128, 2020-02-07:
> > 刷一遍潘多拉，然后
> > ```
> > mtd erase ubi
> > mtd write /tmp/xxx.ubi ubi
> > ```
> > 
> > 
> > 再试试吧
> 
> 又试了两遍，还是一样，应该是bootloader这个分区有问题

> > 刷一遍潘多拉，然后
> > ```
> > mtd erase ubi
> > mtd write /tmp/xxx.ubi ubi
> > ```
> > 
> > 
> > 再试试吧
> 
> 又试了两遍，还是一样，应该是bootloader这个分区有问题

pbboot分区是在2M的小芯片里的。LGA1150那个有可能是固件问题，也有可能是分区写错位置了。。。具体不清楚，我的也是无线重启进入pbboot

lloyd18, 2020-02-07:
> > > 刷一遍潘多拉，然后
> > > ```
> > > mtd erase ubi
> > > mtd write /tmp/xxx.ubi ubi
> > > ```
> > > 
> > > 
> > > 再试试吧
> > 
> > 
> > 又试了两遍，还是一样，应该是bootloader这个分区有问题
> 
> > > 刷一遍潘多拉，然后
> > > ```
> > > mtd erase ubi
> > > mtd write /tmp/xxx.ubi ubi
> > > ```
> > > 
> > > 
> > > 再试试吧
> > 
> > 
> > 又试了两遍，还是一样，应该是bootloader这个分区有问题
> 
> pbboot分区是在2M的小芯片里的。LGA1150那个有可能是固件问题，也有可能是分区写错位置了。。。。具体不清楚，用不了我就一直用原厂的最后第二版了。我的也是无限重启进入pbboot。并不能进入openwrt

看来没办法了，已经尽力了，就让它寿终正寝吧。虽然有点可惜。你用的官方哪个版本，最新版我浏览器进后台操作经常卡住。

AmadeusGhost, 2020-02-07:
http://tmp.link/f/5e3d6760a9c10
估计我抄过去有问题
这是1150源码的固件，你们试试

lloyd18, 2020-02-07:
> 估计我抄过去有问题
> 这是1150大佬源码的固件，你们试试
> http://tmp.link/f/5e3d6760a9c10

老样子，无限重启

AmadeusGhost, 2020-02-07:
> > 估计我抄过去有问题
> > 这是1150大佬源码的固件，你们试试
> > http://tmp.link/f/5e3d6760a9c10
> 
> 老样子，无限重启

你的极X是正式版吗？

lloyd18, 2020-02-07:
> > > 估计我抄过去有问题
> > > 这是1150大佬源码的固件，你们试试
> > > http://tmp.link/f/5e3d6760a9c10
> > 
> > 
> > 老样子，无限重启
> 
> 你的极X是正式版吗？

特意翻看了下闲鱼的购买记录，说是工程版

lloyd18, 2020-02-07:
> > > 估计我抄过去有问题
> > > 这是1150大佬源码的固件，你们试试
> > > http://tmp.link/f/5e3d6760a9c10
> > 
> > 
> > 老样子，无限重启
> 
> 你的极X是正式版吗？

给大佬汇报个好消息，我一顿操作，发现可以进去了，但是无线用不了
[image]

AmadeusGhost, 2020-02-08:
工程版有什么不同吗？一顿操作是怎么搞，好奇

lloyd18, 2020-02-08:
> 工程版有什么不同吗？一顿操作是怎么搞，好奇

应该没什么不同，把我也搞懵了，我自己也编译了几次，先写入ubi，重启后进不去LUCI，但是可以用winscp进路由目录 再用命令sysupgrade xxx.bin就能进了后台密码还是潘多拉的，5G信号能中继，2G没反应

ch1993128, 2020-02-09:
> > 工程版有什么不同吗？一顿操作是怎么搞，好奇
> 
> 应该没什么不同，把我也搞懵了，我自己也编译了几次，先写入ubi，重启后进不去LUCI，但是可以用winscp进路由目录 再用命令sysupgrade xxx.bin就能进了后台密码还是潘多拉的，5G信号能中继，2G没反应

改一下驱动配置就应该能用了，驱动应该没选全。sysupgrade xxx.bin指的是？？？我记得编译出来只有写入ubi分区的文件哦。

ch1993128, 2020-02-09:
> > > 工程版有什么不同吗？一顿操作是怎么搞，好奇
> > 
> > 
> > 应该没什么不同，把我也搞懵了，我自己也编译了几次，先写入ubi，重启后进不去LUCI，但是可以用winscp进路由目录 再用命令sysupgrade xxx.bin就能进了后台密码还是潘多拉的，5G信号能中继，2G没反应
> 
> 改一下驱动配置就应该能用了，驱动应该没选全。sysupgrade xxx.bin指的是？？？我记得编译出来只有写入ubi分区的文件哦。

> > 工程版有什么不同吗？一顿操作是怎么搞，好奇
> 
> 应该没什么不同，把我也搞懵了，我自己也编译了几次，先写入ubi，重启后进不去LUCI，但是可以用winscp进路由目录 再用命令sysupgrade xxx.bin就能进了后台密码还是潘多拉的，5G信号能中继，2G没反应

哈哈哈，极X要涨价了。噗

AmadeusGhost, 2020-02-09:
> 改一下驱动配置就应该能用了，驱动应该没选全。sysupgrade xxx.bin指的是？？？我记得编译出来只有写入ubi分区的文件哦。照这么说我是luci没有编译选进去。进去固件了，但是只是编译固件包没选全。。。。。正在系统的看这个，等学会了看看能不能做点事情

sysupgrade openwrt-ipq40xx-hiwifi_c526a-squashfs-nand-sysupgrade.bin
想得美，驱动是全的，估计是工程版有些电路或者GPIO不一样

ch1993128, 2020-02-09:
> > 改一下驱动配置就应该能用了，驱动应该没选全。sysupgrade xxx.bin指的是？？？我记得编译出来只有写入ubi分区的文件哦。照这么说我是luci没有编译选进去。进去固件了，但是只是编译固件包没选全。。。。。正在系统的看这个，等学会了看看能不能做点事情
> 
> sysupgrade openwrt-ipq40xx-hiwifi_c526a-squashfs-nand-sysupgrade.bin
> 想得美，驱动是全的，估计是工程版有些电路或者GPIO不一样
我的是正式版。。。。工程版gpio和spi驱动都不一样

AmadeusGhost, 2020-02-09:
> 我的是正式版。。。。工程版gpio和spi驱动都不一样

正式版能刷op啊，先刷潘多拉，然后
```
mtd erase ubi
mtd write /tmp/xxx.ubi ubi
```
最后拔电重启

ch1993128, 2020-02-09:
以前试过。。也是差不多的问题，机器不在手上，具体得到学校了再说

lloyd18, 2020-02-10:
刷回官方版，不折腾了。

closed 2020-02-10 by lloyd18

ch1993128, 2020-02-11:
嗯，官方版信号好很多，不知道啥原因，没仔细研究，op能扶墙而已

lgt5684, 2020-02-17:
哪位大佬能发份UBOOT给我。搞没了。谢谢！lgt_8888@qq.com

alphavip, 2020-05-27:
> > 我的是正式版。。。。工程版gpio和spi驱动都不一样
> 
> 正式版能刷op啊，先刷潘多拉，然后
> 
> ```
> mtd erase ubi
> mtd write /tmp/xxx.ubi ubi
> ```
> 
> 最后拔电重启

大佬，怎么改极路由x openwrt的分区啊，我编译得openwrt软件包空间只有2M，clash下载好内核就没空间了

AmadeusGhost, 2020-05-27:
> 怎么改极路由x openwrt的分区啊，我编译得openwrt软件包空间只有2M，clash下载好内核就没空间了

挂载点那把 hiwifi_opt 设置成 overlay

alphavip, 2020-05-28:
> > 怎么改极路由x openwrt的分区啊，我编译得openwrt软件包空间只有2M，clash下载好内核就没空间了
> 
> 挂载点那把 hiwifi_opt 设置成 overlay

感谢指导，我把hiwifi_opt那个分区格式化然后挂载成overlay 30多m空间了
顺便学习了一下openwrt得分区知识

sealandwu, 2021-02-17:
> > > 怎么改极路由x openwrt的分区啊，我编译得openwrt软件包空间只有2M，clash下载好内核就没空间了
> > 
> > 
> > 挂载点那把 hiwifi_opt 设置成 overlay
> 
> 感谢指导，我把hiwifi_opt那个分区格式化然后挂载成overlay 30多m空间了
> 顺便学习了一下openwrt得分区知识

大佬，能顺便讲一下格式化及挂载overlay的操作步骤吗？自己编辑的固件只有5M 的软件空间。

root@OpenWrt:~# cat /proc/partitions
major minor  #blocks  name

  31        0      45056 mtdblock0
  31        1      45056 mtdblock1
  31        2      40960 mtdblock2
  31        3        256 mtdblock3
  31        4        128 mtdblock4
  31        5        384 mtdblock5
  31        6         64 mtdblock6
  31        7         64 mtdblock7
  31        8         64 mtdblock8
  31        9        512 mtdblock9
  31       10         64 mtdblock10
  31       11         64 mtdblock11
  31       12         64 mtdblock12
  31       13        128 mtdblock13
  31       14         64 mtdblock14
  31       15         64 mtdblock15
  31       16         64 mtdblock16
 254        0      31620 ubiblock0_1
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 02c00000 00020000 "ubi"
mtd1: 02c00000 00020000 "hiwifi_rootfs"
mtd2: 02800000 00020000 "hiwifi_opt"
mtd3: 00040000 00010000 "SBL1"
mtd4: 00020000 00010000 "MIBIB"
mtd5: 00060000 00010000 "QSEE"
mtd6: 00010000 00010000 "CDT"
mtd7: 00010000 00010000 "DDRPARAMS"
mtd8: 00010000 00010000 "APPSBLENV"
mtd9: 00080000 00010000 "APPSBL"
mtd10: 00010000 00010000 "ART"
mtd11: 00010000 00010000 "bdinfo"
mtd12: 00010000 00010000 "debug"
mtd13: 00020000 00010000 "oem"
mtd14: 00010000 00010000 "Factory"
mtd15: 00010000 00010000 "hw_panic"
mtd16: 00010000 00010000 "upgrade_config"

wujunze, 2021-02-23:
Hello 各位路由器玩家朋友    极路由X 现在有 稳定 可用的 OpenWrt  固件了吗