coolsnowwolf/lede #2948: ssr plus+对ss支持好不是很友好

mmeiwenlong, 2020-02-04:
我有个机场，在paswall上面正常，而在ssr plus+上面，ss节点也正常下载下来了，就是不通，后来用v2ray节点，能正常显示延迟时间，就是不起效果，后来问机场管理，说你的这个可能不支持SIP 003/ simple-obfs  请问后续的版本能让他支持么？感谢！

hcym, 2020-02-04:
混淆参数，自己看看帖子，大雕的支持标准ss，加参数要自己改动，

jyy5206, 2020-02-06:
https://github.com/coolsnowwolf/lede/issues/2864

coolsnowwolf, 2020-02-06:
这个主要是我没有测试环境，自己也用不到，你可以根据 
https://github.com/coolsnowwolf/lede/issues/2864

这里添加测试，如果没问题发一个 PR

jyy5206, 2020-02-09:
尊敬的 @coolsnowwolf ，

谢谢您的关注。

还望您（或其它大佬）有空的时候，亲自把 SS+obfs 组合塞进源码。

因为我不是 IT 业者，仅会用一些初步的命令如 vi、ls，再多就不会了。
#2864 也是我在众多网友的指导下，瞎蒙成功的。

更新后的2个文件：
https://raw.githubusercontent.com/jyy5206/SS_obfs/master/client-config.lua
https://raw.githubusercontent.com/jyy5206/SS_obfs/master/shadowsocksr
但是我不懂如何 PR。而且也不确定这 2 个文件是否和其他路由器兼容？

另：还需要增加 obfs 插件。
我用的是： http://openwrt-dist.sourceforge.net/packages/base/mips_24kc/simple-obfs_0.0.5-4_mips_24kc.ipk
就我的理解，不同路由器，此文件是不同的。（openwrt-dist.sourceforge.net 下都有）
如何在您的源码中正确的塞进去，我也不懂了。。。

谢谢您一如既往的贡献！

此致
        敬礼！

jyy5206, 2020-02-12:
今天发现 SS+obfs 组合功能已经合并到源码了。
一次性编译成功。
刷机测试后，一切OK！

就我使用的 Netgear 4300V1 路由，共享固件如下：

全新安装: https://github.com/jyy5206/OpenWRT-firmware-R20.1.24-for-Netgear-4300V1/blob/master/openwrt-ar71xx-nand-wndr4300-ubi-factory.img?raw=true

更新安装: https://github.com/jyy5206/OpenWRT-firmware-R20.1.24-for-Netgear-4300V1/blob/master/openwrt-ar71xx-nand-wndr4300-squashfs-sysupgrade.tar?raw=true

再次感谢 Mr. Lean 和众多专家一如既往的贡献！

hcym, 2020-02-12:
明天试试看看，cdn的反代

closed 2020-02-19 by coolsnowwolf