coolsnowwolf/lede #2951: 哪位大佬帮看看这错误是什么问题，world文件是源码的，还是我添加的软件的问题

authon, 2020-02-05:
哪位大佬帮看看这错误是什么问题，world文件是源码的，还是我添加的软件的问题，action在线编译出现的问题

make: *** [world] Error 1
/home/runner/work/OpenWrt-Actions-Lean/OpenWrt-Actions-Lean/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
##[error]Process completed with exit code 2.

CYaNu, 2020-02-05:
不是源发地方 往上面翻

authon, 2020-02-05:
> 不是源发地方 往上面翻

Run cd openwrt
  cd openwrt
  echo -e "$(nproc) thread compile"
  make -j$(nproc)
  shell: /bin/bash -e {0}
2 thread compile
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..8b05d9f36ddfba2c9d2af532bfe0088c7e19f325
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..8b05d9f36ddfba2c9d2af532bfe0088c7e19f325
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..8b05d9f36ddfba2c9d2af532bfe0088c7e19f325
make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile

CYaNu, 2020-02-05:
建议重新下源码

authon, 2020-02-05:
> 建议重新下源码

fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..8b05d9f36ddfba2c9d2af532bfe0088c7e19f325

重新新编译还是这个，我是action自动编译的，源码实时下载

closed 2020-02-19 by coolsnowwolf