coolsnowwolf/lede #2955: ssrplus+ trojan 插件 大陆白名单 部分网站无法访问的问题

samloya, 2020-02-05:
反馈一下：
1. ssrplus+ 插件 使用trojan 采用大陆白名单的方式   部分网站 无法正常打开，比如  233blog.com  btbtt.co    这两个网站
2. 如果是使用 vr2ay  ss  类型 则没有以上问题，都能顺利打开网页，现在如果要使用trojan 打开以上两个网站，只能使用静态解析 将网站域名与IP绑定，才能顺利打开。。请L大雕  排查一下原因。 
3. ssr v2ray  的白名单  是不是 跟 trojan  用的不是一套地址列表？

coolsnowwolf, 2020-02-06:
一样的

armandlee, 2020-02-08:
同样问题

OUZYcn, 2020-02-09:
> 反馈一下：
> 
> 1. ssrplus+ 插件 使用trojan 采用大陆白名单的方式   部分网站 无法正常打开，比如  233blog.com  btbtt.co    这两个网站
> 2. 如果是使用 vr2ay  ss  类型 则没有以上问题，都能顺利打开网页，现在如果要使用trojan 打开以上两个网站，只能使用静态解析 将网站域名与IP绑定，才能顺利打开。。请L大雕  排查一下原因。
> 3. ssr v2ray  的白名单  是不是 跟 trojan  用的不是一套地址列表？

btbtt.co NXDOMAIN 他们家已经变为 http://www.88btbtt.com/

samloya, 2020-02-10:
> 一样的

如果是用的同一套地址列表，那为什么同一个服务器，v2ray 能打开， trojan 不行，但是如果我用trojan 的电脑 客户端  就没有这种情况，是分流造成的问题么？

jargion, 2020-02-12:
我也遇到这个问题

closed 2020-02-19 by coolsnowwolf