coolsnowwolf/lede #2956: uhttpd无法启用https

CodeRhm, 2020-02-05:
编译的固件中已经选中了luci-ssl-openssl，证书使用的是阿里云的免费证书，选择的证书文件为apache的key及public.crt，也尝试使用了other中的证书并转换为der编码，但是还是无法启用，现象为访问一直停留在建立链接的过程，对应端口（443）似乎也没有开始，查了官方文档似乎连包用的都不一样，也并没有发现uhttpd的日志怎么查看，哪位大佬知道怎么解决麻烦告知下，万分感谢！

hcym, 2020-02-05:
大雕的默认值是0，改成1应该可以，没事用什么s，不怕麻烦

CodeRhm, 2020-02-05:
@

> 大雕的默认值是0，改成1应该可以，没事用什么s，不怕麻烦

麻烦问下是uhttpd的配置文件么？哪个值？
主要是想外网偶尔访问，http明文有点慌

trackycn, 2020-02-05:
> @
> 
> > 大雕的默认值是0，改成1应该可以，没事用什么s，不怕麻烦
> 
> 麻烦问下是uhttpd的配置文件么？哪个值？
> 主要是想外网偶尔访问，http明文有点慌

/etc/config/uhttpd
或者安装 luci-app-uhttpd 可以在UI里编辑

另外, 运营商是封80 8080 443 这些端口的, 你最好做个映射

CodeRhm, 2020-02-05:
映射已经做过了，以前用的别人编译的版本没有问题，自己编译的目前有问题，配置文件是手动改的，也装过面板改过，配置如下:
GNU nano 3.1                        /etc/config/uhttpd                                  

config uhttpd 'main'
        option home '/www'
        option max_requests '3'
        option max_connections '100'
        option cert '/etc/uhttpd.crt'
        option key '/etc/uhttpd.key'
        option cgi_prefix '/cgi-bin'
        list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
        option script_timeout '60'
        option network_timeout '30'
        option tcp_keepalive '1'
        option ubus_prefix '/ubus'
        list index_page 'cgi-bin/luci'
        list listen_http '0.0.0.0:80'
        list listen_http '[::]:80'
        list listen_http '0.0.0.0:443'
        list listen_http '[::]:443'
        option rfc1918_filter '0'
        option redirect_https '1'
翻了文档和帖子，看起来应该没错，能麻烦大佬看下有问题么？

CodeRhm, 2020-02-05:
@trackycn 映射已经做过了，以前用的别人编译的版本没有问题，自己编译的目前有问题，配置文件是手动改的，也装过面板改过，配置如下:
GNU nano 3.1 /etc/config/uhttpd

config uhttpd 'main'
option home '/www'
option max_requests '3'
option max_connections '100'
option cert '/etc/uhttpd.crt'
option key '/etc/uhttpd.key'
option cgi_prefix '/cgi-bin'
list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
option script_timeout '60'
option network_timeout '30'
option tcp_keepalive '1'
option ubus_prefix '/ubus'
list index_page 'cgi-bin/luci'
list listen_http '0.0.0.0:80'
list listen_http '[::]:80'
list listen_http '0.0.0.0:443'
list listen_http '[::]:443'
option rfc1918_filter '0'
option redirect_https '1'
翻了文档和帖子，看起来应该没错，能麻烦帮忙看下有问题么？

slitilanber, 2020-02-05:
list listen_http '0.0.0.0:80'
        list listen_http '[::]:80'
        list listen_https '0.0.0.0:端口'
        list listen_https '[::]:端口'
        option redirect_https '0'
        option home '/www'
        option rfc1918_filter '1'

这个配置: 
内网: 192.168.1.1，且只能IP, 不能域名
外网: https://域名:端口

trackycn, 2020-02-05:
> @trackycn 映射已经做过了，以前用的别人编译的版本没有问题，自己编译的目前有问题，配置文件是手动改的，也装过面板改过，配置如下:
> GNU nano 3.1 /etc/config/uhttpd
>
> config uhttpd 'main'
> option home '/www'
> option max_requests '3'
> option max_connections '100' [...]

配置文件不对, 把
`list listen_http '0.0.0.0:443'
list listen_http '[::]:443'`
改成
`	list listen_https '0.0.0.0:443'
	list listen_https '[::]:443'`

CodeRhm, 2020-02-05:
@trackycn 这么低级的错误:joy:
感谢提醒，可以close了:sweat_smile:

closed 2020-02-05 by CodeRhm