coolsnowwolf/lede #2957: luci-app-nfs #2908 似乎无法工作

Aes64X, 2020-02-05:
https://github.com/coolsnowwolf/lede/pull/2908
之前一直是手动修改 exports
尝试了一下，似乎是汉化文件夹名称有误，汉化是失效的，我改为 zh-cn 后汉化正常
他也是编辑 exports 文件，但填入的任何路径、参数、允许的IP什么的都无效，保存应用后也没发现exports文件有任何的变化。
如果不是个案，建议移除。

hxlls, 2020-02-05:
我也发现了，好像etc下没有exports 文件，但是在etc/config/下有个nfs文件，这个文件在luci 下修改生效，但是nfs实际没有生效

hxlls, 2020-02-05:
我安装 nfs 发现有提示错误：
root@OpenWrt:~# opkg install nfs-kernel-server
Installing nfs-kernel-server (2.4.1-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/nfs-kernel-server_2.4.1-1_x86_64.ipk
Multiple packages (librt and librt) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (librt and librt) providing same name marked HOLD or PREFER. Using latest.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for nfs-kernel-server:
 *      kernel (= 4.19.101-1-f2c57a4079c1d2a25ef373f71a66a4de)
 * opkg_install_cmd: Cannot install package nfs-kernel-server.

内核版本不够高？

hxlls, 2020-02-05:
好像找到问题了，那个luci-app-nfs 勾选时候，好像不会自动勾选相关的内核依赖，也就是只有壳，没程序

Aes64X, 2020-02-06:
> 好像找到问题了，那个luci-app-nfs 勾选时候，好像不会自动勾选相关的内核依赖，也就是只有壳，没程序

于依赖无关（前提是你有依赖才能用，它这个不检测依赖），手动编辑exports文件能生效就不缺依赖，它也是编辑这个文件达到控制NFS共享的。

hxlls, 2020-02-06:
已经解决
修改了部分代码，可以生效了，那个luci 的代码有错误，造成无法生效，但是还是有点小bug ,只能添加一行配置，没法添加两行，要手动选择 nfs-kernel-server一起编译

AmadeusGhost, 2020-02-13:
这个提交有一些错误，```zh_Hans```是官方luci的写法，依赖也没加上，还有一些文件的权限不对
luci-app-nfs我不知道是干啥的，github上有人写了，不知道能不能用
```
cd package
git clone https://github.com/simonsmh/luci-app-nfs.git
```
这个源码有些地方写的是```/mnt/sda1```，可能需要改成```/mnt```

hxlls, 2020-02-13:
> 这个提交有一些错误，`zh_Hans`是官方luci的写法，依赖也没加上，还有一些文件的权限不对
> luci-app-nfs我不知道是干啥的，github上有人写了，不知道能不能用
> 
> ```
> cd package
> git clone https://github.com/simonsmh/luci-app-nfs.git
> ```
> 
> 这个源码有些地方写的是`/mnt/sda1`，可能需要改成`/mnt`

不单只是这里，控制的文件也有错误
[image]
我看了你提到这个源码的，也是错误的，这个我倒是修改了nfs server 能用了，但是只有第一行配置生效，所以就没提交，还有客户端没测试，最近沉迷游戏，迟点再研究下

springsunx, 2020-02-15:
我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。
我测试过了多行配置，没有问题。
[image]
[luci-app-nfs.zip](https://github.com/coolsnowwolf/lede/files/4207729/luci-app-nfs.zip)
这是添加依赖后的源码，欢迎大家测试。

springsunx, 2020-02-15:
> 已经解决
> 修改了部分代码，可以生效了，那个luci 的代码有错误，造成无法生效，但是还是有点小bug ,只能添加一行配置，没法添加两行，要手动选择 nfs-kernel-server一起编译

有空可以再测试一下，我上传了新的附件。

AmadeusGhost, 2020-02-15:
> 我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。

”内核变动经常导致编译无法通过“是什么意思？有些依赖内核的就更新源码一起编译

springsunx, 2020-02-15:
> > 我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。
> 
> ”内核变动经常导致编译无法通过“是什么意思？有些依赖内核的就更新源码一起编译
> luci的话这边还是旧版的，没更新，有些写法不一样

官方代码更新比较快，有时候可能测试不完全，会导致编译无法通过。

hxlls, 2020-02-15:
> > > 我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。
> > 
> > 
> > ”内核变动经常导致编译无法通过“是什么意思？有些依赖内核的就更新源码一起编译
> > luci的话这边还是旧版的，没更新，有些写法不一样
> 
> 官方代码更新比较快，有时候可能测试不完全，会导致编译无法通过。
[image]
你确定这个源码没错误？控制文件好像就没对，不是应该是/etc/exports?

springsunx, 2020-02-15:
> > > > 我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。
> > > 
> > > 
> > > ”内核变动经常导致编译无法通过“是什么意思？有些依赖内核的就更新源码一起编译
> > > luci的话这边还是旧版的，没更新，有些写法不一样
> > 
> > 
> > 官方代码更新比较快，有时候可能测试不完全，会导致编译无法通过。
> > [image]
> > 你确定这个源码没错误？控制文件好像就没对，不是应该是/etc/exports?

没有问题，配置写在nfs里，动态添加到exports的。

hxlls, 2020-02-15:
> > > > > 我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。
> > > > 
> > > > 
> > > > ”内核变动经常导致编译无法通过“是什么意思？有些依赖内核的就更新源码一起编译
> > > > luci的话这边还是旧版的，没更新，有些写法不一样
> > > 
> > > 
> > > 官方代码更新比较快，有时候可能测试不完全，会导致编译无法通过。
> > > [image]
> > > 你确定这个源码没错误？控制文件好像就没对，不是应该是/etc/exports?
> 
> 没有问题，配置写在nfs里，动态添加到exports的。

已经测试，看图，这种情况下，这个划线的两个没有生效
[image]

hxlls, 2020-02-15:
还存在一个问题，依赖的NFS 服务内核可能是nfs 源码问题/etc/init.d/nfs 编译出来默认权限不够，是666，需要手动改成775，才能正常工作

hxlls, 2020-02-15:
找到原因了
root@OpenWrt:~# /etc/init.d/nfsd restart
root@OpenWrt:~# /etc/init.d/nfs restart
这两个启动有点问题

springsunx, 2020-02-16:
> > > > > > 我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。
> > > > > 
> > > > > 
> > > > > ”内核变动经常导致编译无法通过“是什么意思？有些依赖内核的就更新源码一起编译
> > > > > luci的话这边还是旧版的，没更新，有些写法不一样
> > > > 
> > > > 
> > > > 官方代码更新比较快，有时候可能测试不完全，会导致编译无法通过。
> > > > [image]
> > > > 你确定这个源码没错误？控制文件好像就没对，不是应该是/etc/exports?
> > 
> > 
> > 没有问题，配置写在nfs里，动态添加到exports的。
> 
> 已经测试，看图，这种情况下，这个划线的两个没有生效
> [image]

第一个，首先保证你的路径没问题，一般硬盘挂载都在/mnt里，不会直接在根目录下/sda3这样。
第二个，你要确保你的服务没有问题，正常访问且你的路径是对的。在提供服务的机器上运行exportfs查看。

springsunx, 2020-02-16:
> 找到原因了
> root@OpenWrt:~# /etc/init.d/nfsd restart root@OpenWrt:~# /etc/init.d/nfs restart
> 这两个启动有点问题

不明白这个什么意思，nfs这个不能正常运行吗？还是什么问题？nfsd是系统的服务。

springsunx, 2020-02-16:
> 还存在一个问题，依赖的NFS 服务内核可能是nfs 源码问题/etc/init.d/nfs 编译出来默认权限不够，是666，需要手动改成775，才能正常工作
我这里编译没有问题，会自动修改权限，应该和编译过程有关系。你可以在解压完成后先修改权限再编译，这样可以暂时解决问题。

closed 2020-02-19 by coolsnowwolf

loungee, 2020-02-26:
> 我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。
> 我测试过了多行配置，没有问题。
> [image]
> [luci-app-nfs.zip](https://github.com/coolsnowwolf/lede/files/4207729/luci-app-nfs.zip)
> 这是添加依赖后的源码，欢迎大家测试。

这个代码默认状态下可以编译成功，包依赖也是全的，但实际没用，因为新增的nfs脚本没有执行权限，只要chmod +x /etc/init.d/nfs就行了，请修正

hxlls, 2020-02-26:
> > 我没有添加依赖是因为，内核变动经常导致编译无法通过。luci我在官方版本上使用没有问题。当添加配置完成后，用exportfs命令检测是否运行成功。
> > 我测试过了多行配置，没有问题。
> > [image]
> > [luci-app-nfs.zip](https://github.com/coolsnowwolf/lede/files/4207729/luci-app-nfs.zip)
> > 这是添加依赖后的源码，欢迎大家测试。
> 
> 这个代码默认状态下可以编译成功，包依赖也是全的，但实际没用，因为新增的nfs脚本没有执行权限，只要chmod +x /etc/init.d/nfs就行了，请修正

编译时候添加一条到启动执行里面去就行了，但是还是又BUG，这个NFS 共享出去，只能生效一条，如果是nfs共享出去2个位置，后面的就无法生效，倒是NFS客户端 挂载可以多条，但是必须写一条参数应用保存一次

loungee, 2020-02-27:
你的/sda3是错误的目录，和人家的程序没关系，正常应该是/mnt/sda3，只要你的exports文件是正确就和这个luci没关系，这段源码纯粹就是增加包依赖和改写/etc/exports而已

loungee, 2020-02-27:
> 编译时候添加一条到启动执行里面去就行了，但是还是又BUG，这个NFS 共享出去，只能生效一条，如果是nfs共享出去2个位置，后面的就无法生效，倒是NFS客户端 挂载可以多条，但是必须写一条参数应用保存一次

你的/sda3是错误的目录，和人家的程序没关系，正常应该是/mnt/sda3，只要你的exports文件是正确就和这个luci没关系，这段源码纯粹就是增加包依赖和改写/etc/exports而已

hxlls, 2020-02-27:
> 你的/sda3是错误的目录，和人家的程序没关系，正常应该是/mnt/sda3，只要你的exports文件是正确就和这个luci没关系，这段源码纯粹就是增加包依赖和改写/etc/exports而已

exports只是服务端配置文件，和nfs 客户端的管理是没关系，我说的luci 的bug 还提到了一个nfs 挂载远程，一次添加多条信息后保存应用，实际生效才有一条

[image]
看到生成的配置信息，里面是10行，
[image]
也就是不能同时一次生成多行配置信息并且写入