coolsnowwolf/lede #2967: 编译出错

EstFoxs, 2020-02-06:
make[4]: Entering directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04'
make  all-recursive
make[5]: Entering directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04'
Making all in grub-core/lib/gnulib
make[6]: Entering directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make  all-recursive
make[7]: Entering directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[8]: Entering directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
depbase=`echo argp-fmtstream.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I. -I../../..   -I/home/bbq/lede/staging_dir/host/include -I/home/bbq/lede/staging_dir/hostpkg/include -I/home/bbq/lede/staging_dir/target-x86_64_musl/host/include -D_FILE_OFFSET_BITS=64  -O2 -I/home/bbq/lede/staging_dir/host/include -I/home/bbq/lede/staging_dir/hostpkg/include -I/home/bbq/lede/staging_dir/target-x86_64_musl/host/include -MT argp-fmtstream.o -MD -MP -MF $depbase.Tpo -c -o argp-fmtstream.o argp-fmtstream.c &&\
mv -f $depbase.Tpo $depbase.Po
In file included from argp-fmtstream.c:31:
./wchar.h:593:6: error: #if with no expression
  593 | # if
      |      ^
make[8]: *** [Makefile:1640: argp-fmtstream.o] Error 1
make[8]: Leaving directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[7]: *** [Makefile:1662: all-recursive] Error 1
make[7]: Leaving directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[6]: *** [Makefile:1461: all] Error 2
make[6]: Leaving directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[5]: *** [Makefile:11290: all-recursive] Error 1
make[5]: Leaving directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04'
make[4]: *** [Makefile:3329: all] Error 2
make[4]: Leaving directory '/home/bbq/lede/build_dir/hostpkg/grub-2.04'
make[3]: *** [Makefile:32: /home/bbq/lede/build_dir/hostpkg/grub-2.04/.built] Error 2
make[3]: Leaving directory '/home/bbq/lede/package/boot/grub2/grub2'
time: package/boot/grub2/grub2/host-compile#0.25#0.06#0.32
make[2]: *** [package/Makefile:107: package/boot/grub2/grub2/host/compile] Error 2
make[2]: Leaving directory '/home/bbq/lede'
make[1]: *** [package/Makefile:103: /home/bbq/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bbq/lede'
make: *** [/home/bbq/lede/include/toplevel.mk:218: world] Error 2
bbq@bbq:~/lede$

lloyd18, 2020-02-06:
/home/lan/lede/scripts/ipkg-build -c -o 0 -g 0 /home/lan/lede/build_dir/target-x86_64_musl/pcre-8.41/ipkg-x86_64/libpcre /home/lan/lede/bin/packages/x86_64/packages
Packaged contents of /home/lan/lede/build_dir/target-x86_64_musl/pcre-8.41/ipkg-x86_64/libpcre into /home/lan/lede/bin/packages/x86_64/packages/libpcre_8.41-2_x86_64.ipk
echo "libpcre" >> /home/lan/lede/staging_dir/target-x86_64_musl/pkginfo/pcre.default.install
rm -rf /home/lan/lede/tmp/stage-pcre
mkdir -p /home/lan/lede/tmp/stage-pcre/host /home/lan/lede/staging_dir/target-x86_64_musl/packages /home/lan/lede/staging_dir/host/packages
install -d -m0755 /home/lan/lede/tmp/stage-pcre/usr/bin
install -m0755 /home/lan/lede/build_dir/target-x86_64_musl/pcre-8.41/ipkg-install/usr/bin/pcre-config /home/lan/lede/tmp/stage-pcre/usr/bin/
install -d -m0755 /home/lan/lede/tmp/stage-pcre/host/bin
ln -sf /home/lan/lede/staging_dir/target-x86_64_musl/usr/bin/pcre-config /home/lan/lede/tmp/stage-pcre/host/bin
install -d -m0755 /home/lan/lede/tmp/stage-pcre/usr/include
cp -fpR /home/lan/lede/build_dir/target-x86_64_musl/pcre-8.41/ipkg-install/usr/include/pcre*.h /home/lan/lede/tmp/stage-pcre/usr/include/
install -d -m0755 /home/lan/lede/tmp/stage-pcre/usr/lib
cp -fpR /home/lan/lede/build_dir/target-x86_64_musl/pcre-8.41/ipkg-install/usr/lib/libpcre*.{a,so*} /home/lan/lede/tmp/stage-pcre/usr/lib/
install -d -m0755 /home/lan/lede/tmp/stage-pcre/usr/lib/pkgconfig
cp -fpR /home/lan/lede/build_dir/target-x86_64_musl/pcre-8.41/ipkg-install/usr/lib/pkgconfig/libpcre*.pc /home/lan/lede/tmp/stage-pcre/usr/lib/pkgconfig/
find /home/lan/lede/tmp/stage-pcre -name '*.la' | xargs -r rm -f; 
if [ -f /home/lan/lede/staging_dir/target-x86_64_musl/packages/pcre.list ]; then /home/lan/lede/scripts/clean-package.sh "/home/lan/lede/staging_dir/target-x86_64_musl/packages/pcre.list" "/home/lan/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/lan/lede/tmp/stage-pcre ]; then (cd /home/lan/lede/tmp/stage-pcre; find ./ > /home/lan/lede/tmp/stage-pcre.files); 	SHELL= flock /home/lan/lede/tmp/.staging-dir.flock -c ' mv /home/lan/lede/tmp/stage-pcre.files /home/lan/lede/staging_dir/target-x86_64_musl/packages/pcre.list && cp -fpR /home/lan/lede/tmp/stage-pcre/* /home/lan/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/lan/lede/tmp/stage-pcre
touch /home/lan/lede/staging_dir/target-x86_64_musl/stamp/.pcre_installed
make[3]: Leaving directory '/home/lan/lede/feeds/packages/libs/pcre'
time: package/feeds/packages/pcre/compile#47.36#5.74#62.13
make[2]: Leaving directory '/home/lan/lede'
make[1]: *** [package/Makefile:103: /home/lan/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lan/lede'
make: *** [/home/lan/lede/include/toplevel.mk:218：world] 错误 2

yoseg, 2020-02-06:
make  all-recursive
make[7]: Entering directory `/home/yoseg/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[8]: Entering directory `/home/yoseg/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
depbase=`echo argp-ba.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	gcc -DHAVE_CONFIG_H -I. -I../../..   -I/home/yoseg/lede/staging_dir/host/include -I/home/yoseg/lede/staging_dir/hostpkg/include -I/home/yoseg/lede/staging_dir/target-x86_64_musl/host/include -D_FILE_OFFSET_BITS=64  -O2 -I/home/yoseg/lede/staging_dir/host/include -I/home/yoseg/lede/staging_dir/hostpkg/include -I/home/yoseg/lede/staging_dir/target-x86_64_musl/host/include -MT argp-ba.o -MD -MP -MF $depbase.Tpo -c -o argp-ba.o argp-ba.c &&\
	mv -f $depbase.Tpo $depbase.Po
depbase=`echo argp-eexst.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	gcc -DHAVE_CONFIG_H -I. -I../../..   -I/home/yoseg/lede/staging_dir/host/include -I/home/yoseg/lede/staging_dir/hostpkg/include -I/home/yoseg/lede/staging_dir/target-x86_64_musl/host/include -D_FILE_OFFSET_BITS=64  -O2 -I/home/yoseg/lede/staging_dir/host/include -I/home/yoseg/lede/staging_dir/hostpkg/include -I/home/yoseg/lede/staging_dir/target-x86_64_musl/host/include -MT argp-eexst.o -MD -MP -MF $depbase.Tpo -c -o argp-eexst.o argp-eexst.c &&\
	mv -f $depbase.Tpo $depbase.Po
depbase=`echo argp-fmtstream.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	gcc -DHAVE_CONFIG_H -I. -I../../..   -I/home/yoseg/lede/staging_dir/host/include -I/home/yoseg/lede/staging_dir/hostpkg/include -I/home/yoseg/lede/staging_dir/target-x86_64_musl/host/include -D_FILE_OFFSET_BITS=64  -O2 -I/home/yoseg/lede/staging_dir/host/include -I/home/yoseg/lede/staging_dir/hostpkg/include -I/home/yoseg/lede/staging_dir/target-x86_64_musl/host/include -MT argp-fmtstream.o -MD -MP -MF $depbase.Tpo -c -o argp-fmtstream.o argp-fmtstream.c &&\
	mv -f $depbase.Tpo $depbase.Po
In file included from argp-fmtstream.c:31:0:
./wchar.h:593:6: error: #if with no expression
 # if 
      ^
make[8]: *** [argp-fmtstream.o] Error 1
make[8]: Leaving directory `/home/yoseg/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory `/home/yoseg/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[6]: *** [all] Error 2
make[6]: Leaving directory `/home/yoseg/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/yoseg/lede/build_dir/hostpkg/grub-2.04'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/yoseg/lede/build_dir/hostpkg/grub-2.04'
make[3]: *** [/home/yoseg/lede/build_dir/hostpkg/grub-2.04/.built] Error 2
make[3]: Leaving directory `/home/yoseg/lede/package/boot/grub2/grub2'
time: package/boot/grub2/grub2/host-compile#44.46#12.20#76.77
make[2]: *** [package/boot/grub2/grub2/host/compile] Error 2
make[2]: Leaving directory `/home/yoseg/lede'
make[1]: *** [/home/yoseg/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/yoseg/lede'
make: *** [world] 错误 2

happyzhang1995, 2020-02-06:
同grub编译错误

lyblitee, 2020-02-06:
哪里问题？

royalwei, 2020-02-06:
make[8]: Entering directory '/home/openwrt/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
depbase=`echo argp-fmtstream.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I. -I../../..   -I/home/openwrt/lede/staging_dir/host/include -I/home/openwrt/lede/staging_dir/hostpkg/include -I/home/openwrt/lede/staging_dir/target-x86_64_musl/host/include -D_FILE_OFFSET_BITS=64  -O2 -I/home/openwrt/lede/staging_dir/host/include -I/home/openwrt/lede/staging_dir/hostpkg/include -I/home/openwrt/lede/staging_dir/target-x86_64_musl/host/include -MT argp-fmtstream.o -MD -MP -MF $depbase.Tpo -c -o argp-fmtstream.o argp-fmtstream.c &&\
mv -f $depbase.Tpo $depbase.Po
In file included from argp-fmtstream.c:31:0:
./wchar.h:593:6: error: #if with no expression
 # if
      ^
Makefile:1640: recipe for target 'argp-fmtstream.o' failed
make[8]: *** [argp-fmtstream.o] Error 1
make[8]: Leaving directory '/home/openwrt/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:1662: recipe for target 'all-recursive' failed
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory '/home/openwrt/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:1461: recipe for target 'all' failed
make[6]: *** [all] Error 2
make[6]: Leaving directory '/home/openwrt/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:11290: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/openwrt/lede/build_dir/hostpkg/grub-2.04'
Makefile:3329: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/openwrt/lede/build_dir/hostpkg/grub-2.04'
Makefile:31: recipe for target '/home/openwrt/lede/build_dir/hostpkg/grub-2.04/.built' failed
make[3]: *** [/home/openwrt/lede/build_dir/hostpkg/grub-2.04/.built] Error 2
make[3]: Leaving directory '/home/openwrt/lede/package/boot/grub2/grub2'
time: package/boot/grub2/grub2/host-compile#0.21#0.05#0.25
package/Makefile:107: recipe for target 'package/boot/grub2/grub2/host/compile' failed
make[2]: *** [package/boot/grub2/grub2/host/compile] Error 2
make[2]: Leaving directory '/home/openwrt/lede'
package/Makefile:103: recipe for target '/home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
/home/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

royalwei, 2020-02-06:
我也是

jy3344511, 2020-02-06:
编译不了

Yrp, 2020-02-06:
The same issue.

pbwind, 2020-02-07:
我也是
mv -f $depbase.Tpo $depbase.Po
depbase=`echo argp-fmtstream.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I. -I../../..   -I/home/leo/lede/staging_dir/host/include -I/home/leo/lede/staging_dir/hostpkg/include -I/home/leo/lede/staging_dir/target-x86_64_musl/host/include -D_FILE_OFFSET_BITS=64  -O2 -I/home/leo/lede/staging_dir/host/include -I/home/leo/lede/staging_dir/hostpkg/include -I/home/leo/lede/staging_dir/target-x86_64_musl/host/include -MT argp-fmtstream.o -MD -MP -MF $depbase.Tpo -c -o argp-fmtstream.o argp-fmtstream.c &&\
mv -f $depbase.Tpo $depbase.Po
In file included from argp-fmtstream.c:31:0:
./wchar.h:593:6: error: #if with no expression
 # if
      ^
Makefile:1648: recipe for target 'argp-fmtstream.o' failed
make[8]: *** [argp-fmtstream.o] Error 1
make[8]: Leaving directory '/home/leo/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:1670: recipe for target 'all-recursive' failed
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory '/home/leo/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:1469: recipe for target 'all' failed
make[6]: *** [all] Error 2
make[6]: Leaving directory '/home/leo/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:11298: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/leo/lede/build_dir/hostpkg/grub-2.04'
Makefile:3337: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/leo/lede/build_dir/hostpkg/grub-2.04'
Makefile:31: recipe for target '/home/leo/lede/build_dir/hostpkg/grub-2.04/.built' failed
make[3]: *** [/home/leo/lede/build_dir/hostpkg/grub-2.04/.built] Error 2
make[3]: Leaving directory '/home/leo/lede/package/boot/grub2/grub2'
time: package/boot/grub2/grub2/host-compile#36.97#10.50#185.39
package/Makefile:107: recipe for target 'package/boot/grub2/grub2/host/compile' failed
make[2]: *** [package/boot/grub2/grub2/host/compile] Error 2
make[2]: Leaving directory '/home/leo/lede'
package/Makefile:103: recipe for target '/home/leo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/leo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/leo/lede'
/home/leo/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

hubanya, 2020-02-07:
The same problem.

EstFoxs, 2020-02-07:
1.Ubuntu14 通过编译
2.Ubuntu19 编译失败
都是server版
Generating index for package ./strongswan_5.6.3-2_x86_64.ipk
Generating index for package ./transmission-daemon-openssl_2.94-8_x86_64.ipk
Generating index for package ./transmission-web-control_2019-07-24-c27e3ba9-1_all.ipk
Generating index for package ./ttyd_1.4.0-1_x86_64.ipk
Generating index for package ./wget_1.20.3-2_x86_64.ipk
Generating index for package ./zerotier_1.4.6-1_x86_64.ipk
Signing package index...
make[2]: Leaving directory `/home/bbq/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory `/home/bbq/lede'
make[2]: Leaving directory `/home/bbq/lede'
make[1]: Leaving directory `/home/bbq/lede'
bbq@bbq:~/lede$ git pull
Already up-to-date.
bbq@bbq:~/lede$

liudongjia, 2020-02-08:
同样的问题

StoneForests, 2020-02-08:
一样的问题，有没有大神知道原因？

closed 2020-02-08 by EstFoxs