coolsnowwolf/lede #2968: ssrplus的“不走代理的域名”对gfw_list内的域名无效

Bnosora, 2020-02-06:
现象：
ssrplus使用绕过中国大陆ip模式
在不走代理的域名里添加
google.com
youtube.com
pornhub.com
akamai.com

保存应用之后依旧可以访问www.google.com、www.youtube.com和www.pornhub.com，证明这三条未生效。
whatismyip.akamai.com返回的ip是本地ip，证明这条生效了。（不添加这条时返回的是代理ip）

这个问题在gfw模式和全局模式下均存在。

coolsnowwolf, 2020-02-06:
这个是优先度的问题，同一个域名例如 *.google.com 只能命中一次 ipset

yifax, 2020-02-09:
> 这个是优先度的问题，同一个域名例如 *.google.com 只能命中一次 ipset

借楼咨询一下海外回国模式的逻辑。目前肉身在美，用海外模式无法正常观看爱奇艺和B站的版权视频，但是用全局模式可以。同时，两种模式都无法正常访问吾爱论坛，提示Clinet IP为服务器IP。尝试使用黑名单加入cache.iqiyi.com也不行，而全局模式+白名单则无法正常观看油管。目前比较懵逼。

Bnosora, 2020-02-10:
> 这个是优先度的问题，同一个域名例如 *.google.com 只能命中一次 ipset

请教一下，这个优先级受什么影响吗？试了几天也没明白为什么dnsmasq会把*.google.com的解析结果放进gfwlist这个集合而不是放进whitelist这个集合。明明ipset=/.google.com/whitelist和ipset=/.google.com/gfwlist这两条规则都在……

SharerMax, 2020-02-12:
参考 https://github.com/shadowsocks/luci-app-shadowsocks/issues/197

agutools, 2020-02-16:
提供一思路：曾经遇到过搜狗“被”走代理情况，计划任务中添加SS自动更新后执行
sed -i '/sogou/d' /etc/dnsmasq\.ssr/gfw_base.conf

closed 2020-02-19 by coolsnowwolf