coolsnowwolf/lede #2969: 编译出错

lyblitee, 2020-02-06:
tmp/.config-package.in:78587:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:78587:   symbol PACKAGE_dnsforwarder is selected by PACKAGE_luci-app-dnsforwarder
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:47302:   symbol PACKAGE_luci-app-dnsforwarder depends on PACKAGE_dnsforwarder
tmp/.config-package.in:84087:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:84087:   symbol PACKAGE_v2ray-plugin is selected by PACKAGE_v2ray-plugin
make[2]: Leaving directory '/home/lyblitee/lede'
make[3]: Leaving directory '/home/lyblitee/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.62#1.10#2.95
make[2]: Leaving directory '/home/lyblitee/lede'
package/Makefile:103: recipe for target '/home/lyblitee/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lyblitee/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lyblitee/lede'
Build failed - please re-run with -j1 to see the real error message
/home/lyblitee/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 1

请问是什么错误？请指教，谢谢

jy3344511, 2020-02-06:
一样编译出错。

mbax2zf2, 2020-02-06:
依赖选重了，v2ray和dnsforwader这些不能重复选，一个插件下面选了另一个插件下面就别去选（比如jo的选了lean的就别去选）

Yrp, 2020-02-07:
Build failed - please re-run with -j1 to see the real error message

lyblitee, 2020-02-07:
> Build failed - please re-run with -j1 to see the real error message

The same error

closed 2020-02-19 by coolsnowwolf

zhiwen1987, 2020-04-24:
我这里也是一样的错误 都不知道怎么解决

zhiwen1987, 2020-04-24:
tmp/.config-package.in:80967:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:80967:   symbol PACKAGE_dnsforwarder is selected by PACKAGE_luci-app-dnsforwarder
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:48508:   symbol PACKAGE_luci-app-dnsforwarder depends on PACKAGE_dnsforwarder
求解决方法。。