coolsnowwolf/lede #2975: 编译x86固件总是需要手动选择x86 instructing decoder seiftest

happyzhang1995, 2020-02-07:
[image]

lyblitee, 2020-02-07:
一样

Yrp, 2020-02-07:
The same issue.

pbwind, 2020-02-07:
同样的问题。 
如果输入yes 
最终编译错误。 
如果不输入，直接回车默认。 
也是编译错误。 错误提示如下：
...skipped <pbin.v2/libs/regex/build/gcc-4.8/release/threading-multi/visibility-hidden>libboost_regex.so.1.71.0 for lack of <pbin.v2/libs/regex/build/gcc-4.8/release/threading-multi/visibility-hidden>posix_api.o...
...skipped <p/home/leo/lede/staging_dir/hostpkg/lib>libboost_regex.so.1.71.0 for lack of <pbin.v2/libs/regex/build/gcc-4.8/release/threading-multi/visibility-hidden>libboost_regex.so.1.71.0...
...skipped <p/home/leo/lede/staging_dir/hostpkg/lib>libboost_regex.so for lack of <p/home/leo/lede/staging_dir/hostpkg/lib>libboost_regex.so.1.71.0...
...skipped <p/home/leo/lede/staging_dir/hostpkg/lib>libboost_regex.so.1 for lack of <p/home/leo/lede/staging_dir/hostpkg/lib>libboost_regex.so.1.71.0...
...skipped <p/home/leo/lede/staging_dir/hostpkg/lib>libboost_regex.so.1.71 for lack of <p/home/leo/lede/staging_dir/hostpkg/lib>libboost_regex.so.1.71.0...
...skipped <p/home/leo/lede/staging_dir/hostpkg/lib/cmake/boost_regex-1.71.0>libboost_regex-variant-shared.cmake for lack of <pbin.v2/libs/regex/build/gcc-4.8/release/threading-multi/visibility-hidden>libboost_regex.so.1.71.0...
...failed updating 12 targets...
...skipped 24 targets...
...updated 15393 targets...
make[3]: *** [/home/leo/lede/build_dir/hostpkg/boost_1_71_0/.built] Error 1
make[3]: Leaving directory `/home/leo/lede/feeds/packages/libs/boost'
time: package/feeds/packages/boost/host-compile#125.31#110.55#154.57
make[2]: *** [package/feeds/packages/boost/host/compile] Error 2
make[2]: Leaving directory `/home/leo/lede'
make[1]: *** [/home/leo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/leo/lede'
make: *** [world] 错误 2

lyblitee, 2020-02-07:
> 同样的问题。
> 如果输入yes
> 最终编译错误。
> 如果不输入，直接回车默认。
> 也是编译错误。 错误提示如下：
> ...skipped <pbin.v2/libs/regex/build/gcc-4.8/release/threading-multi/visibility-hidden>libboost_regex.so.1.71.0 for lack of [...]

我也是这个错误，不知道哪里问题

pbwind, 2020-02-07:
> 我也是这个错误，不知道哪里问题

我昨天用VPS ubuntu 18 编译，出现这个问题，今天换成了ubuntu14，还是这个问题。 
有点摸不着头脑。 
这个属于DL库的问题，还是脚本缺了一行代码呢？
有点蒙。

但是很奇怪。 我用github的 OpenWrt-CI 自动编译，每天都成功编译啊

AmadeusGhost, 2020-02-07:
这个可以在x86的config-4.19加一行：
```
CONFIG_X86_DECODER_SELFTEST=n
```
编译出错不是因为这个
我今天清了下硬盘编译x86，不选上qb一次成功

PR #2988 (2020-02-08): Fix #2975 需要手动选择x86 instructing decoder seiftest
Fix #2975
commit fcea97eb (2020-02-08): Fix #2975 需要手动选择x86 instructing decoder seiftest (#2988)

* Update config-4.19

Fix #2975

* Update config-4.19

Fix #2975
closed 2020-02-07 by happyzhang1995