coolsnowwolf/lede #2977: x

happyzhang1995, 2020-02-07:


happyzhang1995, 2020-02-07:
[image]
如图错误提示信息

happyzhang1995, 2020-02-07:
[image]
qt5已选

Yrp, 2020-02-07:
一样的问题

xynu617, 2020-02-07:
X86_32位同样编译qBittorrent出错
make[3]: *** [/home/runner/work/lede/lede/build_dir/target-i386_pentium4_musl/qBittorrent-release-4.1.9/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
package/Makefile:107: recipe for target 'package/lean/qBittorrent/compile' failed
make[2]: Leaving directory '/home/runner/work/lede/lede'
time: package/lean/qBittorrent/compile#0.49#0.20#0.68
package/Makefile:103: recipe for target '/home/runner/work/lede/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:216: recipe for target 'world' failed
make[2]: *** [package/lean/qBittorrent/compile] Error 2
make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make: *** [world] Error 2

YLaido, 2020-02-09:
Same here. Any solution?

coolsnowwolf, 2020-02-09:
fixed

closed 2020-02-10 by happyzhang1995