coolsnowwolf/lede #2979: 编译报错

dlj390, 2020-02-07:
make[3]: Entering directory '/home/dlj390/lede/package/boot/grub2/grub2'
CFLAGS="-O2 -I/home/dlj390/lede/staging_dir/host/include -I/home/dlj390/lede/staging_dir/hostpkg/include -I/home/dlj390/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/dlj390/lede/staging_dir/host/include -I/home/dlj390/lede/staging_dir/hostpkg/include -I/home/dlj390/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/dlj390/lede/staging_dir/host/lib -L/home/dlj390/lede/staging_dir/hostpkg/lib -L/home/dlj390/lede/staging_dir/target-x86_64_musl/host/lib" make  -C /home/dlj390/lede/build_dir/hostpkg/grub-2.04  TARGET_RANLIB=x86_64-openwrt-linux-musl-gcc-ranlib LIBLZMA=/home/dlj390/lede/staging_dir/host/lib/liblzma.a 
make[4]: Entering directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04'
make  all-recursive
make[5]: Entering directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04'
Making all in grub-core/lib/gnulib
make[6]: Entering directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make  all-recursive
make[7]: Entering directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[8]: Entering directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
depbase=`echo argp-fmtstream.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I. -I../../..   -I/home/dlj390/lede/staging_dir/host/include -I/home/dlj390/lede/staging_dir/hostpkg/include -I/home/dlj390/lede/staging_dir/target-x86_64_musl/host/include -D_FILE_OFFSET_BITS=64  -O2 -I/home/dlj390/lede/staging_dir/host/include -I/home/dlj390/lede/staging_dir/hostpkg/include -I/home/dlj390/lede/staging_dir/target-x86_64_musl/host/include -MT argp-fmtstream.o -MD -MP -MF $depbase.Tpo -c -o argp-fmtstream.o argp-fmtstream.c &&\
mv -f $depbase.Tpo $depbase.Po
In file included from argp-fmtstream.c:31:0:
./wchar.h:593:6: error: #if with no expression
 # if
      ^
Makefile:1640: recipe for target 'argp-fmtstream.o' failed
make[8]: *** [argp-fmtstream.o] Error 1
make[8]: Leaving directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:1662: recipe for target 'all-recursive' failed
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:1461: recipe for target 'all' failed
make[6]: *** [all] Error 2
make[6]: Leaving directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
Makefile:11290: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04'
Makefile:3329: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/dlj390/lede/build_dir/hostpkg/grub-2.04'
Makefile:31: recipe for target '/home/dlj390/lede/build_dir/hostpkg/grub-2.04/.built' failed
make[3]: *** [/home/dlj390/lede/build_dir/hostpkg/grub-2.04/.built] Error 2
make[3]: Leaving directory '/home/dlj390/lede/package/boot/grub2/grub2'
time: package/boot/grub2/grub2/host-compile#0.17#0.06#0.23
package/Makefile:107: recipe for target 'package/boot/grub2/grub2/host/compile' failed
make[2]: *** [package/boot/grub2/grub2/host/compile] Error 2
make[2]: Leaving directory '/home/dlj390/lede'
package/Makefile:103: recipe for target '/home/dlj390/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/dlj390/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dlj390/lede'
/home/dlj390/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

fgprodigal, 2020-02-07:
一样

liudongjia, 2020-02-08:
全都出问题了，等大神处理下了

coolsnowwolf, 2020-02-09:
fixed

closed 2020-02-09 by coolsnowwolf