coolsnowwolf/lede #2980: 今天重新clone后，编译多个固件都报错

cjxzdzh, 2020-02-07:
使用原先成功编译的.config文件，再次clone后，几个固件都不能成功编译了

AmadeusGhost, 2020-02-07:
你倒是说具体target啊

hue715, 2020-02-07:
我因为用make clean，但指令不删除之前编译toolchain，导致多个工具版本错误
make[4]: Entering directory '/home/tom/lede/build_dir/target-x86_64_musl/sqlite-autoconf-3230100'
 cd . && /bin/bash /home/tom/lede/build_dir/target-x86_64_musl/sqlite-autoconf-3230100/missing automake-1.15 --foreign
。。。
libtool: Version mismatch error.  This is libtool 2.4.6, but the
libtool: definition of this LT_INIT comes from libtool 2.4.
libtool: You should recreate aclocal.m4 with macros from libtool 2.4.6
libtool: and run autoconf again.
make[4]: *** [Makefile:538: sqlite3.lo] Error 63
make[4]: Leaving directory '/home/tom/lede/build_dir/target-x86_64_musl/sqlite-autoconf-3230100'
make[3]: *** [Makefile:127: /home/tom/lede/build_dir/target-x86_64_musl/sqlite-autoconf-3230100/.built] Error 2
make[3]: Leaving directory '/home/tom/lede/feeds/packages/libs/sqlite3'
time: package/feeds/packages/sqlite3/compile#10.44#7.19#23.25
make[2]: *** [package/Makefile:109: package/feeds/packages/sqlite3/compile] Error 2

blankhang, 2020-02-09:
旧的.config文件删除后 重新生成后再试试

closed 2020-02-19 by coolsnowwolf