coolsnowwolf/lede #2981: automake 也需要特定版本了么

hyird, 2020-02-07:
/home/hyird/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/coreutils-8.23/build-aux/missing: line 81: aclocal-1.14: command not found
WARNING: 'aclocal-1.14' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make[4]: *** [Makefile:5796: aclocal.m4] Error 127

要求1.14 我系统安装的是1.16

hyird, 2020-02-07:
/bin/sh ../libtool  --tag=CXX   --mode=compile arm-openwrt-linux-muslgnueabi-g++ -DHAVE_CONFIG_H -I. -I..  -I../lib -I../intl -I./includes -I./includes -DLOCALEDIR=\"/usr/share/locale\" -DHAVE_CONFIG_H -I/home/hyird/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I../deps/wslay/lib/includes -I../deps/wslay/lib/includes  -I/home/hyird/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I/home/hyird/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/include -I/home/hyird/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/include -I/home/hyird/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/include/fortify -I/home/hyird/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/include  -pipe -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/hyird/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/aria2-1.35.0:aria2-1.35.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto  -MT AbstractDiskWriter.lo -MD -MP -MF $depbase.Tpo -c -o AbstractDiskWriter.lo AbstractDiskWriter.cc &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: Version mismatch error.  This is libtool 2.4.6 Debian-2.4.6-11, but the
libtool: definition of this LT_INIT comes from libtool 2.4.6.42-b88ce.
libtool: You should recreate aclocal.m4 with macros from libtool 2.4.6 Debian-2.4.6-11
libtool: and run autoconf again.

hyird, 2020-02-07:
自己切换到1.14后就ok了

AmadeusGhost, 2020-02-07:
竞斗云正常编译

AmadeusGhost, 2020-02-07:
arch系好像会有这个问题

closed 2020-02-08 by hyird