coolsnowwolf/lede #2984: 请教游戏模式的逻辑

kid424, 2020-02-07:
先在issue搜索了“游戏”关键字，不过没有找到答案，所以发issue叨扰了。

请问“游戏模式UDP中继服务器”在设置以后，是按照以下哪种情况在调用呢？

1. 设置后内网所有设备UDP都按照选定的运行模式分流转发，添加在LAN访问控制中的增强游戏模式的IP对应的设备全局UDP都通过代理转发

2. 设置后内网所有设备UDP都按照选定的运行模式分流转发，添加在LAN访问控制中的增强游戏模式的IP对应的设备UDP按照国内外IP分流转发

3. 设置后只有添加在LAN访问控制中的增强游戏模式的IP对应的设备全局UDP都通过代理转发，别的设备不转发

4. 设置后只有添加在LAN访问控制中的增强游戏模式的IP对应的设备UDP按照国内外IP分流转发，别的设备不转发

Bnosora, 2020-02-07:
应该是你描述里的第二种

因为从iptables里看：

不论默认模式是gfw还是绕过中国大陆ip，直连设备（不走代理的lan ip）一定不走udp代理，全局设备一定走udp代理。（下文不考虑这两种模式）

默认gfw模式下，只有2种情况**会代理**udp：
1、增强游戏模式设备访问的目标ip不在中国大陆ip段内
2、（gfw模式的设备访问的）目标地址位于gfwlist内

默认绕过中国大陆ip模式下，只有3种情况**不会代理**udp：
1、（绕过中国大陆ip模式的设备访问的）目标地址在中国大陆ip段里
2、增强游戏模式设备访问的目标地址在“不走代理的wan ip“里
3、（绕过中国大陆ip模式的设备访问的）访问的目标地址在“不走代理的wan ip”里

另外，”不走/强制走代理的域名“里面的域名好像并不能影响udp转发？因为没在iptables的mangle里看到blacklist和whitelist这两张表……

我也在了解ssrplus的工作流程，如果有错希望各位大佬能指正一下。

kid424, 2020-02-08:
> 应该是你描述里的第二种
>
> 因为从iptables里看：
>
> 不论默认模式是gfw还是绕过中国大陆ip，直连设备（不走代理的lan ip）一定不走udp代理，全局设备一定走udp代理。（下文不考虑这两种模式）
>
> 默认gfw模式下，只有2种情况**会代理**udp：
> 1、增强游戏模式设备访问的目标ip不在中国大陆ip段内
> 2、（gfw模式的设备访问的）目标地址位于gfwlist内 [...]

感谢大佬细致的解答。我拿switch的连接测试做了下实验，初步估计是1或者2。结合大佬在iptables里看到的，应该就是2了。

测试的时候我主服务器和转发服务器的NAT类型都是NAT B，不开SS用我家网络直接测试是NAT A。

1.在大陆白名单模式下无论是否添加IP到增强游戏模式，测试结果都是B。但在没有添加到增强游戏模式下的时候，如果此时转发服务器换成NAT A类型的话，测试结果就会变成NAT A。

2.在gfwlist模式下，如果没有添加到增强游戏模式下，测试结果为NAT A。添加到增强游戏模式后，测试结果与转发服务器NAT类型一致。

先看1，即使没有添加到增强模式下，设备的NAT类型都是跟随转发服务器类型的。可以推测出只要设置了转发服务器，全部设备要么全局UDP被代理，要么按照一定的规律转发。

再看2，由于switch的连接测试服务器应该是在gfwlist以外，而在没有添加增强游戏模式的时候测试结果为A，则可以确定此时UDP是直连的。那么结合上面的结论，可以知道只要设置了转发服务器而没有添加增强游戏模式，则UDP一定是按照运行模式分流，可以排除我主楼的假设3和4.

而在测试2中添加增强游戏模式后，发现测试结果转发服务器NAT类型一致，则可以推测增强游戏模式内的设备的UDP转发分流规则不会跟随运行模式，要么全局，要么有一个白名单规则。

closed 2020-02-18 by kid424