coolsnowwolf/lede #2985: Debian 下给树莓派编译报错，/build_dir/hostpkg/go-1.13.6/.go_bootstrap.

FattyboyN, 2020-02-07:
( cd /home/wrt/work/lede/build_dir/hostpkg/go-1.13.6/src ; GOROOT_FINAL=/home/wrt/work/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/wrt/work/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/wrt/work/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap.
Building Go toolchain1 using /home/wrt/work/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap.
go build bootstrap/cmd/compile/internal/ssa: /home/wrt/work/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap/pkg/tool/linux_amd64/6g: signal: killed
go tool dist: FAILED: /home/wrt/work/lede/build_dir/hostpkg/go-1.13.6/.go_bootstrap/bin/go install -gcflags=-l -tags=math_big_pure_go compiler_bootstrap bootstrap/cmd/...: exit status 1
Makefile:271: recipe for target '/home/wrt/work/lede/build_dir/hostpkg/go-1.13.6/.built' failed
make[3]: *** [/home/wrt/work/lede/build_dir/hostpkg/go-1.13.6/.built] Error 2
make[3]: Leaving directory '/home/wrt/work/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#72.24#7.46#86.74
package/Makefile:107: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/wrt/work/lede'
package/Makefile:103: recipe for target '/home/wrt/work/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile' failed
make[1]: *** [/home/wrt/work/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wrt/work/lede'
/home/wrt/work/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

这是我选了哪个package导致的？
求教

SharerMax, 2020-02-07:
内存不足，清增加内存

coolsnowwolf, 2020-02-08:
编译机内存不足

closed 2020-02-10 by coolsnowwolf