coolsnowwolf/lede #2987: 编译出错

aotoo1989, 2020-02-07:
jie@LEDE:~/lede$ ./scripts/feeds install -a
Collecting package info: done
Collecting target info: done
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'kmod-fs-exfat', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/lean/cifsd-tools/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/ipt2socks/Makefile' has a build dependency on 'libuv', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/lean/libidn2/Makefile' has a dependency on 'libunistring', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-dnsforwarder/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a dependency on 'docker-ce', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'bash', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-filetransfer/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-flowoffload/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-guest-wifi/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-haproxy-tcp/Makefile' has a dependency on 'haproxy', which does not exist
WARNING: Makefile 'package/lean/luci-app-haproxy-tcp/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a dependency on 'strongswan-minimal', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a dependency on 'strongswan-mod-xauth-generic', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'nginx', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'zoneinfo-asia', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'php7', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'php7-fpm', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'php7-mod-curl', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'php7-mod-gd', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'php7-mod-iconv', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'php7-mod-json', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'php7-mod-mbstring', which does not exist
WARNING: Makefile 'package/lean/luci-app-kodexplorer/Makefile' has a dependency on 'php7-mod-opcache', which does not exist
[... 89 lines cut ...]
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/v2ray/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist
Installing all packages from feed packages.
Installing package 'acl' from packages
Installing package 'attr' from packages

[... cut ...]
time: package/lean/qBittorrent/compile#4.49#0.45#4.86
make[2]: *** [package/Makefile:109: package/lean/qBittorrent/compile] Error 2
make[2]: Leaving directory '/home/jie/lede'
make[1]: *** [package/Makefile:103: /home/jie/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jie/lede'
make: *** [/home/jie/lede/include/toplevel.mk:218：world] 错误 2
-----------------------------------------------------------------------------------------------------

默认config 无修改

sealszzz, 2020-02-07:
我也试过好多次都失败，最后qbittorrent那个不编译就成功了

lylxoooooo, 2020-02-07:
试了一下云编译，也是同样的错误

lloyd18, 2020-02-07:
qbittorrent 不能有

liudongjia, 2020-02-08:
编译好几遍都不行，qbittorrent确实有问题？

lesliedu, 2020-02-08:
> qbittorrent 不能有

怎么把qbittorrent去掉？

jy3344511, 2020-02-08:
> 编译好几遍都不行，qbittorrent确实有问题？

没有编译qbittorrent，一样编译错误。

sealszzz, 2020-02-08:
> > 编译好几遍都不行，qbittorrent确实有问题？
> 
> 没有编译qbittorrent，一样编译错误。
我是重新安装虚拟机编译的，可能make clean不能解决问题，重下代码试下吧

d975xbx, 2020-02-08:
[image]
L大的ACtions也报错。

RealKiro, 2020-02-08:
- 临时补丁试试[#2982 ](https://github.com/coolsnowwolf/lede/pull/2982#issue-372274783)
```
wget -O package/lean/qBittorrent/Makefile https://raw.githubusercontent.com/coolsnowwolf/lede/5001036ea61ec091dd990325d946dd7ff2ec6286/package/lean/qBittorrent/Makefile
```

Blueskylang, 2020-02-08:
我的config 里面也没选 qbittorrent 选项，但是编译的时候 出现qbittorrent 错误，把前两天编译通过的再编译一次都不行了，还是这个报错。

closed 2020-02-08 by coolsnowwolf