coolsnowwolf/lede #2993: UDPXY与MWAN冲突

zhangbw2016, 2020-02-08:
1、做iptv单线融合，如果不编译wan和luci-app-syncdial，udpxy可以正常转发组播并可以用http播放iptv
2、如果编译wan和luci-app-syncdial，udpxy不能正常转发组播，http播放iptv失败，但是用rtp协议可以正常播放
3、用openwrt官方固件安装udpxy、mwan和syncdial又是正常的

coolsnowwolf, 2020-02-08:
mark

zhangbw2016, 2020-02-09:
> mark

大佬抽空看看能不能修复，感谢就是多拨与UDPXY不兼容，只要编译了多拨插件，UDPXY就无法播放IPTV了

zhangbw2016, 2020-02-09:
> mark

大佬看看能不能修复，感谢

SoPudge, 2020-02-10:
我这边mwan3和udpxy同时使用正常……

zhangbw2016, 2020-02-10:
> 我这边mwan3和udpxy同时使用正常……

我只要选了syncdial，udpxy 就不行；你是不是没有选这个插件，只选了mwan3？然后自己手动创建的接口和负载均衡？

zhangbw2016, 2020-02-12:
> mark

 我将 mwan3的版本从您的2.6.18更新为2.8.2就解决这个问题了

ihuaer, 2020-02-14:
> mwan3

2.8.2 哪里获得源码

zhangbw2016, 2020-02-14:
> > mwan3
> 
> 2.8.2 哪里获得源码

openwrt 官方源

AmadeusGhost, 2020-02-18:
https://github.com/AmadeusGhost/packages/commit/30ac1d7c61a2f3c30dee5988e6f5b8122cb948fd
试一下这个同步的官方的更改会不会有问题

zhangbw2016, 2020-02-19:
> [AmadeusGhost/packages@30ac1d7](https://github.com/AmadeusGhost/packages/commit/30ac1d7c61a2f3c30dee5988e6f5b8122cb948fd)
> 试一下这个同步的官方的更改会不会有问题

naoki66, 2020-02-26:
mwan3: update to 2.8.2

closed 2020-03-06 by coolsnowwolf