coolsnowwolf/lede #2996: MT7621及MT7628，链路不定时瞬断3s（已排除物理链路问题）

Interestz, 2020-02-08:
首先对L大说一声抱歉，这个问题我认为可能去openwrt官方那里提issue，但是我英文水平有限，只好来这里麻烦L大以及各位开发者们。

前阵子打游戏发现时不时的断线漂移，后来查看op日志发现有不定时、但比较频繁的的link down和link up。
拓扑1
[image]
这是我家的网络拓扑，软路由拨号上网，下接3个AP（2台MT7621极路由B70,1台MT7628极1s），3个LAN口都分别有出现link down在link up的日志，但是网络接口中的LAN口始终是在线的。WAN口始终没出现过这个现象，软路由和AP也都没有开启负载均衡。
起初以为是物理线路问题，于是换了网线、水晶头等都没解决，后来以为是AP本身硬件有问题，于是换了个刷了op的AP还是一样，最后尝试把固件刷回原厂系统后运行一天（这个现象一个6小时内就会有好几次）竟没再出现这个问题（从这里看软路由应该是不可能存在硬件问题）。
我很是不解为什么会有这种问题，谷歌了一圈下来也没找到什么有营养的东西，于是又把AP刷了L大20年1月份的op、18.06.4官版的op以及最新19.01的官版op，仍旧会出现这个现象，但是在刷了潘多拉以后竟然没出现这个问题。

拓扑2
[image]
系统日志：[syslog1.txt](https://github.com/coolsnowwolf/lede/files/4173810/syslog1.txt)
内核日志：[kernellog.txt](https://github.com/coolsnowwolf/lede/files/4173814/kernellog.txt)

这个是我不死心，仍以为是软路由硬件的锅时，最后改变的拓扑。在运行了一段时间后，二级硬路由和上级软路由、下接AP相连的链路都出现了link down再link up的日志，至此，彻底排除嫌疑不大的软路由。

总结：在我的长测下，排除了物理链路存在问题的可能，排除了软路由、硬路由存在硬件问题的可能。在这种情况下，硬路由在op固件下会出现此类问题，但在原厂和潘多拉固件下却没有发现。我不太清楚这是不是我的配置导致的？还是openwrt本身存在这种问题？请L大及各位大佬们帮忙看看，以及看看自己op路由上的日志里是否有此类问题。

Interestz, 2020-02-08:
系统日志：
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00004cb0
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00004cb0
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] Memory: 252528K/262144K available (4982K kernel code, 257K rwdata, 1064K rodata, 212K init, 255K bss, 9616K reserved, 0K cma-reserved, 0K highmem)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.007842] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.074015] pid_max: default: 32768 minimum: 301
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.078829] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.085337] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.094295] Performance counters: mips/1004K PMU enabled, 2 32-bit counters available to each CPU, irq 17
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.104031] Hierarchical SRCU implementation.
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.109193] smp: Bringing up secondary CPUs ...
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.189787] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.189796] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.189807] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Feb  7 22:13:06 2020 kern.info kernel: [    1.189936] CPU1 revision is: 0001992f (MIPS 1004Kc)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.174098] Synchronize counters for CPU 1: done.
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.220221] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.220229] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.220237] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Feb  7 22:13:06 2020 kern.info kernel: [    1.220304] CPU2 revision is: 0001992f (MIPS 1004Kc)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.265046] Synchronize counters for CPU 2: done.
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.309827] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.309835] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.309843] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Feb  7 22:13:06 2020 kern.info kernel: [    1.309920] CPU3 revision is: 0001992f (MIPS 1004Kc)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.350230] Synchronize counters for CPU 3: done.
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.380100] smp: Brought up 1 node, 4 CPUs
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.388419] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.398212] futex hash table entries: 1024 (order: 3, 32768 bytes)
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.404594] pinctrl core: initialized pinctrl subsystem
Fri Feb  7 22:13:06 2020 kern.info kernel: [    0.411255] NET: Registered protocol family 16
Fri Feb  7 22:13:06 2020 kern.debug kernel: [    0.425624] FPU Affinity set after 11720 emulations
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    0.427205] pull PCIe RST: RALINK_RSTCTRL = 0
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    0.731944] release PCIe RST: RALINK_RSTCTRL = 7000000
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    0.736977] ***** Xtal 40MHz *****
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    0.740348] release PCIe RST: RALINK_RSTCTRL = 7000000
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    0.745457] Port 0 N_FTS = 1b105000
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    0.748899] Port 1 N_FTS = 1b105000
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    0.752359] Port 2 N_FTS = 1b102800
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.907735] PCIE2 no card, disable it(RST&CLK)
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.912083]  -> 21007f2
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.914501] PCIE0 enabled
Fri Feb  7 22:13:06 2020 kern.warn kernel: [    1.917109] PCIE1 enabled
Fri Feb  7 22:13:06 2020 kern.info kernel: [    1.919693] PCI host bridge /pcie@1e140000 ranges:
[... cut ...]
Sat Feb  8 07:48:41 2020 kern.info kernel: [34422.111584] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
Sat Feb  8 11:11:17 2020 kern.info kernel: [46577.841476] mtk_soc_eth 1e100000.ethernet eth0: port 4 link down
Sat Feb  8 11:11:20 2020 kern.info kernel: [46580.741929] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
Sat Feb  8 11:37:11 2020 kern.info kernel: [48132.115880] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
Sat Feb  8 11:37:14 2020 kern.info kernel: [48135.234430] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up

Interestz, 2020-02-08:
内核日志：
[    0.000000] Linux version 4.14.149 (a1@ubuntu) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r1519-ac6cb48)) #0 SMP Fri Oct 18 08:37:20 2019
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is HiWiFi HC5962
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 80627980, node_mem_map 81003000
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from start_kernel+0x9c/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu s27536 r8192 d21616 u57344
[    0.000000] pcpu-alloc: s27536 r8192 d21616 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00004cb0
[    0.000000] Readback ErrCtl register=00004cb0
[    0.000000] Memory: 252528K/262144K available (4982K kernel code, 257K rwdata, 1064K rodata, 212K init, 255K bss, 9616K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007842] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.074015] pid_max: default: 32768 minimum: 301
[    0.078829] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.085337] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.094295] Performance counters: mips/1004K PMU enabled, 2 32-bit counters available to each CPU, irq 17
[    0.104031] Hierarchical SRCU implementation.
[    0.109193] smp: Bringing up secondary CPUs ...
[    1.189787] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    1.189796] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    1.189807] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    1.189936] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.174098] Synchronize counters for CPU 1: done.
[    1.220221] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    1.220229] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    1.220237] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    1.220304] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.265046] Synchronize counters for CPU 2: done.
[    1.309827] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    1.309835] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    1.309843] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    1.309920] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.350230] Synchronize counters for CPU 3: done.
[    0.380100] smp: Brought up 1 node, 4 CPUs
[    0.388419] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.398212] futex hash table entries: 1024 (order: 3, 32768 bytes)
[    0.404594] pinctrl core: initialized pinctrl subsystem
[... 264 lines cut ...]
[ 5480.165904] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
[ 5805.083752] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[ 5808.106536] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[15124.598150] mtk_soc_eth 1e100000.ethernet eth0: port 4 link down
[15127.545346] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
[21387.952258] mtk_soc_eth 1e100000.ethernet eth0: port 4 link down
[21390.802381] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
[22341.086546] mtk_soc_eth 1e100000.ethernet eth0: port 4 link down
[22343.927072] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
[22963.316052] mtk_soc_eth 1e100000.ethernet eth0: port 4 link down
[22966.168758] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
[25125.866182] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[... cut ...]
[30669.527992] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[30672.490694] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[34419.026468] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[34422.111584] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[46577.841476] mtk_soc_eth 1e100000.ethernet eth0: port 4 link down
[46580.741929] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
[48132.115880] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[48135.234430] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up

Interestz, 2020-02-08:
实在离不开op的802.11r T_T

monsterzzzz, 2020-02-08:
这个问题也遇到过，感觉像兼容性问题，我这里用新3，只有一台PC这样，也是找不到原因

coolsnowwolf, 2020-02-08:
感觉像物理网卡和MTK 交换机的硬件兼容问题

Interestz, 2020-02-08:
> 感觉像物理网卡和MTK 交换机的硬件兼容问题

感谢L大费心解答！！
我感觉大概率也像是兼容性的问题，不太清楚MTK路由器的网卡驱动是否也包含在MTK芯片里闭源的？如果是的话，我目前能想到的唯一可能就是OP开源驱动的兼容性问题导致的MTK交换机运行不稳定。
我会尝试去openwrt官方反馈，期望能修复这个奇怪的问题。

Interestz, 2020-02-08:
> 这个问题也遇到过，感觉像兼容性问题，我这里用新3，只有一台PC这样，也是找不到原因

能详细说说您的情况吗？您的网络中包括主路由一共有几台刷了op的路由器？新3的各个网口只有连接PC的那个网口出现这种问题吗？

monsterzzzz, 2020-02-08:
> > 这个问题也遇到过，感觉像兼容性问题，我这里用新3，只有一台PC这样，也是找不到原因
> 
> 能详细说说您的情况吗？您的网络中包括主路由一共有几台刷了op的路由器？新3的各个网口只有连接PC的那个网口出现这种问题吗？

我都是拿来做主路由的，有问题的是两台台式机，都是intel CPU，技嘉主板板载网卡，一台是B85M-D3h，另一台稍微新点。B85M那台当时非常频繁，另一台也有这个问题，没那么频繁，感觉是PC刚开机时候出现比较多。接在上面其他的是有线AP，没这个问题。
刷过PADAVAN，没这个问题，所以感觉是兼容性问题。

Interestz, 2020-02-09:
> > > 这个问题也遇到过，感觉像兼容性问题，我这里用新3，只有一台PC这样，也是找不到原因
> > 
> > 
> > 能详细说说您的情况吗？您的网络中包括主路由一共有几台刷了op的路由器？新3的各个网口只有连接PC的那个网口出现这种问题吗？
> 
> 我都是拿来做主路由的，有问题的是两台台式机，都是intel CPU，技嘉主板板载网卡，一台是B85M-D3h，另一台稍微新点。B85M那台当时非常频繁，另一台也有这个问题，没那么频繁，感觉是PC刚开机时候出现比较多。接在上面其他的是有线AP，没这个问题。
> 刷过PADAVAN，没这个问题，所以感觉是兼容性问题。

您接的有线AP是openwrt系统吗？

monsterzzzz, 2020-02-09:
> > > > 这个问题也遇到过，感觉像兼容性问题，我这里用新3，只有一台PC这样，也是找不到原因
> > > 
> > > 
> > > 能详细说说您的情况吗？您的网络中包括主路由一共有几台刷了op的路由器？新3的各个网口只有连接PC的那个网口出现这种问题吗？
> > 
> > 
> > 我都是拿来做主路由的，有问题的是两台台式机，都是intel CPU，技嘉主板板载网卡，一台是B85M-D3h，另一台稍微新点。B85M那台当时非常频繁，另一台也有这个问题，没那么频繁，感觉是PC刚开机时候出现比较多。接在上面其他的是有线AP，没这个问题。
> > 刷过PADAVAN，没这个问题，所以感觉是兼容性问题。
> 
> 您接的有线AP是openwrt系统吗？

不是，是梅林

WinRose-C, 2020-02-09:
我用k2p，也有这个现象。路由交换机设置了3个wan口，两个lan口。其中一个lan口下接千兆交换机，另一lan口直接接移动的机顶盒（百兆）。千兆交换机下的设备无问题，机顶盒在开机一段时间频繁出现网络接口上线，下线问题，过一段时间后，断线频率降低。是不是速率协商方面的问题？

mcucore, 2020-02-10:
你们最新固件，K2P掉配置吗？不死选的斐讯模式

Interestz, 2020-02-10:
> 我用k2p，也有这个现象。路由交换机设置了3个wan口，两个lan口。其中一个lan口下接千兆交换机，另一lan口直接接移动的机顶盒（百兆）。千兆交换机下的设备无问题，机顶盒在开机一段时间频繁出现网络接口上线，下线问题，过一段时间后，断线频率降低。是不是速率协商方面的问题？

把机顶盒也接到交换机下试试？我怀疑是op的mtk网卡驱动兼容性太差，我重新调整，把软路由和有线PC全去掉，再看看运行情况

hcym, 2020-02-10:
兼容问题只能迁就，我的两个软路由，一个蜗牛只能用官方op，大雕的pdnsd经常无症状失效，一个5020u只能用大雕的，一点毛病没有。用官方op运行v2会卡cpu，真的没天理

Interestz, 2020-02-12:
> 感觉像物理网卡和MTK 交换机的硬件兼容问题

L大您好，我又有了一些新进展，烦请您帮忙看看。

我再次做了网络结构变更进行测试：
[image]
我将J1900软路由撤掉，只剩下3台硬路由（2台MT7621的HC5962，1台MT7628的HC5661A），主路由拨号，另外两台做AP，物理上皆为直连。在这样的拓扑下运行2天，主路由日志里只观察到port 2多次出现了前述的瞬断现象。（对应MT7621 AP日志里也有）

于是我又在主路由和MT7621 AP之间加了一台千兆交换机：
[image]
在这样的拓扑下运行一天多以后，竟没有再出现瞬断现象！

根据此前多次实验积累之下，我猜测2种可能：
1、物理网卡、OP驱动的兼容性问题。
最初怀疑的是这个问题，但是在2台相同硬件、相同系统的设备（soc为MT7621的HC5962）直连都出现瞬断现象的情况下，我认为兼容性问题的可能性相对较低。
2、OP系统与OP系统之间的千兆链路存在一定问题。
这个猜测的依据有3点。第一：本次实验中，在两台OP系统下的千兆路由之间加一台千兆交换机之后，再也没出现瞬断现象。第二：第一次实验中，J1900软路由（OP系统）为主路由，下接3个AP（均为OP系统）时，在千兆链路上频繁出现瞬断现象。第三：多次实验中，主路由的WAN口从来没瞬断过（200M宽带，千兆链路）。
在此再次附上第一次实验的拓扑图：

由此我认为第二种的可能性较大一些，并且我注意到在我电脑开机时，内核日志中也出现了和瞬断现象一样的日志，并且down和up的间隔均为2s+。我猜测我遇到的瞬断问题，会不会是OP系统与OP系统之间存在某种链路协商方面的问题？

我本人并非从事开发，对代码一知半解，暂不知道如何下手查看。还请L大能否帮忙分析排查一下？感激不尽！！

kongzheng333, 2020-02-12:
有点意思 同性相斥异性相吸？

Interestz, 2020-02-13:
> 有点意思 同性相斥异性相吸？

别搞玄学。。。您有2台千兆口的OP路由能帮忙测试一下吗？我想看看是MT7621单方面的问题还是OP系统本身的问题。

ch1993128, 2020-02-13:
19.几的是测试版。。。18.07是op稳定版，可以试着git 18.07源码，feeds和包改成lean的就完了。

Interestz, 2020-02-13:
> 19.几的是测试版。。。18.07是op稳定版，可以试着git 18.07源码，feeds和包改成lean的就完了。

尝试过L大19年10月21日的固件、L大20年1月份的op、OP官版的18.06.4的以及OP官版的19.07.1。都出现了上述问题。

ch1993128, 2020-02-14:
那不是很清楚了，我没有遇到上述问题

z826540272, 2020-02-18:
遇到同样问题
K2P和WR1200JS之间链路存在瞬断

adocwang, 2020-02-18:
我家里两个路由，一个K3做所有的事，另一个小米R3G做单纯的AP。然后用一段时间之后，大概一两天，做AP的R3G就会不断地断线重连，一小时一次的样子。后来仔细研究了一下发现，主路由和AP设置里面，Flow Offloading只要不同时开启就好了。现在把AP的关掉了，到目前为止超过48小时了，没什么问题。楼主有没有试过这个方案？

Interestz, 2020-02-19:
> 我家里两个路由，一个K3做所有的事，另一个小米R3G做单纯的AP，两个都刷的L大固件。然后用一段时间之后，大概一两天，做AP的R3G就会不断地断线重连，一小时一次的样子。后来仔细研究了一下发现，主路由和AP设置里面，Flow Offloading只要不同时开启就好了。现在把AP的关掉了，到目前为止超过48小时了，没什么问题。楼主有没有试过这个方案？

此前只试过关闭单侧的flow offloading，我立马修改测试一下！

Interestz, 2020-02-19:
> 遇到同样问题
> K2P和WR1200JS之间链路存在瞬断

都是op固件吗？尝试下把两端的flow offloading都关闭试试

z826540272, 2020-02-19:
一台高恪,一台K2P,
去年初期WR1200JS刚适配的时候我的电脑和WR有断LAN现象,后来再也没有了,看到这个issues发现果然两台路由器之间仍然存在瞬断,链路是
LAN TO LAN, WR用作交换机+AP+网关

>

adocwang, 2020-02-19:
> > 我家里两个路由，一个K3做所有的事，另一个小米R3G做单纯的AP，两个都刷的L大固件。然后用一段时间之后，大概一两天，做AP的R3G就会不断地断线重连，一小时一次的样子。后来仔细研究了一下发现，主路由和AP设置里面，Flow Offloading只要不同时开启就好了。现在把AP的关掉了，到目前为止超过48小时了，没什么问题。楼主有没有试过这个方案？
> 
> 此前试过关闭单侧的flow offloading，好像还是会发生，我再测试一下试试

不好意思，我记错了，我现在是，主路由和AP的Flow Offloading都关闭了。又过了12小时了，目前依然稳健。

Interestz, 2020-02-19:
> > > 我家里两个路由，一个K3做所有的事，另一个小米R3G做单纯的AP，两个都刷的L大固件。然后用一段时间之后，大概一两天，做AP的R3G就会不断地断线重连，一小时一次的样子。后来仔细研究了一下发现，主路由和AP设置里面，Flow Offloading只要不同时开启就好了。现在把AP的关掉了，到目前为止超过48小时了，没什么问题。楼主有没有试过这个方案？
> > 
> > 
> > 此前试过关闭单侧的flow offloading，好像还是会发生，我再测试一下试试
> 
> 不好意思，我记错了，我现在是，主路由和AP的Flow Offloading都关闭了。又过了12小时了，目前依然稳健。

今天把单侧关闭和双侧关闭都试了下，都还是会出现瞬断的现象。
我之前有一次大概有2天多的时间没出现瞬断，然后又一直瞬断的。感觉大概率还是OP和OP系统之间存在某些问题，目前如果想要正常使用，必须在OP和OP之间加一台交换机，尤其千兆OP之间，希望有一天OP能解决这个问题把。。。

Interestz, 2020-02-19:
> 一台高恪,一台K2P, 去年初期WR1200JS刚适配的时候我的电脑和WR有断LAN现象,后来再也没有了,看到这个issues发现果然两台路由器之间仍然存在瞬断,链路是 LAN TO LAN, WR用作交换机+AP+网关
> […](#)

目前来看还是OP和OP之间会瞬断，就是不清楚是否只有MTK存在这个问题

closed 2020-02-20 by coolsnowwolf

YorkPeng, 2020-03-22:
我也是K2P，主板也是技嘉的B85，具体表现就是LAN口每隔几个小时就会断开一次。用lean的pandorabox和其他固件就不会出现这种问题