coolsnowwolf/lede #2999: 编译提示grud2.04错误

hxlls, 2020-02-08:
环境，ubuntu14 lts  默认配置，提示如下
     gcc -DHAVE_CONFIG_H -I. -I../../..   -I/home/ling/lede/staging_dir/host/include -I/home/ling/lede/staging_dir/hostpkg/include -I/home/ling/lede/staging_dir/target-x86_64_musl/host/include -D_FILE_OFFSET_BITS=64  -O2 -I/home/ling/lede/staging_dir/host/include -I/home/ling/lede/staging_dir/hostpkg/include -I/home/ling/lede/staging_dir/target-x86_64_musl/host/include -MT argp-fmtstream.o -MD -MP -MF $depbase.Tpo -c -o argp-fmtstream.o argp-fmtstream.c &&\
        mv -f $depbase.Tpo $depbase.Po
In file included from argp-fmtstream.c:31:0:
./wchar.h:593:6: error: #if with no expression
 # if
      ^
make[8]: *** [argp-fmtstream.o] Error 1
make[8]: Leaving directory `/home/ling/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory `/home/ling/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[6]: *** [all] Error 2
make[6]: Leaving directory `/home/ling/lede/build_dir/hostpkg/grub-2.04/grub-core/lib/gnulib'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/ling/lede/build_dir/hostpkg/grub-2.04'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/ling/lede/build_dir/hostpkg/grub-2.04'
make[3]: *** [/home/ling/lede/build_dir/hostpkg/grub-2.04/.built] Error 2
make[3]: Leaving directory `/home/ling/lede/package/boot/grub2/grub2'
time: package/boot/grub2/grub2/host-compile#0.41#0.13#0.54
make[2]: *** [package/boot/grub2/grub2/host/compile] Error 2
make[2]: Leaving directory `/home/ling/lede'
make[1]: *** [/home/ling/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/ling/lede'
make: *** [world] Error 2
已经更新最新pull,feed也更新最新，尝试过make clean 等操作，无果，尝试返回到没跳内核版本的
git reset --hard 5540ae406dc9eda7afb620276cecb209b40f5284 后可以正常编译，能否修复下？

IsscaPan, 2020-02-08:
删除dl目录下的grub库，重新编译就好了

hxlls, 2020-02-08:
解决了，谢谢

closed 2020-02-09 by coolsnowwolf