coolsnowwolf/lede #3002: IPV6问题，建议升级odhcp6c 等源码

WinRose-C, 2020-02-08:
使用IPV6时，其中一个客户端一直在系统日志中产生如下日志
Sat Feb  8 18:56:13 2020 daemon.warn odhcpd[1828]: DHCPV6 REQUEST IA_NA from 0001000*****星号为隐去地址*****d on br-lan: not on-link
Sat Feb  8 18:56:18 2020 daemon.warn odhcpd[1828]: DHCPV6 REQUEST IA_NA from 0001000*****星号为隐去地址*****d on br-lan: not on-link
Sat Feb  8 18:56:23 2020 daemon.warn odhcpd[1828]: DHCPV6 REQUEST IA_NA from 0001000*****星号为隐去地址*****d on br-lan: not on-link
类似的日志一直会出现
在系统中更新odhcp6c与odhcpd-ipv6only（编译版本中为2018年版，更新为2019及2020年版本），日志恢复正常

hcym, 2020-02-08:
手动对焦没问题？我是没看过日志，一直用的正常

WinRose-C, 2020-02-08:
有没有从源码更新这两个组件的方法呢

AmadeusGhost, 2020-02-08:
https://github.com/AmadeusGhost/lede/commit/17479a4e5620465099d93a52404c2d6adb27cec5
来做测试
没问题了我在发pr

WinRose-C, 2020-02-08:
> [AmadeusGhost@5f5fe56](https://github.com/AmadeusGhost/lede/commit/5f5fe5635cc779ea43865006a56f4799572166ba)
> 来做测试，没问题了我再发pr

可以提示下大概是什么问题导致的吗？

AmadeusGhost, 2020-02-08:
> > [AmadeusGhost@5f5fe56](https://github.com/AmadeusGhost/lede/commit/5f5fe5635cc779ea43865006a56f4799572166ba)
> > 来做测试，没问题了我再发pr
> 
> 可以提示下大概是什么问题导致的吗？

不知道，官方的commit内容只有一行字：
```
e199804 dhcpv6: sanitize oro options
```
你试下我的更改有没有问题

WinRose-C, 2020-02-08:
> > > [AmadeusGhost@5f5fe56](https://github.com/AmadeusGhost/lede/commit/5f5fe5635cc779ea43865006a56f4799572166ba)
> > > 来做测试，没问题了我再发pr
> > 
> > 
> > 可以提示下大概是什么问题导致的吗？
> 
> 不知道，官方的commit内容只有这么几行字：
> 
> ```
> d2e247d odhcp6c: align further with RFC8415
> ce83a23 dhcpv6: avoid parsing unncessary IAs
> b079733 dhcpv6: set cnt to correct IOV enum
> 41494da dhcpv6: get rid of request_prefix
> f7437e4 dhcpv6: sanitize option request list
> e199804 dhcpv6: sanitize oro options
> ```
> 
> 你试下我的更改有没有问题

好的，现在没办法，家里领导在用网，不能断，明天编译后刷了反馈吧。谢谢。
你是直接把官方的源码拿过来的吗？odhcpd-ipv6only是不是就是odhcpd，我在lean的源中没有找到

AmadeusGhost, 2020-02-08:
> 好的，现在没办法，家里领导在用网，不能断，明天编译后刷了反馈吧。谢谢。
> 你是直接把官方的源码拿过来的吗？odhcpd-ipv6only是不是就是odhcpd，我在lean的源中没有找到

odhcpd-ipv6only的定义在odhcpd的makefile那，你不说我差点漏了改hhh

hyird, 2020-02-09:
我觉得  这个应该是 dhcpv6 请求不到ipv6地址而报错 你可以尝试 在lan 的ipv6设置 那个全部关掉试试
我猜测你是不是把ula前缀删掉 而你没有ipv6 
DHCPV6 REQUEST IA_NA from 000100011cf4f76ed0509958acf6 on br-lan: ok fde0:1ca1:b6e0::dce/128
你看这个 
以上纯属猜测

WinRose-C, 2020-02-09:
> 我觉得 这个应该是 dhcpv6 请求不到ipv6地址而报错 你可以尝试 在lan 的ipv6设置 那个全部关掉试试
> 我猜测你是不是把ula前缀删掉 而你没有ipv6
> DHCPV6 REQUEST IA_NA from 000100011cf4f76ed0509958acf6 on br-lan: ok fde0:1ca1:b6e0::dce/128
> 你看这个
> 以上纯属猜测

1.是有PV6的，lan内支持ipv6的设备（包含日志中有问题的设备）都获得了IPV6
2.lan中其他获得或没有获得（不支持）ipv6的设备都不会出现类似日志
3.确实删除了全局网络选项中的前缀
4.不动任何设置，只在软件包中更新了odhcp6c和odhcpd-ipv6only，日志中不再出现错误

WinRose-C, 2020-02-09:
> > 好的，现在没办法，家里领导在用网，不能断，明天编译后刷了反馈吧。谢谢。
> > 你是直接把官方的源码拿过来的吗？odhcpd-ipv6only是不是就是odhcpd，我在lean的源中没有找到
> 
> odhcpd-ipv6only的定义在odhcpd的makefile那，你不说我差点漏了改hhh

已测试，没有问题了。可以pull

closed 2020-02-09 by WinRose-C