coolsnowwolf/lede #3006: 希望增加2.5Gbps的USB网卡RTL8156驱动

jcxq520, 2020-02-08:
希望增加2.5Gbps的USB网卡RTL8156驱动

hxlls, 2020-02-08:
不是有么？好像用的是8152的驱动
Prompt: kmod-usb-net-rtl8152                                                                                                                   │
  │   Location:                                                                                                                                    │
  │     -> Kernel modules                                                                                                                          
  │       -> USB Support

coolsnowwolf, 2020-02-09:
已有

jcxq520, 2020-02-09:
[ 1316.867982] usb 4-1: USB disconnect, device number 2
[ 1320.854666] usb 4-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
root@OpenWrt:~# ls /lib/modules/`uname -r`/r815*
/lib/modules/4.19.88/r8152.ko
root@OpenWrt:~# modprobe r8152
r8152 is already loaded
root@OpenWrt:~# dmesg | grep r8152
[    9.127396] usbcore: registered new interface driver r8152
root@OpenWrt:~# lsusb
-ash: lsusb: not found
root@OpenWrt:~# 
如果是用的8152的驱动，固件默认就有，可是我USB插进去就是看不到新的eth

hyird, 2020-02-09:
> 已有

busybox的 lsusb和lspci 默认能加上么  还挺有用的

Her0R0cket, 2020-02-09:
8156用的应该就是8152的驱动模块

AmadeusGhost, 2020-02-09:
> busybox的 lsusb和lspci 默认能加上么 还挺有用的

Utilities的usbutils和pciutils选上就行

closed 2020-02-19 by coolsnowwolf

qingxuan111, 2020-04-12:
求救各位大神，我是威联通的的453bmini，买了2.5g的绿联的用这个芯片的usb网卡，在威联通上不能用，不过我看到有群晖上的，还有linux的源码的，有高手可以帮忙编译下吗。根本不会编程。买了两个网卡划了三百，不想浪费啊。谢谢高手了

qingxuan111, 2020-04-12:
> > busybox的 lsusb和lspci 默认能加上么 还挺有用的
> 
> Utilities的usbutils和pciutils选上就行
求救各位大神，我是威联通的的453bmini，买了2.5g的绿联的用这个芯片的usb网卡，在威联通上不能用，不过我看到有群晖上的，还有linux的源码的，有高手可以帮忙编译下吗。根本不会编程。买了两个网卡划了三百，不想浪费啊。谢谢高手了

qingxuan111, 2020-04-12:
> 8156用的应该就是8152的驱动模块
求救各位大神，我是威联通的的453bmini，买了2.5g的绿联的用这个芯片的usb网卡，在威联通上不能用，不过我看到有群晖上的，还有linux的源码的，有高手可以帮忙编译下吗。根本不会编程。买了两个网卡划了三百，不想浪费啊。谢谢高手了

sniper3209, 2020-07-18:
> [ 1316.867982] usb 4-1: USB disconnect, device number 2
> [ 1320.854666] usb 4-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
> root@OpenWrt:~# ls /lib/modules/`uname -r`/r815* [...]

请问你的问题解决了吗？我现在遇到和你一样的问题

alexliu269, 2020-07-26:
> 请问你的问题解决了吗？我现在遇到和你一样的问题

同问

stark131, 2020-11-20:
我昨天编译了BCM2711（树莓派4）的snapshot，里头的kmod-net-usb-r8152驱动好像还是旧版的，还是不支持RTL8156；而且RTL8153配合这个旧版驱动跑大流量时会出问题，请问可以确认下snapshot里头的驱动更新了吗？