coolsnowwolf/lede #3012: 请问用Mac可以编译吗？

McJayFeng, 2020-02-09:
我有一台Mac，请问可以编译吗？

zhangbw2016, 2020-02-09:
我用过mac编译 总是在编译fullcontent的时候报错：*** No rule to make target 'tools/objtool/objtool'

luckyyyyy, 2020-02-09:
mac编译比较麻烦 请参考openwrt官方手册
没有经验请勿尝试，并不是说跟着手册就可以没有问题。

https://openwrt.org/docs/guide-developer/buildroot.exigence.macosx

qlmyang-boop, 2020-02-09:
我用的Mac 连接 Google VPS，编译超快。

McJayFeng, 2020-02-09:
> 我用的Mac 连接 Google VPS，编译超快。

我在AWS编译了一下，编译了半天🤣

9l, 2020-02-10:
github action 多省心

ch1993128, 2020-02-12:
mac也可以装ubuntu

McJayFeng, 2020-02-12:
> mac编译比较麻烦 请参考openwrt官方手册
> 没有经验请勿尝试，并不是说跟着手册就可以没有问题。
> 
> https://openwrt.org/docs/guide-developer/buildroot.exigence.macosx

上次看了文档，几年前的文档了，感觉MacOS10.15不一定能用，昨天干脆装了个Ubuntu的虚拟机

McJayFeng, 2020-02-12:
> mac也可以装ubuntu

哈哈哈是的，昨天装好了~

zhaianan, 2020-02-12:
mac可以编译的

zhaianan, 2020-02-12:
https://p3terx.com/archives/compiling-openwrt-with-macos.html可以参考这个

McJayFeng, 2020-02-13:
> [https://p3terx.com/archives/compiling-openwrt-with-macos.html可以参考这个](https://p3terx.com/archives/compiling-openwrt-with-macos.html%E5%8F%AF%E4%BB%A5%E5%8F%82%E8%80%83%E8%BF%99%E4%B8%AA)

嗯嗯，多谢多谢~ 上次看了 https://openwrt.org/docs/guide-developer/buildroot.exigence.macosx 这个文档，最近的更新也是16年了，所以感觉会失败，于是就直接装了个Ubuntu的虚拟机

closed 2020-03-06 by coolsnowwolf