coolsnowwolf/lede #3017: luci配置端口转发不生效，点击保存&应用没有任何内容，配置文件中也没有相关配置

NanFour, 2020-02-09:
近日发现luci配置端口转发不生效，点击保存&应用界面没有任何内容，配置文件中也没有相关配置，系统日志中也没有相关错误信息，如上，感谢各位大佬的辛勤。

[image]

luckyyyyy, 2020-02-09:
ssh 进去 执行 ps 把内容发出来看一下

NanFour, 2020-02-09:
> ssh 进去 执行 ps 把内容发出来看一下

[image]
[image]
如上。

luckyyyyy, 2020-02-09:
哦 那和这个无关 我以为有和另外一个相同的问题

NanFour, 2020-02-09:
> 哦 那和这个无关 我以为有和另外一个相同的问题

刚又发现luci添加静态路由后，/etc/config/network生成了相关配置文件，但ip route list 中并没有生成相关路由条目，shell下用ip route add添加条目后倒是可以生效。

luckyyyyy, 2020-02-09:
> > 哦 那和这个无关 我以为有和另外一个相同的问题
> 
> 刚又发现luci添加静态路由后，/etc/config/network生成了相关配置文件，但ip route list 中并没有生成相关路由条目，shell下用ip route add添加条目后倒是可以生效。

因为在Lienol的固件 我发现有网易云restart导致的fw死锁，导致任何后续操作都不生效的问题，你这里我没看到。

NanFour, 2020-02-09:
> > > 哦 那和这个无关 我以为有和另外一个相同的问题
> > 
> > 
> > 刚又发现luci添加静态路由后，/etc/config/network生成了相关配置文件，但ip route list 中并没有生成相关路由条目，shell下用ip route add添加条目后倒是可以生效。
> 
> 因为在Lienol的固件 我发现有网易云restart导致的fw死锁，导致任何后续操作都不生效的问题，你这里我没看到。

谢谢你的回复。

coolsnowwolf, 2020-02-10:
点击 “添加”

NanFour, 2020-02-10:
> 点击 “添加”

🤣见笑了，操作问题。

closed 2020-02-10 by NanFour