coolsnowwolf/lede #3020: R20版本SSR无法订阅

YingJieKang, 2020-02-10:
反映个问题，不知道是不是都一样：编译了很多个X86的固件都是最新的R20的，SSR订阅那里填写了URL后点击更新URL更新服务器之后无反应，更新不出来节点，重新编译还是一样的问题

edmondfr, 2020-02-10:
填完订阅地址，先点击保存，再更新试一下

YingJieKang, 2020-02-10:
> 填完订阅地址，先点击保存，再更新试一下

试过了，还是不行，很奇怪，你没遇到过这个问题吗？固件可以正常编译出来，但是就是订阅这里出不来服务器

edmondfr, 2020-02-10:
> > 填完订阅地址，先点击保存，再更新试一下
> 
> 试过了，还是不行，很奇怪，你没遇到过这个问题吗？固件可以正常编译出来，但是就是订阅这里出不来服务器

最新的版本没试过，2月份的版本编译过几次，SSR翻不了，暂时回到2月1号的版本。

SharerMax, 2020-02-10:
1. 添加订阅地址
2. 保存
3. 点击 `更新订阅 URL 列表`
4. 点击 `更新所有订阅服务器节点`
5. 切到日志页，查看日志

YingJieKang, 2020-02-10:
> 1. 添加订阅地址
> 2. 保存
> 3. 点击 `更新订阅 URL 列表`
> 4. 点击 `更新所有订阅服务器节点`
> 5. 切到日志页，查看日志

2020-02-10 21:15:42 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:354: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:305: in function </usr/share/shadowsocksr/subscribe.lua:226>
	[C]: in function 'assert'
	/usr/share/shadowsocksr/subscribe.lua:356: in function </usr/share/shadowsocksr/subscribe.lua:354>
2020-02-10 21:15:42 stack traceback:
2020-02-10 21:15:42 /usr/share/shadowsocksr/subscribe.lua:305: node result is empty
显示这个

SharerMax, 2020-02-10:
订阅节点未返回有效数据，清检查订阅链接已经订阅返回的信息是否正确

jxph1123, 2020-02-10:
有可能你的订阅地址被墙了，手动增加一个代理，然后通过代理更新订阅试试。

YingJieKang, 2020-02-10:
> 订阅链接未返回有效数据，请检查订阅链接以及订阅返回的信息是否正确

信息都是正确的，用之前的版本没问题

RealKiro, 2020-02-10:
> 信息都是正确的，用之前的版本没问题

你终端命令覆盖一下试试，我之前好像有遇见这个问题https://github.com/coolsnowwolf/lede/pull/2960
```
wget -O /usr/share/shadowsocksr/subscribe.lua https://raw.githubusercontent.com/coolsnowwolf/lede/b9ee8de9d6907fedb5b6c51f407deb4e2b200f72/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua
```

closed 2020-02-11 by YingJieKang

nettiger2019, 2020-02-12:
请问解决了嘛？我也有一样的问题

bigmi777, 2020-03-23:
> > 信息都是正确的，用之前的版本没问题
> 
> 你终端命令覆盖一下试试，我之前好像有遇见这个问题#2960
> 
> ```
> wget -O /usr/share/shadowsocksr/subscribe.lua https://raw.githubusercontent.com/coolsnowwolf/lede/b9ee8de9d6907fedb5b6c51f407deb4e2b200f72/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua
> ```

这个方法可行，问题已解决！！！