coolsnowwolf/lede #3024: 编译好的固件无法安装其它插件

authon, 2020-02-10:
编译好的固件无法安装其它插件，或者winscp上传配置文件，提示scp: /etc/config/********: No space left on device，文件才2kb，请问该怎么设置呢？

打开ssr文件出现
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/shadowsocksr'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/shadowsocksr/client.lua:18: bad argument #1 to 'upper' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

这个问题是储存不足导致的吗

coolsnowwolf, 2020-02-10:
对

authon, 2020-02-10:
> 对

大神，我是PVE虚拟机里面装的OP，用的img2kvm转的系统，是编译的什么步骤错了吗

Aes64X, 2020-02-11:
> > 对
> 
> 大神，我是PVE虚拟机里面装的OP，用的img2kvm转的系统，是编译的什么步骤错了吗。
> 
> 或者说怎么设置更大的空余空间

你看的帖子有问题，PVE加载OP不需要 img2kvm 来转换，自带有转换命令 qm importdisk 。一般PVE上OP我都是用EXT4格式。

authon, 2020-02-11:
> > > 对
> > 
> > 
> > 大神，我是PVE虚拟机里面装的OP，用的img2kvm转的系统，是编译的什么步骤错了吗。
> > 或者说怎么设置更大的空余空间
> 
> 你看的帖子有问题，PVE加载OP不需要 img2kvm 来转换，自带有转换命令 qm importdisk 。一般PVE上OP我都是用EXT4格式。

好的昨天爬到这个命令了，qm importdisk命令出来的就是ext4了吗，这中加载方式不会提示储存不够吗？

然后今天发现可以在pve里面调整op系统的的大小，然后fdisk /dev/sda 加载空余硬盘，
然后系统——挂载点——添加
选择启用挂载点——UUID选择刚刚分好的磁盘——挂载点自定义为  作为外部overlay使用
发现有空间可以下载adgurdhome的组件了

我这两种操作，是qm importdisk生成的系统生效还是第二种方法生效的，同时操作的，不知道是哪个起作用

Aes64X, 2020-02-11:
> > > > 对
> > > 
> > > 
> > > 大神，我是PVE虚拟机里面装的OP，用的img2kvm转的系统，是编译的什么步骤错了吗。
> > > 或者说怎么设置更大的空余空间
> > 
> > 
> > 你看的帖子有问题，PVE加载OP不需要 img2kvm 来转换，自带有转换命令 qm importdisk 。一般PVE上OP我都是用EXT4格式。
> 
> 好的昨天爬到这个命令了，qm importdisk命令出来的就是ext4了吗，这中加载方式不会提示储存不够吗？
> 
> 然后今天发现可以在pve里面调整op系统的的大小为5GB，然后fdisk /dev/sda 加载空余硬盘，
> 然后系统——挂载点——添加
> 选择启用挂载点——UUID选择刚刚分好的磁盘——挂载点自定义为 作为外部overlay使用
> 发现有空间可以下载adgurdhome的组件了
> 
> 我这两种操作，是qm importdisk生成的系统生效还是第二种方法生效的，同时操作的，不知道是哪个起作用
> 
> df -h信息
> Filesystem Size Used Available Use% Mounted on
> tmpfs 1.9G 8.2M 1.9G 0% /tmp
> overlayfs:/overlay 91.4M 22.0M 62.5M 26% /
> /dev/sda1 15.7M 3.6M 11.8M 24% /boot
> tmpfs 512.0K 0 512.0K 0% /dev
> /dev/loop0 91.4M 22.0M 62.5M 26% /mnt/loop0
> /dev/sda1 15.7M 3.6M 11.8M 24% /mnt/sda1

第二个，扩展空间，qm importdisk 命令并不会改变格式，格式是编译时指定的。

authon, 2020-02-11:
> 第二个，扩展空间，qm importdisk 命令并不会改变格式，格式是编译时指定的。

OK，那我扩容了，感谢回答

closed 2020-02-11 by authon